intel/media-driver #1378: [Tiger Lake] VAInfo fails with vaInitialize code 1

Parmjot-Singh, 2022-04-04:
## System information
- CPU information: `11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz`
- GPU information: `Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49] (rev 01)`
- Display server if rendering to display: `X.org`

## Issue behavior
### Describe the current behavior
I run `vainfo`, because I want to enable Hardware Acceleration on my laptop, and I need to verify it, but it comes up with this error:
`[hexagon@dell ~]$ vainfo
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
`
### Describe the expected behavior
It should have come up with information about the driver? This example is taken from ArchWiki:
`libva info: VA-API version 0.39.4
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
vainfo: VA-API version: 0.39 (libva 1.7.3)
vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 1.7.3
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice`
## Debug information
- What's libva/libva-utils/gmmlib/media-driver version? `intel-media-driver 22.2.2-1 from Arch Repository`
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
`[hexagon@dell ~]$ ls /dev/dri
by-path  card0  renderD128`
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
`libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit`
- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
`[hexagon@dell ~]$ vainfo
libva info: Open new log file /tmp/libva_trace.log.204235.thd-0x00000af4 for the thread 0x00000af4
libva info: LIBVA_TRACE is on, save log into /tmp/libva_trace.log.204235.thd-0x00000af4
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit`
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`? Doubt this is the case.
- Do you want to contribute a patch to fix the issue? (yes/no): yes

PHLAK, 2022-04-05:
I'm seeing the same problem

```
$ vainfo 
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code -1 (unknown libva error),exit
```

Here's the trace log but I don't think it will be of much help.

```
[41335.292354][ctx       none]==========va_TraceInitialize
[41335.292357][ctx       none]=========vaInitialize ret = VA_STATUS_ERROR_UNKNOWN, unknown libva error 
[41335.292359][ctx       none]==========va_TraceTerminate
[41335.292365][ctx       none]=========vaTerminate ret = VA_STATUS_SUCCESS, success (no error)
```

dblaber, 2022-04-05:
Having the same issue (also on arch)
**CPU**: Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz

[1] dmb@thinkpad> lspci -nn|grep "VGA\|3D"                                                                                                                                                                     ~
00:02.0 VGA compatible controller [0300]: Intel Corporation CometLake-U GT2 [UHD Graphics] [8086:9b41] (rev 02)
2d:00.0 3D controller [0302]: NVIDIA Corporation GP108GLM [Quadro P520] [10de:1d34] (rev a1)

dmb@thinkpad> vainfo --display drm --device /dev/dri/card1                                                                                                                                                     ~
libva info: Open new log file out.txt.173121.thd-0x00008f3d for the thread 0x00008f3d
libva info: LIBVA_TRACE is on, save log into out.txt.173121.thd-0x00008f3d
vaInitialize failed with error code -1 (unknown libva error),exit
                                                                                                                                                        ~
dmb@thinkpad> vainfo --display drm --device /dev/dri/card0                                                                                                                                                     ~
libva info: Open new log file out.txt.173147.thd-0x00008f4f for the thread 0x00008f4f
libva info: LIBVA_TRACE is on, save log into out.txt.173147.thd-0x00008f4f
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit           

Log shows exactly as what @PHLAK  and @Parmjot-Singh shared.

jike212, 2022-04-05:
Had the same issue while testing Fedora 36 using the RPMFusion repos on my Tiger Lake computer. `vainfo` returns the same error as mentioned above. 

I then installed the latest RPM from here for 22.3.1 and the error persisted.

LaserEyess, 2022-04-06:
FYI for those on arch, this is a [packaging issue ](https://bugs.archlinux.org/task/74341?project=5&string=intel-media-driver)

Parmjot-Singh, 2022-04-06:
So if I were to build from source, I won't get this error?

LaserEyess, 2022-04-06:
Yes, you can either rebuild intel-media-driver from source and install it, or you can downgrade intel-gmmlib to 22.0.3. I assume the arch packager will do a revbump soon, though.

Parmjot-Singh, 2022-04-07:
It doesn't build.
`
In file included from /home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:32:
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.h:307:30: error: ‘VAEncPictureParameterBufferAV1’ does not name a type; did you mean ‘VADecPictureParameterBufferAV1’?
  307 |     VAStatus CheckCDEF(const VAEncPictureParameterBufferAV1 *picParams,
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                              VADecPictureParameterBufferAV1
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/cp/ddi/media_libva_cp_interface.cpp.o
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_VP.dir/agnostic/gen9_cml/vp/kernel/tgp/igvpkrn_g9_cml_tgp.c.o
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp: In member function ‘VAStatus DdiEncodeAV1::ParseSeqParams(void*)’:
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:407:5: error: ‘VAEncSequenceParameterBufferAV1’ was not declared in this scope; did you mean ‘VAEncSequenceParameterBufferVP9’?
  407 |     VAEncSequenceParameterBufferAV1 *seqParams = (VAEncSequenceParameterBufferAV1 *)ptr;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     VAEncSequenceParameterBufferVP9
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_VP.dir/agnostic/gen9_cml/vp/kernel/igvpkrn_g9_cml.c.o
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:407:38: error: ‘seqParams’ was not declared in this scope; did you mean ‘MfeParams’?
  407 |     VAEncSequenceParameterBufferAV1 *seqParams = (VAEncSequenceParameterBufferAV1 *)ptr;
      |                                      ^~~~~~~~~
      |                                      MfeParams
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:407:84: error: expected primary-expression before ‘)’ token
  407 |     VAEncSequenceParameterBufferAV1 *seqParams = (VAEncSequenceParameterBufferAV1 *)ptr;
      |                                                                                    ^
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/cp/ddi/media_libva_caps_cp_interface.cpp.o
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp: In member function ‘virtual VAStatus DdiEncodeAV1::ParsePicParams(DDI_MEDIA_CONTEXT*, void*)’:
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:448:5: error: ‘VAEncPictureParameterBufferAV1’ was not declared in this scope; did you mean ‘VADecPictureParameterBufferAV1’?
  448 |     VAEncPictureParameterBufferAV1 *picParams = (VAEncPictureParameterBufferAV1 *)ptr;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     VADecPictureParameterBufferAV1
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:448:37: error: ‘picParams’ was not declared in this scope; did you mean ‘MfeParams’?
  448 |     VAEncPictureParameterBufferAV1 *picParams = (VAEncPictureParameterBufferAV1 *)ptr;
      |                                     ^~~~~~~~~
      |                                     MfeParams
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:448:82: error: expected primary-expression before ‘)’ token
  448 |     VAEncPictureParameterBufferAV1 *picParams = (VAEncPictureParameterBufferAV1 *)ptr;
      |                                                                                  ^
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp: In member function ‘VAStatus DdiEncodeAV1::ParseTileGroupParams(void*, uint32_t)’:
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:698:5: error: ‘VAEncTileGroupBufferAV1’ was not declared in this scope
  698 |     VAEncTileGroupBufferAV1 *vaEncTileGroupParams = (VAEncTileGroupBufferAV1 *)ptr;
      |     ^~~~~~~~~~~~~~~~~~~~~~~
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_VP.dir/agnostic/gen9_cmpv/vp/kernel/igvpkrn_g9_cmpv.c.o
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:698:30: error: ‘vaEncTileGroupParams’ was not declared in this scope; did you mean ‘ParseTileGroupParams’?
  698 |     VAEncTileGroupBufferAV1 *vaEncTileGroupParams = (VAEncTileGroupBufferAV1 *)ptr;
      |                              ^~~~~~~~~~~~~~~~~~~~
      |                              ParseTileGroupParams
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp:698:79: error: expected primary-expression before ‘)’ token
  698 |     VAEncTileGroupBufferAV1 *vaEncTileGroupParams = (VAEncTileGroupBufferAV1 *)ptr;
      |                                                                               ^
/home/hexagon/build-media-driver/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_av1.cpp: In member function ‘VAStatus DdiEncodeAV1::ParseSegMapParams(void*)’:
[... cut ...]
[  8%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/cp/os/mos_os_cp_interface_specific.cpp.o
[  8%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/ddi/media_ddi_base.cpp.o
[  8%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_VP.dir/agnostic/gen10/vp/kdll/hal_kernelrules_g10.c.o
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/build.make:4430: media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/codec/ddi/media_ddi_encode_av1.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
`

Parmjot-Singh, 2022-04-13:
Closed because `intel-media-driver` got updated on the Arch Repo so it should work for everyone.

closed 2022-04-13 by Parmjot-Singh