intel/media-driver #1379: Inter11 generation of CPU on Synology system can't use Hardware decode

zailushang2008, 2022-04-06:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`):N5095
- GPU information(`lspci -nn | grep -E 'VGA|isplay`):UHD
- Display server if rendering to display(X or wayland):

## Issue behavior
### Describe the current behavior

### Describe the expected behavior

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?YES
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
- Do you want to contribute a patch to fix the issue? (yes/no):YES

I couldn't enable GUC/NUC because of the older kernel used by Synology,Is there a solution to this？thank you very much!

XinfengZhang, 2022-04-08:
need more information,  1. vatrace 2. test cases and command line or reproducer 3. how about the result of vainfo?

according https://ark.intel.com/content/www/us/en/ark/products/212322/intel-celeron-processor-n5095-4m-cache-up-to-2-90-ghz.html
it is JSL with device ID 0x4E55
it do support Hardware decoding

about GuC/Huc firware enabling, suppose , you need provide the kernel version and firmware version

Jexu, 2023-04-21:
It has no update for long time, and i will close this issue now. Please re-try again with latest media driver, you could re-open it again if still having any issues or questions for this.