intel/media-driver #1380: [TGL]VPP sharpen doesn't support conversion on input for YV12->NV12/I420

fanli-hub, 2022-04-08:
## System information
- CPU information:
11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
- GPU information:
0000:00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:9a49] (rev 01)

Gst cmd:
gst-launch-1.0 -vf filesrc location=/opt/media/src/assets/otc-media/sanity/yuv/1920x1080.yv12 num-buffers=1 ! rawvideoparse format=yv12 width=1920 height=1080 ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=I420 ! msdkvpp hardware=true detail=99 ! video/x-raw,format=YV12 ! checksumsink2 file-checksum=false qos=false frame-checksum=false plane-checksum=false dump-output=true dump-location=w1920_h1080_YV12_1920x1080_sharpen_99_YV12.yuv eos-after=1

According to the media driver wiki, vpp/sharpen doesn't support the input format of YV12, but supports the input format of NV12. 
In the gstreamer cmd, it inserts a conversion on input for YV12->NV12/I420. So the gst cmd should work, and the input and output video's md5 values should be different. 

Output:
root@media# md5sum w1920_h1080_YV12_1920x1080_sharpen_99_YV12.yuv
4fc7610410207c17e5bea5f8302aa635 w1920_h1080_YV12_1920x1080_sharpen_99_YV12.yuv
root@mediai# md5sum /opt/media/ci/assets/otc-media/sanity/yuv/1920x1080.yv12
4fc7610410207c17e5bea5f8302aa635  /opt/media/ci/assets/otc-media/sanity/yuv/1920x1080.yv12

However, the values of md5 are the same. It would be better if the driver specifies it's unsupported.

FurongZhang, 2022-04-10:
ok, will check that.

LhGu, 2022-05-24:
Have contacted others to double confirm it and change media driver docs

LhGu, 2022-05-25:
@uartie @fanli-hub  Others have double confirmed. And media driver docs has been changed. 
We could not support YV12 as output format for sharpness. If you do it , sharpness will be skipped.   If no other concerns, could you help close this issue? Thanks.

fanli-hub, 2022-05-27:
Could you share media driver docs with me?  Thanks.

LhGu, 2022-05-27:
> Could you share media driver docs with me? Thanks.

@fanli-hub Maybe I could not paste inner wiki link here. Its link is the same as you reference in the past.

fanli-hub, 2022-05-27:
Thank you for your update. I will close this issue.

closed 2022-05-27 by fanli-hub