intel/media-driver #1383: "readlink -errno 2" segfault when device does not have "driver"

omcaif, 2022-04-12:
## System information
- Impacted GPU platform: All?
- Host machine if it's descreted card: ?
- Good libva/libva-utils/gmmlib/media-driver version? media-driver 22.1.0
- Bad libva/libva-utils/gmmlib/media-driver version? media-driver 22.3.1

## Issue behavior
### Describe the current behavior
When trying to initialize iHD vaapi (e.g. running `vainfo`), it seems to try opening every card in `/dev/dri` then going to its `/sys/devices/pci...` entry to do a `readlink()` on a `driver` file, which it assumes exists. If it does not, then readlink fails and the program (e.g. `vainfo`) segfaults.

The `driver` file does not exist because it goes into my NVIDIA card, which I have unbound from the nvidia module when I am not using it. This is useful for dual GPU setups on laptops where power management is a consideration.

### Describe the expected behavior
If `readlink()` fails or if the `driver` file/link does not exist, then it should ignore this device/card and try another one.

XinfengZhang, 2022-04-17:
the latest vainfo will check two device nodes under /dev/dri and check whether it is render node,  the problem is one of the node is Nvidia card also is valid DRM node?

omcaif, 2022-04-17:
The problem is not only with `vainfo`. I have also seen it happen in FFmpeg and MPV. I tried downgrading only the media driver from 22.3.1 to 22.1.0 and then the issue does not happen.

The problem is that it goes into the Nvidia card `/sys..` entry to use `readlink()` on the `driver` link; however, that link/file does not exist because I have my Nvidia card unbound (using, `echo "0000:01:00.0" >/sys/bus/pci/drivers/nvidia/unbind`). Then `readlink()` fails and puts nothing in the return butter because the file does not exist, but the parsing still continues using that empty buffer, which eventually causes a segfault in `snprintf()`.

XinfengZhang, 2022-04-18:
what's the gmmlib version ?did you use correlated gmmlib version for 22.3.1 and 22.1.0? there are on ABI broken issue with gmmlib, https://github.com/intel/gmmlib/issues/93 , the change was revert finally, but it already impact the tags between gmmlib 22.0.3-22.1.1 .
I suspect the issue is caused by ABI broken which caused random behavior

omcaif, 2022-04-18:
I used gmmlib version 22.1.2 with both media driver version 22.1.0 and 22.3.1.

In pull request #1384 I have included a patch which makes 22.3.1 work for me, which, I think, at least points out where the issue is.

cqian2, 2022-04-21:
Seems like PR https://github.com/intel/media-driver/pull/1384 can solve the problem. Please help try again.

omcaif, 2022-04-21:
I was the one who made that PR, and I have been using that patch since then. No problems so far. It seems to work as expected.

cqian2, 2022-04-21:
thanks, @omcaif

PR #1384 (2022-04-27): [CM] Skip parsing info when device is unbound
When a device is not bound to a driver, the `driver` link does not
exist, so we can stop parsing the `sysfs` files, as that device is not
useful, and return.

Fixes #1383,