intel/media-driver #1387: Regression: Build failure since 22.1.1

th0ma7, 2022-04-16:
## System information
- Impacted GPU platform:  any
- Host machine if it's descreted card: any
- Good libva/libva-utils/gmmlib/media-driver version? `libva`+`libva-utils`: 2.14.0, `gmmlib`: 22.1.2, `intel-med-driver`: 22.1.**0**
- Bad libva/libva-utils/gmmlib/media-driver version? `intel-media-driver`: 22.1.**1** and above
- GCC: 4.9.4 (using Synology DSM linux toolchain for cross-compiling)

## Issue behavior
### Describe the current behavior
Build failure with (see detailed log attached [build-x64-6.1.log](https://github.com/intel/media-driver/files/8499114/build-x64-6.1.log)):
```
[ 65%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/os/memory_policy_manager_specific.cpp.o
In file included from /home/spksrc/ffmpeg44/spksrc/cross/intel-media-driver/work-x64-6.1/media-driver-intel-media-22.1.1/media_driver/linux/common/os/mos_interface.cpp:37:0:
/home/spksrc/ffmpeg44/spksrc/cross/intel-media-driver/work-x64-6.1/media-driver-intel-media-22.1.1/cmrtlib/linux/hardware/drm_device.h:79:0: error: "DRM_DEV_NAME" redefined [-Werror]
 #define DRM_DEV_NAME  "%s/card%d"
 ^
In file included from /home/spksrc/ffmpeg44/spksrc/cross/intel-media-driver/work-x64-6.1/media-driver-intel-media-22.1.1/media_common/linux/common/os/mos_os_specific.h:40:0,
                 from /home/spksrc/ffmpeg44/spksrc/cross/intel-media-driver/work-x64-6.1/media-driver-intel-media-22.1.1/media_common/agnostic/common/shared/user_setting/media_user_setting_configure.h:32,
                 from /home/spksrc/ffmpeg44/spksrc/cross/intel-media-driver/work-x64-6.1/media-driver-intel-media-22.1.1/media_common/agnostic/common/shared/user_setting/media_user_setting.h:51,
                 from /home/spksrc/ffmpeg44/spksrc/cross/intel-media-driver/work-x64-6.1/media-driver-intel-media-22.1.1/media_common/agnostic/common/os/mos_os.h:48,
                 from /home/spksrc/ffmpeg44/spksrc/cross/intel-media-driver/work-x64-6.1/media-driver-intel-media-22.1.1/media_softlet/agnostic/common/os/mos_interface.h:51,
                 from /home/spksrc/ffmpeg44/spksrc/cross/intel-media-driver/work-x64-6.1/media-driver-intel-media-22.1.1/media_driver/linux/common/os/mos_interface.cpp:27:
/home/spksrc/ffmpeg44/spksrc/cross/intel-media-driver/work-x64-6.1/install/usr/local/intel-media/include/xf86drm.h:92:0: note: this is the location of the previous definition
 #define DRM_DEV_NAME          "%s/" DRM_PRIMARY_MINOR_NAME "%d"
 ^
...
```

XinfengZhang, 2022-04-16:
it is strange, these two definition already existing long time.

th0ma7, 2022-04-18:
I can confirm that it fails to build with both GCC-7.5 & GCC-4.9.4 (e.g. Synology Linux DSM-7.0 and DSM-6.1).  I did a "manual" git-bisect and found the culprit commit:
```
[       ]  31c1b995f49ef118a4b02f0b8e45bccdacf0d15e (tag: intel-media-22.1.1, origin/intel-media-22.1) Readme update to add Alchemist/ATS-M information.
[       ]  292d764e82903a8a5ceffcb78e6dfa607def263b Bump up version to 22.1.1
[       ]  0b3e3401733693b1394dc380a09d75e4f5ecddc7 [Encode] GLK enable VP8
[       ]  16e1e2348ca27aa40addc79d4425af42687b48a2 [Media Common] Resolve build issues for Alchemist open source free-kernel build
[ ERROR ]  2f786a5ec2747339578a7f9b69db8b2cf2a9f39d Revert "[MOS] Remove the IPC in CreateAssociationNode"
[ ERROR ]  c80a451697ef9196d7c0ce6c3c3eb4a89eea71a0 [Media Common] Reduce library dependency
[  OK   ]  019f74c0d893e6e2b799121147357d7ebbf30998 [Media Common] SetTileModebyForce
[       ]  2d91b661364c052e40167ae7e2f8d9fd4a58fa1d [Decode] Upstream deissue interface of Alchemist
[       ]  0aaa724f94b763ed0c29025dd81be210a7fe98eb [Media Common] VECS PPC Flush
[       ]  9358c601336b613d07c0db451fc29015507f1ed3 [VP] Force scalingPreferred to sfc for non avs sampler case
[       ]  5d98387ff614500cfad4f7fd387ce10f9bfa26b6 [MOS] Dump bindingtable together with command buffer
[       ]  2a78d3eb3d2eaa4b32877289de84f979a7b1aafd [VP] Fix Field1Dual surface missed issue for fc iscaling/filed weaving related cases
[       ]  857128e857dd8925e42fa4a7d668058200bb17bb [VP] Fix typo for eufusionbypass logic
[  OK   ]  b269bc099479d870f05ed35ab3da1b71c783f318 (tag: intel-media-22.1.0) [Encode] Rework AVC slice state
```

Commit details @kankanzh :
```
commit c80a451697ef9196d7c0ce6c3c3eb4a89eea71a0
Author: kankanzh <kankan.zheng@intel.com>
Date:   Thu Dec 30 23:30:02 2021 +0800

    [Media Common] Reduce library dependency
    
    Reduce library dependency and add corresponding implementation
```

cqian2, 2022-05-26:
Kankan and I have been debugging this issue, will provide PR when test passed internally.

kankanzh, 2022-06-06:
i built with below config and no error:
libva+libva-utils: 2.14.0, 
gmmlib: 22.1.2, 
intel-med-driver: 22.1.2.

Could you have a try with clean compile environment? As your error message is redefinition, maybe your early installed header file result in the compile error.

th0ma7, 2022-06-09:
First off, thnx for your time in looking into this.  Just as an FYI, the spksrc SynoCommunity framework runs as a container and is always "clean" when invoked through github PR continuous integration checks.  On my local linux, I'm running that same container and when re-invoking a build I'm always using a clean tree as well.

- libpciaccess: 0.16,
- libva+libva-utils: 2.14.0,
- gmmlib: 22.1.2,
- libdrm: 2.4.110

As it fails on "DRM" section of the code I tried variations on that as well.  This bug was found using libdrm 2.4.110.  I updated to latest 2.4.111 and previously used 2.5.107 but got the exact same results.

intel-med-driver:
```
- 22.4.3 FAIL    (libdrm 2.4.110)
- 22.3.1 FAIL    (libdrm 2.4.110)
- 22.2.2 FAIL    (libdrm 2.4.107, 2.4.110, 2.4.111)
- 22.1.2 -NOT AVAILABLE-
- 22.1.1 FAIL    (libdrm 2.4.107)
- 22.1.0 -> c80a451697ef9196d7c0ce6c3c3eb4a89eea71a0 FAIL    (libdrm 2.4.111)
- 22.1.0 -> 019f74c0d893e6e2b799121147357d7ebbf30998 SUCCESS (libdrm 2.4.111)
- 22.1.0 SUCCESS (libdrm 2.4.107, 2.4.111) -> version we're currently shipping in our packages
```
I also looked if there wheren't any conflicting headers part of Synology toolchain but I couldn't find anything relevant.

kankanzh, 2022-06-13:
Could you try with below change? I made the change based on your shared log as I cannot reproduce the build failure from my side.

Source/media/media_common/linux/common/os/mos_os_specific.h | 1 -
 1 file changed, 1 deletion(-)

diff --git a/Source/media/media_common/linux/common/os/mos_os_specific.h b/Source/media/media_common/linux/common/os/mos_os_specific.h
index 246a005ff9af..323f3cc380dd 100755
--- a/Source/media/media_common/linux/common/os/mos_os_specific.h
+++ b/Source/media/media_common/linux/common/os/mos_os_specific.h
@@ -37,7 +37,6 @@
 #endif
 #include "i915_drm.h"
 #include "mos_bufmgr.h"
-#include "xf86drm.h"
 
 #include <vector>

th0ma7, 2022-06-25:
Sorry for the delay in anwering.  Tried it and still getting the same error:  [build-x64-7.0.log](https://github.com/intel/media-driver/files/8984745/build-x64-7.0.log)

You can easily test it against the spksrc framework:
1. Just clone our repo https://github.com/SynoCommunity/spksrc
2. You'll need a docker or lxc container (see info above)
3. Move to `spksrc/cross/intel-media-driver`
4. Invoke build `make arch-x64-7.0` or `make arch-x64-6.1` (7.0 and 6.1 refers to Synology Linux DSM version)
5. Change the version by editing the `Makefile` then `make clean && make digests`
6. re-invoke the build & repeat.

th0ma7, 2022-09-01:
@kankanzh and @cqian2 somewhere something changed for the best!  Using the following now works just fine:
- intel-media-driver: 22.4.4
- gmmlib: 22.1.4
- libdrm: 2.4.113
- libva: 2.15.0

No clue what hapen exactly but no change happend on our framework side :shrug:

closed 2022-09-01 by th0ma7