intel/media-driver #1389: VPP: Tonemapping can not work with CSC/Scale at the same time.

HeJunyan, 2022-04-22:
## System information
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): All Gen9.5+ platform

## Issue behavior
### Describe the current behavior
We find that the VAProcFilterHighDynamicRangeToneMapping filter can not work with the other filters such as VAProcFilterSharpening at one shot. We can neither do the tonemapping with CSC or scale at the same time.

### Describe the expected behavior
We want the tonemapping work together with other VPP features such as CSC, scale within on one vaBeginPicture/vaEndPicture pair.

The current tonemapping can not work if the input and output have different format or resolution. For example, we need to use two steps of vpp to complete the conversion of 4k-HDR --> 400x300-SDR:

gst-launch-1.0 -vf filesrc location=Sony_Swordsmith.mp4 ! qtdemux ! h265parse ! vah265dec ! vapostproc hdr-tone-mapping=1 ! vapostproc ! video/x-raw,format=NV12,width=400,height=300 !  filesink location=sdr.yuv

We expect to use one step to complete this.

FurongZhang, 2022-04-26:
@HeJunyan , VAProcFilterHighDynamicRangeToneMapping will not be combined with a few VP features. We have verified, Tone Mapping should be working together with scaling. 

As for CSC, what kinds of color space conversion do you need?

HeJunyan, 2022-04-26:
@FurongZhang Very thanks. I have already verified on TGL, the VPP can really do the CSC(not all format, but already meet our requirement), scaling and ToneMapping at the same time. That's really good for me.

And if you do not want to implement the combination with other VP filters(I think may be HW limitation), can you help to declare it in some place of the driver docs or on its home page? Then we can have a solid reason to avoid ToneMapping and other VP filters happening at the same time in FFMpeg and Gstreamer.

FurongZhang, 2022-04-27:
@HeJunyan , thanks for the information. Sure, Let me think about it, then reply to you.

softworkz, 2022-05-19:
> I have already verified on TGL, the VPP can really do the CSC(not all format, but already meet our requirement), scaling and ToneMapping at the same time. That's really good for me.

Doing scaling before tone mapping (in a separate VPP) should be faster in total (when scaling is done after tm as indicated in the new comment).

FurongZhang, 2022-05-19:
We have already had such optimization in the current driver. But this optimization is only for scaling + 3dlut. As of now, not applied to all the features.

FurongZhang, 2022-05-19:
@HeJunyan , if you don't mind, close this issue? Thanks!

closed 2022-05-22 by FurongZhang