intel/media-driver #1390: Encoder: HEVC: Low Power encoder generates broken stream if B frames' reference number is bigger than 3

HeJunyan, 2022-04-23:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): Tiger Lake
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): Gen12(Or all the platforms which support B and LDB)

## Issue behavior
### Describe the current behavior
For HEVC encoder, if we set the reference number of B frame > 3,  the output coded stream is broke, as attached.
The VAConfigAttribEncMaxRefFrames return 3 for L0 and 3 for L1, this is correct for Low Delay B frame, which has the same L0 and L1 list. So in fact, there are only 3 reference frames for LDB kind frame.
But for normal B frame, the user does not know the HW limitation of the reference number, and still feed 3 forward frames as List0 and 1 or 2 backward frames as list1. So the total reference number is 3+1=4 or 3+2=5, which may exceeds the HW limitation and we get a broken coded result for this B frame.
The middleware  or application does not know the max reference number it can provide to driver and always fully use the query result of VAConfigAttribEncMaxRefFrames to achieve the best compression.

The attached file:
When the limit total reference number to 3, the result.
If the reference number is bigger than 3, the B frames are broken.

### Describe the expected behavior
The driver should add some protection to avoid the reference exceeds the HW limitation.

HeJunyan, 2022-05-10:
Sorry, the result is correct.

Tianhaol, 2022-05-27:
Could you provide the tools to reproduce this issue?

HeJunyan, 2022-05-27:
By now, only the gstreamer can produce this.
You may need to clone
https://gitlab.freedesktop.org/He_Junyan/gstreamer
and switch to H265_encoder_dev branch.

Because of this and several other bugs, I can not send the MR now. Please use my dev branch.

Then, use the command line:
gst-launch-1.0 -vf videotestsrc num-buffers=500 ! video/x-raw,height=1280,width=720,format=NV12 ! vah265lpenc key-int-max=60 b-frames=7 ref-frames=5 ! fakesink

Xiaogangli-intel, 2022-05-30:
Thank you @HeJunyan, so could you help confirm if #1420 can fix this issue?

Tianhaol, 2022-06-02:
The gstream built by me,  there are no features when query vaapi.
bin/gst-inspect-1.0 ./lib64/gstreamer-1.0/libgstvaapi.so
Plugin Details:
  Name                     vaapi
  Description              VA-API based elements
  Filename                 ./lib64/gstreamer-1.0/libgstvaapi.so
  Version                  1.21.0.1
  License                  LGPL
  Source module            gstreamer-vaapi
  Binary package           gstreamer-vaapi
  Origin URL               Unknown package origin

  0 features:

uartie, 2022-06-02:
> The gstream built by me, there are no features when query vaapi. bin/gst-inspect-1.0 ./lib64/gstreamer-1.0/libgstvaapi.so Plugin Details: Name vaapi Description VA-API based elements Filename [...]

I encountered this same behavior the other day on Ubuntu build environment.  Found that libudev-dev was missing but build still succeeds and then 0 features appear.  After installing libudev-dev and rebuild, it worked for me.  Guess it's a bug in gstreamer meson configure.

HeJunyan, 2022-06-02:
This bug need to use VA plugins, not vaapi. Do you just want to use old vaapi plugins as a comparison?

Tianhaol, 2022-06-02:
@uartie  Thank you.  I'm installing libudev-dev and try again.

Tianhaol, 2022-06-02:
@HeJunyan  Are the VA plugins are enabled by default?  Are the old vaapi plugins are abandoned?

HeJunyan, 2022-06-02:
Yes, you only need to fix va plugins, vaapi will be deprecated soon.  And the old vaapi only set 3 as its max reference num and no this issue can be found.

VA plugins are default enabled in -bad.