intel/media-driver #1391: Encoder: HEVC: The VAConfigAttribValEncHEVCBlockSizes and VAConfigAttribValEncHEVCFeatures is not precise for VME

HeJunyan, 2022-04-23:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): Tiger Lake
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): Gen12

## Issue behavior
### Describe the current behavior
The query results of VAConfigAttribValEncHEVCBlockSizes and VAConfigAttribValEncHEVCFeatures for VME(VAEntrypointEncSlice) are not very precise. For example, log2_max_coding_tree_block_size_minus3 is still set to 2, which causes the CTU size still be 32.
In fact, according to MSDK's result, this can be set to 3 and set CTU size to 64. This obsolete parameters do not block our encoding result, but the compression performance may not be good enough.

### Describe the expected behavior
Please update these parameters setting, for example, sync with MSDK's setting.

Xiaogangli-intel, 2022-06-02:
Thanks @HeJunyan, will correct the value of log2_max_coding_tree_block_size_minus3. Except this, did you notice others?

HeJunyan, 2022-06-02:
And VAConfigAttribValEncHEVCFeatures.cu_qp_delta, MSDK use 3 but the driver gives 2

Xiaogangli-intel, 2022-06-02:
Hi @HeJunyan, VAConfigAttribValEncHEVCFeatures.cu_qp_delta should be 2 (VA_FEATURE_REQUIRED, VA_FEATURE_SUPPORTED=1, VA_FEATURE_NOT_SUPPORTED =0). So this field is correct.

HeJunyan, 2022-06-02:
OK. I just wonder where MSDK get the 3 for this.

PR #1429 (2022-06-14): Optimize SCC capability report for Gen12 HEVC
1. optimize call flow
2. add a new attribute list for SCC to correct some incorrect values
3. fixes #1391