intel/media-driver #1392: VPP: Should not report VAProcFilterHighDynamicRangeToneMapping ability on Gen9- platform.

HeJunyan, 2022-04-23:
## System information
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): Gen9 and older platform.

## Issue behavior
### Describe the current behavior
It seems that the tone mapping feature is introduced in Gen9.5, so the old platform should not report that ability.
The driver reports this feature now. But it can not really work when we try to use them on Gen9.

### Describe the expected behavior
The driver should not report this feature on Gen9- platform.

FurongZhang, 2022-04-24:
HDR Tone Mapping is supported from KBL.

HeJunyan, 2022-04-24:
KBL is gen9.5 and SKL is Gen9, which should not have this feature.

FurongZhang, 2022-04-26:
> KBL is gen9.5 and SKL is Gen9, which should not have this feature.

Got it. Understand the issue. Will change that. Thanks for the report.

FurongZhang, 2022-05-17:
Fix PR: https://github.com/intel/media-driver/pull/1409

closed 2022-05-22 by FurongZhang