intel/media-driver #1394: Regression: media-driver >= 22.3.0 crashes in KernelDll_AllocateStates in builds with -DENABLE_NONFREE_KERNELS=OFF

sebastinas, 2022-04-24:
## System information
- Impacted GPU platform: KBL
- Host machine if it's descreted card:
- Good libva/libva-utils/gmmlib/media-driver version? Anything before media-driver 22.3.0
- Bad libva/libva-utils/gmmlib/media-driver version? media-driver 22.3.0 and newer

## Issue behavior
Forwarding a bug from a Debian user. See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1008776 for the original report.

### Describe the current behavior
media-driver crashes on KBL with a SIGSEGV when built with -DENABLE_NONFREE_KERNELS=OFF:
```
$ gdb --args gst-play-1.0 /home/kai/Downloads/pony.mp4 --videosink=glimagesink
Thread 14 "qtdemux0:sink" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb77fe640 (LWP 22147)]
0x00007fffb5f906be in KernelDll_AllocateStates (pKernelBin=<optimized out>, uKernelSize=0, pFcPatchCache=0x0, 
    uFcPatchCacheSize=<optimized out>, pDefaultRules=0x0, ModifyFunctionPointers=0x0)
    at ./media_driver/agnostic/common/vp/kdll/hal_kerneldll.c:2791
Download failed: Invalid argument.  Continuing without source file ./obj-x86_64-linux-gnu/media_driver/./media_driver/agnostic/common/vp/kdll/hal_kerneldll.c.
2791    ./media_driver/agnostic/common/vp/kdll/hal_kerneldll.c: No such file or directory.
```

### Describe the expected behavior
No crash

Xiaogangli-intel, 2022-04-25:
Thanks, @sebastinas. Is that possible that provide the input stream Or the codec(AVC or ?).

kwbr, 2022-04-25:
(I am the original reporter of this bug in Debian)
One example is a video file downloaded with yt-dlp from `https://mobile.twitter.com/julius_riemann/status/1504369766470983683`.

Complete output of `mediainfo` attached.

[log.txt](https://github.com/intel/media-driver/files/8552978/log.txt)

vanvugt, 2022-04-29:
Downstream Ubuntu bug: https://launchpad.net/bugs/1962630

Also seems similar to #1132.

LhGu, 2022-05-30:
> @Xiaogangli-intel Any news here?

This issue WIP.

LhGu, 2022-06-01:
@sebastinas @kwbr  I root-caused it. In KBL, media-driver doesn’t support VP shaders with with ENABLE_NONFREE_KERNELS=OFF.  In this free kernel driver, pKernelbinary will be a null pointer, and mem size is 0.  I will add some codes to avoid  driver crash.

commit e25478c4 (2022-06-09): Fix driver crash with no kernel binary

Fixes #1394
closed 2022-06-09 by intel-mediadev