intel/media-driver #1395: Hevc encoder fails to set hierarchical_level_plus1 on old platform such as Gen9.

HeJunyan, 2022-04-25:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): SKL
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): 00:02.0 VGA compatible controller [0300]: Intel Corporation Skylake GT2 [HD Graphics 520] [8086:1916] (rev 07)

## Issue behavior
### Describe the current behavior
If we set the `hierarchical_level_plus1` field of VAEncPictureParameterBufferHEVC  >= 1, the encoder will faill.
We get error:
[CODEC]:CRITICAL - GetFrameBrcLevel:1529: FrameLevel can only be 0 for I type for LDB

While on TigetLake, according to MSDK result. The I and LDB set this value to 1, and B set to 2 and higher.

### Describe the expected behavior
Please unify the usage of this flag and at least do not report a error.

Xiaogangli-intel, 2022-05-05:
Hi @HeJunyan, we checked the source code again, seems it's not an issue, but different behavior on different platform because of the required input parameter of media kernel. 0/1 have different meaning on SKL and TGL.

HeJunyan, 2022-05-05:
All the open source code like FFMpeg and Gstreamer does not check the platform. How can we use these flags?
Though I know the HW may have different requirements, it's one of the driver's important job to provide a uniform API and usage to user. At least, you should not let the encode fail.

HeJunyan, 2022-05-25:
If you want to have a try by yourself, you need to use 
https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/2036
MR of gstreamer and cmd is:

gst-launch-1.0 -vf videotestsrc ! video/x-raw,format=NV12,width=400,height=300 ! vah265enc b-frames=3 b-pyramid=1 ! fakesink

Xiaogangli-intel, 2022-05-30:
Thank you @HeJunyan, so could you help confirm if https://github.com/intel/media-driver/pull/1420 can fix this issue?

HeJunyan, 2022-05-30:
Yes, verified, OK to run.

PR #1420 (2022-06-09): Avoid incorrect error reporting for LDB when HierarchicalFlag > 0
When HierarchicalFlag > 0 for I frame LDB, will ignore the vaule of HierarchLevelPlus1. For compatibility purpose, we keep m_hevcPicParams->HierarchLevelPlus1 == 0 here. Fixes #1395
closed 2022-06-09 by intel-mediadev