intel/media-driver #1397: H264 encoder can not handle level idc >= 6

HeJunyan, 2022-04-26:
## System information
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): All platforms.

## Issue behavior
### Describe the current behavior
The driver can not handle the level IDC bigger than 6.0
According to the newest H264 spec:
https://www.itu.int/rec/T-REC-H.264-201906-S/en

It adds 3 new levels for 4K streams:
6 4 177 920 139 264 696 320 240 000 240 000 8 192 2 16
6.1 8 355 840 139 264 696 320 480 000 480 000 8 192 2 16
6.2 16 711 680 139 264 696 320 800 000 800 000 8 192 2 16

### Describe the expected behavior
Driver should add support for Level IDC bigger than 6

## Debug information
The command line will fail:
gst-launch-1.0 -vf videotestsrc num-buffers=2000 ! video/x-raw,format=NV12,width=3840,height=2160 ! vah264enc rate-control=vbr bitrate=142857 cpb-size=285714 ! video/x-h264,profile=main ! fakesink

because it will select level IDC 6.1

and driver report error
[CODEC]:CRITICAL - CodecHalAvcEncode_GetProfileLevelMaxFrameSize:529: Unsupported LevelIDC setting.

Xiaogangli-intel, 2022-04-29:
Hi @HeJunyan, do you have real usage requirement for these new levels now?

HeJunyan, 2022-04-29:
Yes,

gst-launch-1.0 -vf videotestsrc num-buffers=2000 ! video/x-raw,format=NV12,width=3840,height=2160 ! vah264enc rate-control=vbr bitrate=142857 cpb-size=285714 ! video/x-h264,profile=main ! fakesink

will fill because it choose 6.1 level.
cpb-size exceed the level limitation of 5.2. In theory, user can specify this value.

OttoYang, 2023-03-28:
The max frame size of AVC supported in hardware is 4k, so the max CodecLevel is 5.2 
For such hardware limitations, we return unsupported and expect the user to know this and correct the parameter.  
And could we close this issue?

HeJunyan, 2023-03-28:
OK, make it as a known HW limitation.

closed 2023-03-28 by HeJunyan