intel/media-driver #1398: [VDSFC] leak of support NULL VARectangle parameters.

feiwan1, 2022-04-27:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`):Intel(R) Core(TM) i5-1035G1 CPU @ 1.00GHz
- GPU information(`lspci -nn | grep -E 'VGA|isplay`):lspci -nn | grep -E 'VGA|isplay
- Display server if rendering to display(X or wayland):x

## Issue behavior
### Describe the current behavior
According to va_vpp.h, when set output_region/surface_region as NULL in VAProcPipelineParameterBuffer, that means to use the whole surface:
https://github.com/intel/libva/blob/2aa18e3125ad3dd04173fc6292a955d041fcc6c7/va/va_vpp.h#L902

But media-driver now directly use output_region/surface_region without checking if it is NULL pointers. So when app set the pointers as NULL, it will segment fault:
https://github.com/intel/media-driver/blob/a1888324a833b2a2671221909a8dc68f59832471/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp#L129

I tried to add the NULL check in above code before call output_region/surface_region, the dumped sfc output will have some black frames.

I used ffmpeg for my test:
https://github.com/intel-media-ci/ffmpeg/pull/524
And need change current code's output_region/surface_region to NULL in vaapi_hevc.c:306/307.

Cmdline:
$ make examples
$ ffmpeg/doc/examples/vaapi_decode 1920x1080_20.h265 out.yuv width=640:height=480:format=argb sfc.yuv

### Describe the expected behavior
When set output_region/surface_region to NULL, the dumped yuv can be played without black frames.

Jexu, 2022-06-08:
Yeah, it should use the whole surface if there is no region provided. Can you help try above PR for your test.

feiwan1, 2022-06-08:
Your patch works for me. Thanks @Jexu .

Jexu, 2022-06-08:
There is still a question, 2 frames sfc fox example, first frame provides region info while second one doesn't?

feiwan1, 2022-06-08:
Not yet, ffmpeg only support one sfc output now.

Jexu, 2022-06-09:
Ok, I will get the PR merged later and close this issue after that.

PR #1431 (2022-06-10): [Decode] Use the whole surface for processing if there is no region p…
…rovided

Fixes #1398
commit 8249039b (2022-06-10): [Decode] Use the whole surface for processing if there is no region provided

Fixes #1398
closed 2022-06-10 by intel-mediadev