intel/media-driver #1402: JPEGBaseline decoder doesn't expose RGBP pixel format for RGB16/RGB32 rt format

ceyusa, 2022-05-05:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): model name      : Intel(R) Core(TM) i5-10210U CPU @ 1.60GHz
- GPU information(`lspci -nn | grep -E 'VGA|isplay`):  00:02.0 VGA compatible controller [0300]: Intel Corporation CometLake-U GT2 [UHD Graphics] [8086:9b41] (rev 02)
- Display server if rendering to display(X or wayland): Wayland

## Issue behavior
### Describe the current behavior

While querying for `vaQuerySurfaceAttributes`  for config `VAProfileJPEGBaseline`/`VAEntrypointVLD` and rt format is any supported, the same pixel formats are always shown.

For example, with RGB16 and RGB32 with [vadumpcaps](https://github.com/fhvwy/vadumpcaps/commits/master):

```
                       {
                            "rt_format": "RGB16",
                            "max_width": 16384,
                            "max_height": 16384,
                            "memory_types": [
                                "VA",
                                "USER_PTR",
                                "KERNEL_DRM",
                                "DRM_PRIME",
                                "DRM_PRIME_2",
                            ],
                            "pixel_formats": [
                                "NV12",
                                "IMC3",
                                "Y800",
                                "411P",
                                "422H",
                                "422V",
                                "444P",
                            ],
                        },
                        {
                            "rt_format": "RGB32",
                            "max_width": 16384,
                            "max_height": 16384,
                            "memory_types": [
                                "VA",
                                "USER_PTR",
                                "KERNEL_DRM",
                                "DRM_PRIME",
                                "DRM_PRIME_2",
                            ],
                            "pixel_formats": [
                                "NV12",
                                "IMC3",
                                "Y800",
                                "411P",
                                "422H",
                                "422V",
                                "444P",
                            ],
                        },
```

While decoding a RGB jpeg with MediaSDK (gst-msdk) the surfaces is created with RGBP pixel format (0x50424752):

(with `LIBVA_TRACE`):

```
[16970.527444][ctx       none]==========va_TraceCreateSurfaces
[16970.527446][ctx       none]  width = 320
[16970.527447][ctx       none]  height = 256
[16970.527447][ctx       none]  format = 1048576
[16970.527447][ctx       none]  num_surfaces = 1
[16970.527448][ctx       none]      surfaces[0] = 0x00000000
[16970.527449][ctx       none]  attrib_list[0] =
[16970.527449][ctx       none]      type = 1
[16970.527450][ctx       none]      flags = 2
[16970.527450][ctx       none]      value.type = 1
[16970.527450][ctx       none]      value.value.i = 0x50424752
[16970.527451][ctx       none]  attrib_list[1] =
[16970.527451][ctx       none]      type = 8
[16970.527451][ctx       none]      flags = 2
[16970.527452][ctx       none]      value.type = 1
[16970.527452][ctx       none]      value.value.i = 0x00000001
[16970.527482][ctx       none]=========vaCreateSurfaces ret = VA_STATUS_SUCCESS, success (no error) 
```

**But RGBP is not exposed when querying the surface attributes!**

### Describe the expected behavior

While querying the surfaces attributes for the JPEG decoding config with RGB rt formats, RGBP pixel format should be advertised.

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version? Debian testing: media-driver 22.4.0+ds1-1 / gmmlib 22.1.2+ds1-1
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?

```
libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.4.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```

- Do you want to contribute a patch to fix the issue? (yes/no): If I find time for it.

ceyusa, 2022-06-08:
I've tested it, and it works as expected

Jexu, 2022-06-09:
Ok, I will get above PR merged.

closed 2022-06-10 by intel-mediadev