intel/media-driver #1407: VPP: Should not report VAProcFilterHighDynamicRangeToneMapping ability on Gemini Lake Graphics (all or some)

softworkz, 2022-05-16:
This is similar to https://github.com/intel/media-driver/issues/1392 but about a different platform

The case where we've seen this is with a Celeron CPU J4025.
(I don't know whether there are other Gemini Lake CPUs where it is working)

Via VAAPI we get the information that VAProcFilterHighDynamicRangeToneMapping would be supported, but when trying to use, it fails.

It is very important that this information is reliable because incorrect information leads to failure which is difficult to identify.

FurongZhang, 2022-05-17:
Since KBL, HDR Tone Mapping is supported in media driver.

softworkz, 2022-05-17:
So you say it should be working with a Celeron J4025?

Also, how about ELK and JSL?

FurongZhang, 2022-05-17:
ELK and JSL No due to lack of VEBOX.

softworkz, 2022-05-17:
And that Celeron?

softworkz, 2022-05-17:
This is our detection output from a Jasper Lake system:

```JSON
"Filters": [
	{
		"FilterId": 1,
		"FilterName": "Noise reduction filter",
		"Properties": [
			{
				"ParamIndex": 0,
				"MinValue": "0.000",
				"MaxValue": "64.000",
				"DefaultValue": "0.000",
				"Step": "1.000"
			}
		],
		"IsSupported": 1
	},
	{
		"FilterId": 2,
		"FilterName": "Deinterlacing filter",
		"Properties": [
			{
				"DeinterlacingType": 1
			},
			{
				"DeinterlacingType": 3
			},
			{
				"DeinterlacingType": 4
			}
		],
		"IsSupported": 1
	},
	{
		"FilterId": 3,
		"FilterName": "Sharpening filter",
		"Properties": [
			{
				"ParamIndex": 0,
				"MinValue": "0.000",
				"MaxValue": "64.000",
				"DefaultValue": "44.000",
				"Step": "1.000"
			}
		],
		"IsSupported": 1
	},
	{
		"FilterId": 4,
		"FilterName": "Color balance parameters",
		"Properties": [
			{
				"ColorBalanceType": 1,
				"MinValue": "-180.000",
				"MaxValue": "180.000",
				"DefaultValue": "0.000",
				"Step": "0.100"
			},
			{
				"ColorBalanceType": 2,
				"MinValue": "0.000",
				"MaxValue": "10.000",
				"DefaultValue": "1.000",
				"Step": "0.010"
			},
			{
				"ColorBalanceType": 3,
				"MinValue": "-100.000",
				"MaxValue": "100.000",
				"DefaultValue": "0.000",
				"Step": "0.100"
			},
			{
				"ColorBalanceType": 4,
				"MinValue": "0.000",
				"MaxValue": "10.000",
				"DefaultValue": "1.000",
				"Step": "0.010"
			},
			{
				"ColorBalanceType": 7,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			}
		],
		"IsSupported": 1
	},
	{
		"FilterId": 5,
		"FilterName": "Skin Tone Enhancement",
		"Properties": [
			{
				"ParamIndex": 0,
				"MinValue": "0.000",
				"MaxValue": "9.000",
				"DefaultValue": "3.000",
				"Step": "1.000"
			}
		],
		"IsSupported": 1
	},
	{
		"FilterId": 6,
		"FilterName": "Total Color Correction",
		"Properties": [
			{
				"ColorCorrectionType": 1,
				"MinValue": "0.000",
				"MaxValue": "255.000",
				"DefaultValue": "160.000",
				"Step": "1.000"
			},
			{
				"ColorCorrectionType": 2,
				"MinValue": "0.000",
				"MaxValue": "255.000",
				"DefaultValue": "160.000",
				"Step": "1.000"
			},
			{
				"ColorCorrectionType": 3,
				"MinValue": "0.000",
				"MaxValue": "255.000",
				"DefaultValue": "160.000",
				"Step": "1.000"
			},
			{
				"ColorCorrectionType": 4,
				"MinValue": "0.000",
				"MaxValue": "255.000",
				"DefaultValue": "160.000",
				"Step": "1.000"
			},
			{
				"ColorCorrectionType": 5,
				"MinValue": "0.000",
				"MaxValue": "255.000",
				"DefaultValue": "160.000",
				"Step": "1.000"
			},
			{
				"ColorCorrectionType": 6,
				"MinValue": "0.000",
				"MaxValue": "255.000",
				"DefaultValue": "160.000",
				"Step": "1.000"
			}
		],
		"IsSupported": 1
	},
	{
		"FilterId": 7,
		"FilterName": "Human Vision System(HVS) Noise reduction filter",
		"Properties": [
			{
				"ParamIndex": 0,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "255.000",
				"Step": "160.000"
			},
			{
				"ParamIndex": 1,
				"MinValue": "160.000",
				"MaxValue": "1.000",
				"DefaultValue": "0.000",
				"Step": "10.000"
			},
			{
				"ParamIndex": 2,
				"MinValue": "0.000",
				"MaxValue": "-0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 3,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "255.000",
				"Step": "160.000"
			},
			{
				"ParamIndex": 4,
				"MinValue": "160.000",
				"MaxValue": "1.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 5,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 6,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 7,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "-nan",
				"Step": "-nan"
			},
			{
				"ParamIndex": 8,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 9,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "-nan",
				"Step": "-nan"
			},
			{
				"ParamIndex": 10,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 11,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 12,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 13,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 14,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			},
			{
				"ParamIndex": 15,
				"MinValue": "0.000",
				"MaxValue": "0.000",
				"DefaultValue": "0.000",
				"Step": "0.000"
			}
		],
		"IsSupported": 1
	},
	{
		"FilterId": 8,
		"FilterName": "High Dynamic Range Tone Mapping",
		"Properties": [
			{
				"MatadataType": 1,
				"CapFlags": 7
			}
		],
		"IsSupported": 1
	},
	{
		"FilterId": 9,
		"FilterName": "Unknown Filter",
		"IsSupported": 0
	}
]
```

And this is the information of the device:

```js
"DeviceInfo": {
	"VendorName": "Intel Corporation",
	"DeviceName": "JasperLake [UHD Graphics]",
	"SubsytemVendorName": "Gigabyte Technology Co., Ltd",
	"VendorId": 32902,
	"DeviceId": 20081,
	"SubsytemVendorId": 5208,
	"SubsytemDeviceId": 4096,
	"DevPath": "/sys/bus/pci/devices/0000:00:02.0",
	"DrmCard": "/dev/dri/card0",
	"DrmRender": "/dev/dri/renderD128",
	"IsEnabled": 1,
	"IsBootVga": 1,
	"ApiVersionMajor": 1,
	"ApiVersionMinor": 13,
	"Driver": "Intel iHD driver for Intel(R) Gen Graphics - 21.4.3 ()"
},
```

softworkz, 2022-05-17:
I don't have that output for the Celeron J4025 (Gemini Lake), but it reported that tone mapping is supported but then it failed.

The very first point we need to find out, is of course: Is it actually **supposed** to work?

(and for JSL, we can be already sure that the filter support API returns incorrect results)

FurongZhang, 2022-05-17:
https://github.com/intel/media-driver/pull/1409/

FurongZhang, 2022-05-17:
Hi @softworkz , I felt sorry. In the original code, it missed the caps report. I added in the PR https://github.com/intel/media-driver/pull/1409/. I am verifying my PR. Please let me know if any comment from you.

softworkz, 2022-05-17:
> Hi @softworkz , I felt sorry. In the original code, it missed the caps report. I added in the PR #1409. I am verifying my PR. Please let me know if any comment from you.

Hi @FurongZhang - thanks a lot for the quick response and fix!

> I added in the PR #1409. I am verifying my PR. Please let me know if any comment from you.

I would help reviewing, but I don't know what is supposed to work and what is not - so I cannot contribute anything to it 😄 

But looking at your pull request 

> GLK (this is a specific requests, perf can not be guarantee due to limited EUs);

does that mean that Gemini Lake should be able to do HDR-to-SDR tone mapping?
Even that Celeron J4025 CPU?

Thanks again,
softworkz

FurongZhang, 2022-05-17:
Gemini Lake, there was a request 2-3 years ago. It should be working.

softworkz, 2022-05-17:
I belongs to "Goldmont Plus" / "Gemini Lake Refresh"

https://en.wikipedia.org/wiki/Goldmont_Plus#:~:text=Goldmont%20Plus%20is%20a%20microarchitecture,platform%20on%20November%204%2C%202019.

softworkz, 2022-05-17:
The "Pentium Silver" CPUs have UHD Graphics 605
the Celeron CPUs have UHD Graphics 600

FurongZhang, 2022-05-17:
Since it is GLK, it should work. I added GLK due to some request from users.

FurongZhang, 2022-05-19:
PR merged. If no more concern or issues, I will close the issue by next week. Thank you!

closed 2022-05-19 by softworkz

FurongZhang, 2022-05-20:
Thank you so much @softworkz  for the information. If possible, May I know what kind of product you are working on? What is it for? Thanks a lot in advance!

FurongZhang, 2022-05-20:
Meanwhile, I will request our validation to fully validate that and enabled that on the automatic system.

softworkz, 2022-05-20:
> Thank you so much @softworkz for the information. If possible, May I know what kind of product you are working on? What is it for? Thanks a lot in advance!

Hi Furong, thanks a lot for asking. I'll respond privately.
Kind regards,
softworkz