intel/media-driver #1408: vaCreateBuffer() should not hold a BufferLock in executing memcpy

rosetta-jpn, 2022-05-17:
### What Feature?

vaCreateBuffer(), if a given |data| is not null, copies |data| into a created buffer.
[The current vaCreateBuffer() implementation](https://github.com/intel/media-driver/blob/427e7910a7528a9220c10586cb26cccf28efafde/media_driver/linux/common/ddi/media_libva.cpp#L3242) acquires a BufferLock in the begging of the function and releases in the end of the function.
Therefore, memcpy is executed while the lock is held.
The bufferlock is basically required to operate on a bufferHeap, if I understand correctly.
So the lock scope should be more narrow; at least memcpy should be performed without holding the lock.

### What's the usage scenario would be benifited?

Video Conference

### What impacted?

The concurrent function calls e.g. vaCreateBuffer() will not be blocked by memcpy in vaCreateBuffer().

rosetta-jpn, 2022-05-17:
@zlma7001 FYI

XinfengZhang, 2023-05-06:
I agree with rosetta-jpn , @MicroYY  please help to narrow the protection area.