intel/media-driver #1412: Package version conflict in Ubuntu 22.04

thetick1234567890, 2022-05-22:
Sorry for the cross post with Intel Compute Runtime but due to libigdgmm12 both the latest  Intel Compute Runtime and Intel Media Driver can not be installed on the Latest Ubuntu LTS 22.04.

Any chance both Intel teams could add post package install to their corresponding  CI/CD pipelines before releasing.  And if needed coordinate together when dependencies change ?

Reference: https://github.com/intel/compute-runtime/issues/538

Sherry-Lin, 2022-05-24:
Thanks for reporting this issue. I believe we do have the package installing checking but not validate all combinations. 
Compute team is preparing a patch to replace "=" with ">=" which should resolve this issue.

thetick1234567890, 2022-06-04:
Latest release is working fine with updated Jammy, 5.18 kernel and last night mesa build.

closed 2022-06-04 by thetick1234567890