intel/media-driver #1415: [Bug]: HEVC transcoding fails on DG1 (0x4905)

eero-t, 2022-05-25:
### Which component impacted?

Decode

### What happened?

**Use-cases**

* `ffmpeg  -hwaccel vaapi -hwaccel_output_format vaapi -i GTAV_1920x1080_60_yuv420p.h265 -c:v h264_vaapi -f null -`
* `sample_multi_transcode -i::h265 GTAV_1920x1080_60_yuv420p.h265 -o::h264 /dev/null`

**Expected outcome**

Both of above do transcoding at hundreds of FPS, like is the case with TGL iGPU, with exactly the same setup.  Or if I change the input to H.264 one.

**Actual outcome**

* FFmpeg transcoding happens at 2 FPS
* OneVPL transcoding fails to:
```
$ sample_multi_transcode -i::h265 /media/GTAV_1920x1080_60_yuv420p.h265 -o::h264 /dev/null
Multi Transcoding Sample Version 8.4.27.0

CONFIGURE LOADER: required implementation: hw 
CONFIGURE LOADER: required implementation mfxAccelerationMode: MFX_ACCEL_MODE_VIA_VAAPI 
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
Session 0:
Loaded Library configuration: 
    Version: 2.7 
    ImplName: mfx-gen 
    Adapter number : 0 
    Adapter type: integrated
    DRMRenderNodeNum: 128 
Used implementation number: 0 
Loaded modules:
   0: /usr/local/lib/libmfxhw64.so.1.35 
   1: /usr/local/lib/libmfx-gen.so.1.2.7 

Pipeline surfaces number (DecPool): 10
Input  video: HEVC
Output video: AVC 

Session 0 was NOT joined with other sessions

Transcoding started

[... 117 lines cut ...]
**Notes**

There are no GPU hangs.  Kernel driver output / settings:
```
# dmesg |grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/drm_intel root=/dev/nvme0n1p2 rootwait fsck.repair=yes i915.enable_guc=3 i915.force_probe=4905 ro
[    0.026206] Kernel command line: BOOT_IMAGE=/boot/drm_intel root=/dev/nvme0n1p2 rootwait fsck.repair=yes i915.enable_guc=3 i915.force_probe=4905 ro
[    2.582581] i915 0000:03:00.0: [drm] VT-d active for gfx access
[    2.582586] fb0: switching to i915 from EFI VGA
[    2.582684] i915 0000:03:00.0: vgaarb: deactivate vga console
[    2.582707] i915 0000:03:00.0: [drm] Local memory IO size: 0x00000000fb800000
[    2.582708] i915 0000:03:00.0: [drm] Local memory available: 0x00000000fb800000
[    2.597310] i915 0000:03:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    2.600584] i915 0000:03:00.0: [drm] Finished loading DMC firmware i915/dg1_dmc_ver2_02.bin (v2.2)
[    2.667756] i915 0000:03:00.0: [drm] GuC firmware i915/dg1_guc_70.1.1.bin version 70.1
[    2.667759] i915 0000:03:00.0: [drm] HuC firmware i915/dg1_huc_7.9.3.bin version 7.9
[    2.673370] i915 0000:03:00.0: [drm] HuC authenticated
[    2.673789] i915 0000:03:00.0: [drm] GuC submission enabled
[    2.673790] i915 0000:03:00.0: [drm] GuC SLPC enabled
[    2.674046] i915 0000:03:00.0: [drm] GuC RC: enabled
```

### Do you want to contribute a patch to fix the issue?

No.

eero-t, 2022-05-27:
I've also tested Sysman functionality and simple OpenCL programs.  Those work fine, so in general drm-tip kernel seems to work fine.

Btw. media-driver README still states following:
> Media-driver requires special i915 kernel mode driver (KMD) version to support the following new platforms since upstream version of i915 KMD does not fully support them (pending patches upstream):
>
>    DG1/SG1
 >   Alchemist(DG2)/ATSM
>
> By default, media-driver builds against upstream i915 KMD and will miss support for the platforms listed above. To enable new platforms which require special i915 KMD and specify ENABLE_PRODUCTION_KMD=ON (default: OFF) build configuration option.

Although AFAIK that has not been true for DG1 for over a half a year, since this media-driver commit: https://github.com/intel/media-driver/commit/db5a8706bf9214ef12542eaab1896dacfdedfceb

And DG2 / ATS-M support being already in public kernel (for some of their variants, and requiring force-probing for now).

Xiaogangli-intel, 2022-05-31:
Hi @eero-t, I noticed i915.enable_guc=3 in your kernel parameters, could you try i915.enable_guc=2? Seems GuC submission doesn't work on DG1.

Xiaogangli-intel, 2022-05-31:
Hi @eero-t , media still have some issues on drm-tip KMD.
Could you please try this KMD at https://github.com/intel-gpu/intel-gpu-i915-backports, and need to build media driver with  ENABLE_PRODUCTION_KMD=ON, also i915.enable_guc=2 in kernel boot parameters.

eero-t, 2022-05-31:
Good catch.  I'll test upstream kernel without the GuC option tomorrow, and report back.

(I've intended to clean that out, but had forgotten to do it for all kernel configs on all machines.)

eero-t, 2022-06-01:
When GuC scheduling is explicitly disabled, there's also a GPU hang:
```
[   65.151632] i915 0000:03:00.0: [drm] Resetting vcs1 for preemption time out
[   65.151688] i915 0000:03:00.0: [drm] sample_multi_tr[5533] context reset due to GPU hang
[   65.160360] i915 0000:03:00.0: [drm] GPU HANG: ecode 12:4:28fffffd, in sample_multi_tr [5533]
```

See: [gpu-hang.txt](https://github.com/intel/media-driver/files/8814373/gpu-hang.txt)

> media still have some issues on drm-tip KMD for DG1.

Could you give pointer to more info?

> Could you please try this KMD at https://github.com/intel-gpu/intel-gpu-i915-backports, and need to build media driver with ENABLE_PRODUCTION_KMD=ON, also i915.enable_guc=2 in kernel boot parameters.

Sorry, but I'm not interested about public media-driver on backport kernel, only with what's going to upstream.

Xiaogangli-intel, 2022-06-09:
Hi @eero-t, I noticed the hang issue of HEVC decode. If you really mind using backport kernel,  maybe we have to sync with KMD to check the progress of DG1 patches upstreaming.

eero-t, 2022-06-10:
DG1 has been enabled in upstream kernel (not just drm-tip) for a long time: https://github.com/torvalds/linux/blob/master/include/drm/i915_pciids.h#L630

But kernel docs RFC section still mentions several items: https://www.kernel.org/doc/html/latest/gpu/rfc/index.html

I've asked whether they've landed already upstream (in Linus' tree i.e. should docs have been moved out of RFC section), not just in public drm-tip that I was testing (and with which I was seeing the issues).

eero-t, 2022-06-14:
According to kernel side, status specified in RFC docs applies both to public upstream and drm-tip.  I.e. there are still significant gaps in kernel i915 dGPU support, although GuC scheduling has already been enabled by default.

PS. I just tested latest media driver stack releases, and e.g. FFmpeg still gives 2 FPS with drm-tip (instead of the expected hundreds of FPS).  I haven't updated the kernel side though (will probably do that late summer, when 5.19 nears release).

eero-t, 2022-07-22:
I tested yesterday drm-tip 5.19-rc6 (and few days earlier 5.19-rc6) on DG1, and things have gone downhill.  Instead of 2 FPS HEVC decode, there are lots of failures now with FFmpeg / VA-API:
```
Input #0, hevc, from '/media/GTAV_1920x1080_60_yuv420p.h265':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: hevc (Main), yuv420p(tv), 1920x1080, 60 fps, 60 tbr, 1200k tbn
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> h264 (h264_vaapi))
Press [q] to stop, [?] for help
[hevc @ 0x560f76ef5700] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x560f76ef5700] hardware accelerator failed to decode picture
[hevc @ 0x560f76fa7080] Could not find ref with POC 0
[hevc @ 0x560f76fa7080] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x560f76fa7080] hardware accelerator failed to decode picture
[hevc @ 0x560f76fb8840] Could not find ref with POC 1
[hevc @ 0x560f76fb8840] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x560f76fb8840] hardware accelerator failed to decode picture
[hevc @ 0x560f76fca040] Could not find ref with POC 6
[hevc @ 0x560f76fca040] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x560f76fca040] hardware accelerator failed to decode picture
[hevc @ 0x560f76fdb840] Could not find ref with POC 4
[hevc @ 0x560f76fdb840] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x560f76fdb840] hardware accelerator failed to decode picture
```
(No errors in dmesgs though.)

When using FFmpeg with QSV instead of VA-API, it fails immediately:
```
Input #0, hevc, from '/media/GTAV_1920x1080_60_yuv420p.h265':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: hevc (Main), yuv420p(tv), 1920x1080, 60 fps, 60 tbr, 1200k tbn
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> h264 (h264_qsv))
Press [q] to stop, [?] for help
Output #0, null, to 'pipe:':
  Metadata:
    encoder         : Lavf59.16.100
  Stream #0:0: Video: h264, nv12(tv, progressive), 1920x1080, q=2-31, 1000 kb/s, 60 fps, 60 tbn
    Metadata:
      encoder         : Lavc59.18.100 h264_qsv
    Side data:
      cpb: bitrate max/min/avg: 0/0/1000000 buffer size: 0 vbv_delay: N/A
[h264_qsv @ 0x561d6d5488c0] Unknown FrameType, set pict_type to AV_PICTURE_TYPE_NONE.
[h264_qsv @ 0x561d6d5488c0] Error during encoding: unknown error (-21)
Video encoding failed
Conversion failed!
```

However, exactly the same drm-tip kernel, user-space [1] and test-case still work fine on TGL (with perf in hundreds of FPS).

[1] User-space components:
- libva 2.15.0
- intel-gmmlib-22.1.6
- intel-media-22.5.0
- intel-mediasdk-22.5.0
- FFmpeg n5.0.1

TGL dmesg content:
```
$ dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/drm_intel root=/dev/nvme0n1p2 rootwait fsck.repair=yes i915.enable_guc=2 ro
[    0.037729] Kernel command line: BOOT_IMAGE=/boot/drm_intel root=/dev/nvme0n1p2 rootwait fsck.repair=yes i915.enable_guc=2 ro
[    2.621257] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    2.621363] i915 0000:00:02.0: vgaarb: deactivate vga console
[    2.621412] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    2.623817] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    2.625315] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/tgl_dmc_ver2_12.bin (v2.12)
[    3.298064] i915 0000:00:02.0: [drm] failed to retrieve link info, disabling eDP
[    3.414350] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin version 70.1
[    3.414352] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9
[    3.427536] i915 0000:00:02.0: [drm] HuC authenticated
[... 19 lines cut ...]
[    2.258663] i915 0000:03:00.0: [drm] GuC firmware i915/dg1_guc_70.1.1.bin version 70.1
[    2.258666] i915 0000:03:00.0: [drm] HuC firmware i915/dg1_huc_7.9.3.bin version 7.9
[    2.263872] i915 0000:03:00.0: [drm] HuC authenticated
[    2.264158] i915 0000:03:00.0: [drm] GuC submission enabled
[    2.264160] i915 0000:03:00.0: [drm] GuC SLPC enabled
[    2.264414] i915 0000:03:00.0: [drm] GuC RC: enabled
[    2.301517] [drm] Initialized i915 1.6.0 20201103 for 0000:03:00.0 on minor 0
[    2.302089] i915 0000:0a:00.0: [drm] VT-d active for gfx access
[    2.302121] i915 0000:0a:00.0: [drm] Local memory IO size: 0x00000000fb800000
[    2.302122] i915 0000:0a:00.0: [drm] Local memory available: 0x00000000fb800000
[    2.321256] i915 0000:0a:00.0: [drm] Finished loading DMC firmware i915/dg1_dmc_ver2_02.bin (v2.2)
[    2.327096] fbcon: i915drmfb (fb0) is primary device
[    2.373506] i915 0000:03:00.0: [drm] fb0: i915drmfb frame buffer device
[    2.391770] i915 0000:0a:00.0: [drm] GuC firmware i915/dg1_guc_70.1.1.bin version 70.1
[    2.391772] i915 0000:0a:00.0: [drm] HuC firmware i915/dg1_huc_7.9.3.bin version 7.9
[    2.398253] i915 0000:0a:00.0: [drm] HuC authenticated
[    2.398673] i915 0000:0a:00.0: [drm] GuC submission enabled
[    2.398674] i915 0000:0a:00.0: [drm] GuC SLPC enabled
[    2.398985] i915 0000:0a:00.0: [drm] GuC RC: enabled
[    2.411403] [drm] Initialized i915 1.6.0 20201103 for 0000:0a:00.0 on minor 1
[    2.414748] i915 0000:0a:00.0: [drm] Cannot find any crtc or sizes
[    2.415142] i915 0000:0a:00.0: [drm] Cannot find any crtc or sizes
```

I.e. the main differences are there being 2x DG1 devices, with GuC scheduling being enabled (by default), and THP being enabled only on TGL for some reason, although both have VT-d active.

eero-t, 2022-09-13:
Tested media stack components build from latest release tags (Ubuntu 22.04 based container):
* libVA: 2.15.0
* GMMlib: intel-gmmlib-22.1.8
* media-driver: intel-media-22.5.3
* MediaSDK: intel-mediasdk-22.5.3
* oneVPL: v2022.2.2
* oneVPL GPU: intel-onevpl-22.5.3
* FFmpeg: n5.1.1

And both the FFmpeg VA-API and OneVPL decoding failures are still there, both with slightly older drm-tip v6.0-rc3 kernel, and v6.0-rc5 from yesterday.

OneVPL / MFX error message has changed to match what FFmpeg / VA-API was reporting:
```
Loaded Library configuration: 
    Version: 2.7 
    ImplName: mfx-gen 
    Adapter number : 0 
    Adapter type: integrated
    DRMRenderNodeNum: 128 
Used implementation number: 0 
Loaded modules:
   0: /usr/local/lib/libmfxhw64.so.1.35 
   1: /usr/local/lib/libmfx-gen.so.1.2.7 

Pipeline surfaces number (DecPool): 10
Input  video: HEVC
Output video: AVC 

Session 0 was NOT joined with other sessions

Transcoding started

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), Transcode, Decode<One|Last>Frame failed at /home/nobody/source/oneVPL/tools/legacy/sample_multi_transcode/src/pipeline_transcode.cpp:1933

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), Run, CTranscodingPipeline::Run::Transcode() [0x556090093fa0] failed at /home/nobody/source/oneVPL/tools/legacy/sample_multi_transcode/src/pipeline_transcode.cpp:4868

 session 0 [0x556090093fa0] failed with status MFX_ERR_DEVICE_FAILED shutting down the application...
```

eero-t, 2022-09-13:
drm-tip kernel dmesg shows this on startup, but I guess this is related just to error reporting, not media:
```
[    2.477122] i915 0000:0a:00.0: [drm] *ERROR* Zero GuC log crash dump size!
[    2.477124] i915 0000:0a:00.0: [drm] *ERROR* Zero GuC log debug size!
[    2.478087] i915 0000:0a:00.0: [drm] GuC error state capture buffer maybe too small: 2097152 < 2360316 (min = 786772)
[    2.482243] i915 0000:0a:00.0: [drm] GuC firmware i915/dg1_guc_70.1.1.bin version 70.1.1
```

eero-t, 2022-09-21:
With latest "drm-tip" (6.0-rc5) from 2 days ago, latest media-driver release:
* libVA: 2.15.0
* GMMlib: intel-gmmlib-22.2.0
* media-driver: intel-media-22.5.3

Completely refuses to work on DG1, with strace showing following:
```
...
ioctl(3, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7fff30795080) = 0
ioctl(3, DRM_IOCTL_I915_QUERY, 0x7fff30795070) = 0
ioctl(3, DRM_IOCTL_I915_QUERY, 0x7fff30795070) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7fff30795020) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7fff30795020) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7fff30795020) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7fff30795020) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7fff30795020) = -1 ENOEXEC (Exec format error)
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7fff30795020) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7fff30795020) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7fff30795020) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7fff30795020) = 0
ioctl(3, DRM_IOCTL_I915_QUERY, 0x7fff30795000) = 0
ioctl(3, DRM_IOCTL_I915_QUERY, 0x7fff30795000) = 0
write(2, "libva error: /usr/local/lib/dri/"..., 61libva error: /usr/local/lib/dri/iHD_drv_video.so init failed
) = 61
write(2, "libva info: va_openDriver() retu"..., 38libva info: va_openDriver() returns 1
) = 38
write(2, "vaInitialize failed with error c"..., 62vaInitialize failed with error code 1 (operation failed),exit
) = 62
```
=> "Exec format error" error looks suspicious

Media driver config is following:
```
$ cmake -LH -Wno-dev -G Ninja -DCMAKE_INSTALL_PREFIX=${INSTALL_DIR} -DGEN8=OFF
...
// Build and Install cmrtlib together with media driver
BUILD_CMRTLIB:BOOL=ON

// Choose the type of build, options are: None Debug Release RelWithDebInfo MinSizeRel ...
CMAKE_BUILD_TYPE:STRING=

// Install path prefix, prepended onto install directories.
CMAKE_INSTALL_PREFIX:PATH=/usr/local

// Enabled DG2 support
DG2:BOOL=ON

// Build driver with shaders (kernels) support
ENABLE_KERNELS:BOOL=ON

// Use closed source pre-built  binary shaders (kernels)
ENABLE_NONFREE_KERNELS:BOOL=ON

// Enable Production KMD header files
ENABLE_PRODUCTION_KMD:BOOL=OFF

// Make per-Gen options disable only media-kernels (not driver code) if driver code is known to be required for a successful build
ENABLE_REQUIRED_GEN_CODE:BOOL=ON

// Path to a file.
EXT_CONF:FILEPATH=EXT_CONF-NOTFOUND

// Enable Gen10 support
GEN10:BOOL=OFF

// Enable Gen11 support
GEN11:BOOL=ON

// Enabled ICLLP support (Gen11)
GEN11_ICLLP:BOOL=ON

// Enabled JSL support (Gen11)
GEN11_JSL:BOOL=ON

// Enable Gen12 support
GEN12:BOOL=ON

// Enabled ADLN support (Gen12)
GEN12_ADLN:BOOL=ON

// Enabled ADLP support (Gen12)
GEN12_ADLP:BOOL=ON

// Enabled ADLS support (Gen12)
GEN12_ADLS:BOOL=ON

// Enabled DG1 support (Gen12)
GEN12_DG1:BOOL=ON

// Enabled RKL support (Gen12)
GEN12_RKL:BOOL=ON

// Enabled TGLLP support (Gen12)
GEN12_TGLLP:BOOL=ON

// Enable Gen8 support
GEN8:BOOL=OFF

// Enable Gen9 support
GEN9:BOOL=ON

// Enabled BXT support (Gen9)
GEN9_BXT:BOOL=ON

// Enabled CFL support (Gen9)
GEN9_CFL:BOOL=ON

// Enabled CML support (Gen9)
GEN9_CML:BOOL=ON

// Enabled CMPV support (Gen9)
GEN9_CMPV:BOOL=ON

// Enabled GLK support (Gen9)
GEN9_GLK:BOOL=ON

// Enabled KBL support (Gen9)
GEN9_KBL:BOOL=ON

// Enabled SKL support (Gen9)
GEN9_SKL:BOOL=ON

// Install driver system configuration file
INSTALL_DRIVER_SYSCONF:BOOL=OFF

// Turn compiler warnings into fatal errors
MEDIA_BUILD_FATAL_WARNINGS:BOOL=ON

// Enable hardening (stack-protector, fortify source)
MEDIA_BUILD_HARDENING:BOOL=ON

// run google test module after install
MEDIA_RUN_TEST_SUITE:BOOL=ON

// 
MEDIA_VERSION:STRING=22.5.3

// Enable MHW Interfaces NEXT Support
MHW_INTERFACES_NEXT_SUPPORT:BOOL=ON

// Enabled Xehp_sdv support
XEHP_SDV:BOOL=ON

// Enabled XE_HPG support
XE_HPG:BOOL=ON

// Enabled support for Xehp_sdv+ platforms
Xe_M:BOOL=ON
```

nyanmisaka, 2022-09-26:
Any update on this? I've got a DG1 80EU and it fails decoding any video with VAAPI/QSV through ffmpeg cli. But everything works just fine on Windows.

eero-t, 2022-10-07:
Things still fail with latest "drm-tip" (6.0-rc7) from yesterday, with a matching FW (GuC: 70.5.1, HuC: 7.9.3), and latest media stack releases:
* libVA: 2.16.0
* GMMlib: intel-gmmlib-22.2.0
* media-driver: intel-media-22.5.4
* MediaSDK: intel-mediasdk-22.5.4
* oneVPL: v2022.2.4
* oneVPL-gpu: intel-onevpl-22.5.4
* FFmpeg: n5.1.2

Output from OneVPL tool:
```
$ sample_multi_transcode -i::h265 /media/GTAV_1920x1080_60_yuv420p.h265 -o::h264 /dev/null
...
Loaded Library configuration: 
    Version: 2.7 
    ImplName: mfx-gen 
    Adapter number : 0 
    Adapter type: integrated
    DRMRenderNodeNum: 128 
Used implementation number: 0 
Loaded modules:
   0: /usr/local/lib/libmfxhw64.so.1.35 
   1: /usr/local/lib/libmfx-gen.so.1.2.7 

Pipeline surfaces number (DecPool): 10
Input  video: HEVC
Output video: AVC 

Session 0 was NOT joined with other sessions
Transcoding started

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), Transcode, Decode<One|Last>Frame failed at /home/nobody/source/oneVPL/tools/legacy/sample_multi_transcode/src/pipeline_transcode.cpp:1944

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), Run, CTranscodingPipeline::Run::Transcode() [0x5555799dfdf0] failed at /home/nobody/source/oneVPL/tools/legacy/sample_multi_transcode/src/pipeline_transcode.cpp:4904

 session 0 [0x5555799dfdf0] failed with status MFX_ERR_DEVICE_FAILED shutting down the application...
```

Output from FFmpeg / VA-API:
```
$ ffmpeg -y -an -loglevel verbose -hwaccel vaapi -hwaccel_output_format vaapi -i /media/GTAV_1920x1080_60_yuv420p.h265 -c:v h264_vaapi -f null
...
Input #0, hevc, from '/media/GTAV_1920x1080_60_yuv420p.h265':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: hevc (Main), 1 reference frame, yuv420p(tv, left), 1920x1080, 60 fps, 60 tbr, 1200k tbn
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> h264 (h264_vaapi))
Press [q] to stop, [?] for help
[hevc @ 0x56401e78b500] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x56401e78b500] hardware accelerator failed to decode picture
[hevc @ 0x56401e82a180] Could not find ref with POC 0
[hevc @ 0x56401e82a180] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x56401e82a180] hardware accelerator failed to decode picture
[hevc @ 0x56401e7f15c0] Could not find ref with POC 1
[hevc @ 0x56401e7f15c0] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x56401e7f15c0] hardware accelerator failed to decode picture
[hevc @ 0x56401e802e40] Could not find ref with POC 6
...
```

OneVPL does not show anything in `dmesg`, but FFmpeg does show GPU hangs:
```
[10735.331980] i915 0000:0a:00.0: [drm] GPU HANG: ecode 12:4:00000000, in ffmpeg [19118]
[10735.331984] i915 0000:0a:00.0: [drm] ffmpeg[19118] context reset due to GPU hang
[10741.805987] i915 0000:0a:00.0: [drm] GPU HANG: ecode 12:4:00000000, in ffmpeg [19118]
[10741.805992] i915 0000:0a:00.0: [drm] ffmpeg[19118] context reset due to GPU hang
```

E.g. running simple OpenCL program with latest public compute stack releases does not show any problems.

XinfengZhang, 2022-10-07:
could you help to have a try with #1500

eero-t, 2022-10-07:
> > could you help to have a try with #1500
> 
> Sure, but I'd like to see it first pass at least one of the CI tests... Currently they all fail for it?

Tried it anyway.  "Disable object capture for recoverable context" commit did not help, things fail like before.

gizahNL, 2022-10-12:
Coming here after my issue report on onevpl-intel-gpu:

latest media-driver is completely unusable for me (I'm only interested in encoding). AVC and HEVC encoding both fail when using sample_encode program. 

Up till commit 60001c60a1f13e23cb49278ea757a77c8d743674 (bisected) HEVC encoding works.

gizahNL, 2022-10-12:
> could you help to have a try with #1500

This does fix encoding both AVC and HEVC for me.

eero-t, 2022-10-12:
> > could you help to have a try with #1500
> 
> This does fix encoding both AVC and HEVC for me.

@gizahNL You do not have issue with HEVC decode?

(I'm wondering whether my issue is be specific to my HEVC file, or more generic.)

gizahNL, 2022-10-12:
> > > could you help to have a try with #1500
> > 
> > 
> > This does fix encoding both AVC and HEVC for me.
> 
> @gizahNL You do not have issue with HEVC decode?
> 
> (I'm wondering whether my issue is be specific to my HEVC file, or more generic.)

No, decoding HEVC fails for me as well:

(encoded with sample_encode)
```
ffprobe ./out.h265           
ffprobe version 5.1.1-1ubuntu1 Copyright (c) 2007-2022 the FFmpeg developers
  built with gcc 12 (Ubuntu 12.2.0-1ubuntu1)
  configuration: --prefix=/usr --extra-version=1ubuntu1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --disable-sndio --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-shared
  libavutil      57. 28.100 / 57. 28.100
  libavcodec     59. 37.100 / 59. 37.100
  libavformat    59. 27.100 / 59. 27.100
  libavdevice    59.  7.100 / 59.  7.100
  libavfilter     8. 44.100 /  8. 44.100
  libswscale      6.  7.100 /  6.  7.100
  libswresample   4.  7.100 /  4.  7.100
  libpostproc    56.  6.100 / 56.  6.100
Input #0, hevc, from './out.h265':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: hevc (Main), yuv420p(tv), 1920x1080 [SAR 1:1 DAR 16:9], 30 fps, 30 tbr, 1200k tbn
```

```
sample_decode h265 -in ./out.h265 -o test.yuv
CONFIGURE LOADER: required implementation: hw 
CONFIGURE LOADER: required implementation mfxAccelerationMode: MFX_ACCEL_MODE_VIA_VAAPI 
libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_15
libva info: va_openDriver() returns 0
libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_15
libva info: va_openDriver() returns 0
Loaded Library configuration: 
    Version: 2.7 
    ImplName: mfx-gen 
    Adapter number : 0 
    Adapter type: integrated
    DRMRenderNodeNum: 128 
Used implementation number: 0 
Loaded modules:
   0: /usr/lib/x86_64-linux-gnu/libmfx-gen.so.1.2.7 
   1: /usr/lib/x86_64-linux-gnu/libmfxhw64.so.1.35 

libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_15
libva info: va_openDriver() returns 0
Decoding Sample Version 8.4.27.0

Input video	HEVC
Output format	NV12
Input:
  Resolution	1920x1088
  Crop X,Y,W,H	0,0,1920,1080
Output:
  Resolution	1920x1080
Frame rate	30.00
Memory type		system
MediaSDK impl		hw
MediaSDK version	2.7

Decoding started

[ERROR], sts=MFX_ERR_GPU_HANG(-21), RunDecoding, SyncOperation fail or timeout at ./tools/legacy/sample_decode/src/pipeline_decode.cpp:1666
Frame number:    0, fps: 0.000, fread_fps: 0.000, fwrite_fps: 0.000
[ERROR], sts=MFX_ERR_GPU_HANG(-21), RunDecoding, Unexpected error!! at ./tools/legacy/sample_decode/src/pipeline_decode.cpp:1922

[ERROR], sts=MFX_ERR_GPU_HANG(-21), main, Pipeline.RunDecoding failed at ./tools/legacy/sample_decode/src/sample_decode.cpp:861
```

eero-t, 2022-12-19:
Still an issue with drm-tip v6.0 kernel, and following media stack from end of November:
- libva:    2.16.0
- GMMlib:   intel-gmmlib-22.3.1
- Media-driver:    intel-media-22.6.3
- MediaSDK: intel-mediasdk-22.6.3
- oneVPL:   v2023.1.0
- VPL-GPU:  intel-onevpl-22.6.3
- FFmpeg:   n5.1.2

FFmpeg transcode runs at <1/100th of the expected speed, and oneVPL transcode tools fails to:
```
*** session 0 [0x561608b7d8e0] FAILED (MFX_ERR_GPU_HANG) 3.84671 sec, 7 frames, 1.820 fps
-i::h265 /media/GTAV_1920x1080_60_yuv420p.h265 -o::h264 /dev/null -async 4 
```

PS. `MFX_ERR_GPU_HANG` error is odd, because dmesg does not show any i915 GPU hang...

eero-t, 2023-01-18:
Tested drm-tip "v6.2-rc4" kernel (yesterday Git), and following media stack (latest releases as of today):
* libva: 2.17.0
* GMMlib: intel-gmmlib-22.3.3
* Media-driver: intel-media-23.1.0
* MediaSDK: intel-mediasdk-23.1.0
* oneVPL: v2023.1.1
* VPL-GPU: intel-onevpl-23.1.0
* FFmpeg: n5.1.2

As dGPU support (without force-probing) was enabled in it (already in  "v6.2-rc1"), I tested AVC & HEVC decode + encode also on DG2.  Everything worked fine there, _after_ enabling few extra things: https://gitlab.freedesktop.org/drm/intel/-/issues/7732

However, I could not get that same kernel binary to boot on the machine where DG1 is installed.

When I had earlier tested another build of the same kernel version on DG1, both AVC and HEVC decoding failed. However, that kernel did not have those extra options enabled, so I'm not sure whether that counts.

I did try slightly older drm-tip kernel on DG1 though, v6.1.0 one, with above listed media stack (and older one from 2022 May), and AVC decoding fails with that too, same as HEVC decoding:
```
[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), Transcode, Decode<One|Last>Frame failed at oneVPL/tools/legacy/sample_multi_transcode/src/pipeline_transcode.cpp:1944`
```

I'll try later to get the new kernel config (with I915_PXP & MEI_GSC options) to boot also on DG1 machine, to get a setup comparable to the working DG2 setup.

eero-t, 2023-01-19:
> I did try slightly older drm-tip kernel on DG1 though, v6.1.0 one, with above listed media stack (and older one from 2022 May), and AVC decoding fails with that too, same as HEVC decoding

_v6.1_ drm-tip kernel says in `dmesg` that it *did* load HuC.  Unlike _v6.2-rc4_ did, it does not complain about missing MEI modules, although it does not have DRM_I915_PXP and INTEL_MEI_PXP/_GSC options enabled.

@dvrogozh Is it still possible that HuC does not work properly without kernel MEI modules?

(I don't have easy way to build older drm-tip commits with new configs, as its commit IDs change constantly due to it being rebased to upstream.)

eero-t, 2023-01-20:
That authentication message is there with drm-tip v6.1 kernel:
```
# dmesg | grep HuC
[    2.349771] i915 0000:03:00.0: [drm] HuC firmware i915/dg1_huc.bin version 7.9.3
[    2.355243] i915 0000:03:00.0: [drm] HuC authenticated
[    2.386533] i915 0000:0a:00.0: [drm] HuC firmware i915/dg1_huc.bin version 7.9.3
[    2.392380] i915 0000:0a:00.0: [drm] HuC authenticated
```

So unfortunately that's not explaining the problem.   Maybe MEI modules are really needed only with v6.2...

gizahNL, 2023-01-20:
> However, I could not get that same kernel binary to boot on the machine where DG1 is installed.

I remember looking at the code that removed the force probing requirement and noticing that this was only done for DG2, not DG1

eero-t, 2023-01-20:
Good to know, thanks!

I'm still using "i915.force_probe=4905" on that CML-S machine, but maybe there's a reason why DG1 is not (yet) enabled by default in (drm-tip) v6.2-rc...

(I actually got it to boot first time, but accidentally rebooted it before being able to test it, and it did not come up with that kernel any more on further reboots, only with drm-tip v6.1, and Ubuntu OEM kernel + i915 DKMS. :-/)

nyanmisaka, 2023-02-04:
DG1 has lost the availability of all Media features in upstream kernels. Not just HEVC decoding.

DG1 Xe MAX 8086:4905
Linux 5.15, 6.0, 6.1, 6.2-rc6, drm-tip
media-driver 23.1.0 `ENABLE_PRODUCTION_KMD=OFF`

Tested with FFmpeg:

**DEC:**
`h264_vaapi`: Failed to end picture decode issue: 23 (internal decoding error)
`hevc_vaapi`: Failed to end picture decode issue: 23 (internal decoding error)
`av1_vaapi`: Segfault, Failed to end picture decode issue: 23 (internal decoding error)

**ENC:**
`h264_vaapi`: < 2 fps
`hevc_vaapi`: < 2 fps

**VPP:**
`scale_vaapi`: Failed

**COPY:**
`hwupload`: very slow

Also confirmed with [intel-gpu-i915-backports](https://github.com/intel-gpu/intel-gpu-i915-backports/tree/backport/main) + `ENABLE_PRODUCTION_KMD=ON` everything works fine.

Now that DG2 is already available in upstream.
Can someone at Intel update the progress of DG1 patches upstreaming?

Or provide a list of patches instead of just the squashed commits in backports repo?
It's really hard to tell which lines fixed the media issues on DG1.

gizahNL, 2023-02-04:
We have DG1 Encoding working on 6.0 with intel-onevpl 22.6.1 and mediadriver 22.6.0 with https://github.com/intel/media-driver/pull/1500 applied as patch. Haven't tested with later revisions yet though.

nyanmisaka, 2023-02-04:
> We have DG1 Encoding working on 6.0 with intel-onevpl 22.6.1 and mediadriver 22.6.0 with #1500 applied as patch. Haven't tested with later revisions yet though.

The commit https://github.com/intel/media-driver/commit/f8812f26a35755714d6386df969b0e99fbab56ef is already in 23.1.0.

No issue in UMD for DG1 but in upstream KMD.

gizahNL, 2023-02-04:
> > We have DG1 Encoding working on 6.0 with intel-onevpl 22.6.1 and mediadriver 22.6.0 with #1500 applied as patch. Haven't tested with later revisions yet though.
> 
> The commit [f8812f2](https://github.com/intel/media-driver/commit/f8812f26a35755714d6386df969b0e99fbab56ef) is already in 23.1.0.
> 
> No issue in UMD for DG1 but in upstream KMD.

It was definitely working for me with upstream 6.0 branch, not sure what revision. We're now on 6.0.19 but I haven't used the DG1 in a while in our testing.

nyanmisaka, 2023-02-04:
> Encoding was definitely working for me with upstream 6.0 branch, not sure what revision. We're now on 6.0.19 but I haven't used the DG1 in a while in our testing.

Even if encoding was working with some media-driver + public kernel combinations, it's not enough for our use case.
We need the complete transcoding pipeline from decoding, VPP to encoding.

eero-t, 2023-03-02:
> DG1 has lost the availability of all Media features in upstream kernels. Not just HEVC decoding.
> 
> DG1 Xe MAX 8086:4905 Linux 5.15, 6.0, 6.1, 6.2-rc6, drm-tip media-driver 23.1.0 `ENABLE_PRODUCTION_KMD=OFF`
...
> Also confirmed with [intel-gpu-i915-backports](https://github.com/intel-gpu/intel-gpu-i915-backports/tree/backport/main) + `ENABLE_PRODUCTION_KMD=ON` everything works fine.

I just tried mix of above i.e. latest media stack + `ENABLE_PRODUCTION_KMD=OFF`, with latest backport kernel DKMS modules for Ubuntu 22.04 (=jammy/arc).

Both latest `sample_multi_transcode` and _FFmpeg_ (v6.0) die to `double free or corruption (!prev)` when trying to transcode AVC video.

Strace shows libc to catch memory corruption after ioctl fails, and driver destroys the GPU context:
```
...
ioctl(3, DRM_IOCTL_I915_GEM_MMAP_OFFSET, 0x7fff3ff83970) = -1 EINVAL (Invalid argument)
ioctl(3, DRM_IOCTL_I915_GEM_MADVISE, 0x7fff3ff83aac) = 0
ioctl(3, DRM_IOCTL_I915_GEM_VM_DESTROY, 0x564d1fd65160) = 0
ioctl(3, DRM_IOCTL_I915_GEM_CONTEXT_DESTROY, 0x7fff3ff83d00) = 0
ioctl(3, DRM_IOCTL_I915_GEM_VM_DESTROY, 0x564d1fd67950) = 0
ioctl(3, DRM_IOCTL_I915_GEM_CONTEXT_DESTROY, 0x7fff3ff83dc0) = 0
writev(2, [{iov_base="double free or corruption (!prev"..., iov_len=33}, {iov_base="\n", iov_len=1}], 2double free or corruption (!prev)
) = 34
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fa453564000
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
gettid()                                = 9
getpid()                                = 9
tgkill(9, 9, SIGABRT)                   = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=9, si_uid=65534} ---
+++ killed by SIGABRT +++
```

Why media-driver cannot check and use available APIs like Intel compute driver does?

nyanmisaka, 2023-03-02:
Currently DG1 works only when pairing media-driver(PROD_KMD=ON) with Intel DKMS. Any other combinations will break the transcoding pipeline.

The release note says DG1/SG1 is supported but it seems they only verified on the internal DKMS and ignored the upstream kernel.

[image]

Hyy2001X, 2023-05-16:
> Currently DG1 works only when pairing media-driver(PROD_KMD=ON) with Intel DKMS. Any other combinations will break the transcoding pipeline.
> 
> The release note says DG1/SG1 is supported but it seems they only verified on the internal DKMS and ignored the upstream kernel.
> 
> [image]

Is there any progress in transcoding dg1 on linux？thanks

nyanmisaka, 2023-05-17:
Nothing has changed since then. DG1 is not supported in Linux mainline, so the DKMS is still required.

eero-t, 2023-05-17:
Did quick checking with the latest media driver release on top of the latest _i915 DKMS kernel driver installed from the Intel package repositories_ (using Ubuntu 22.04 Arc repo): https://dgpu-docs.intel.com/driver/client/overview.html

When driver is compiled with `ENABLE_PRODUCTION_KMD=OFF`, it does not recognized DG1, and crashes to double free before exit.

When driver is compiled with `ENABLE_PRODUCTION_KMD=ON`, transcoding succeeds fine _with the public Intel DKMS kernel driver_:
* h264 => h264
* h264 => h265
* h265 => h264
* h265 => h265

=> updated title.

PS. If one is already using Intel driver repo, it normally makes sense to install packages from there.  I just want to make sure that my own driver builds work also, so that I'm not relying on Intel package repo updates & versions (regardless of whether I'm building kernel, or user space).

eero-t, 2023-05-17:
> DG1 has lost the availability of all Media features in upstream kernels. Not just HEVC decoding.

@nyanmisaka You might want to file a separate bug about that (and refer this one), as it's a regression.  Whereas HEVC has AFAIK never worked properly on DG1 with the upstream kernel.

nyanmisaka, 2023-05-17:
We ship the self-built media-driver and VPL with our software so we can’t use the DKMS and PRODUCTION_KMD=ON.

> as it's a regression.

Correct, I read somewhere in this post and compute-runtime, it seems that DG1 did work with the drm-tip kernel when the mainline was on 5.15. But it’s too hard too bisect the drm-tip since it gets updated almost every day.

eero-t, 2023-06-01:
Tested _latest Git_ build of _Mesa 3D driver_ on DG1 & DG2 (Arc).  It aborted Weston & Xwayland with backport DKMS and worked only with upstream kernel. I did not see any Mesa config option for enabling support for backport DKMS (like the media driver option for "production KMD").

=> Media driver working just with backport DKMS ties one also to whatever Mesa version is in the same repository?

PS. On quick testing public compute-runtime built from latest git tag worked with both KMD versions on DG1.

Hyy2001X, 2023-06-26:
> Nothing has changed since then. DG1 is not supported in Linux mainline, so the DKMS is still required.

Have you successfully passed through DG1 to Windows or other systems on platforms such as PVE or ESXi? I think Windows has better support for transcoding, but I have tried several times and failed. Thanks😊

nyanmisaka, 2023-06-26:
I tried DG1 on Windows 10 host last year, where it worked fine.

Sherry-Lin, 2023-07-05:
> Tested _latest Git_ build of _Mesa 3D driver_ on DG1 & DG2 (Arc). It aborted Weston & Xwayland with backport DKMS and worked only with upstream kernel (e.g. v6.3 drm-tip). I did not see any Mesa config option for enabling support for backport DKMS (like the media driver option for "production KMD").
> 
> => Media driver working just with backport DKMS ties one also to whatever Mesa version is in the same repository?
> 
> PS. On quick testing public compute-runtime built from latest git tag worked with both KMD versions on DG1.

@eero-t are you using the Mesa from https://github.com/intel-gpu/Mesa/tags or it's upstream Mesa?

eero-t, 2023-07-05:
@Sherry-Lin I was using upstream Mesa i.e. version that distros will (eventually) include.

Jianshui, 2023-07-26:
Are you using Ubuntu 22.04?
As I understand, the DKMS OOT kernel should work with default Mesa.
You also can install Intel prebuild mesa packages, you can follow the release installation guide. It is verified on Ubuntu 22.04.
https://dgpu-docs.intel.com/driver/client/overview.html

2.1.2. Client Intel package repository configuration[](https://dgpu-docs.intel.com/driver/client/overview.html#client-intel-package-repository-configuration)
For all client scenarios you must configure your system to install client (arc) packages. To add the Ubuntu 22.04 client package repository:

wget -qO - https://repositories.intel.com/graphics/intel-graphics.key | \
  sudo gpg --dearmor --output /usr/share/keyrings/intel-graphics.gpg
echo "deb [arch=amd64,i386 signed-by=/usr/share/keyrings/intel-graphics.gpg] https://repositories.intel.com/graphics/ubuntu jammy arc" | \
  sudo tee /etc/apt/sources.list.d/intel-gpu-jammy.list
sudo apt-get update

2.1.3. Install Compute, Media, and Display runtimes[](https://dgpu-docs.intel.com/driver/client/overview.html#install-compute-media-and-display-runtimes)
This group of usermode packages should be installed for both out-of-tree and upstream driver install scenarios.

Note: Intel’s version of Mesa includes support for the out-of-tree driver. Standard Mesa can be used for the easiest case where the default Ubuntu 23.04 driver is installed.

sudo apt-get install -y \
  libegl-mesa0 libegl1-mesa libegl1-mesa-dev libgbm1 libgl1-mesa-dev libgl1-mesa-dri \
  libglapi-mesa libgles2-mesa-dev libglx-mesa0 libigdgmm12 libxatracker2 mesa-va-drivers \
  mesa-vdpau-drivers mesa-vulkan-drivers
After installing the new version of Mesa, you will need to restart your desktop either through a system reboot or by restarting your window manager.

nyanmisaka, 2023-07-26:
@Jianshui Any ETA about fixing the DG1 media-driver support in non-OOT/upstream kernel?

The OOT kernel maintainer even says they do not support DG1.
https://github.com/intel-gpu/intel-gpu-i915-backports/issues/99#issuecomment-1620558473

It is difficult to understand why DG1 is still not fully supported in upstream as a product of the same period as TGL.

Jianshui, 2023-07-26:
sorry, it's out of my scope.  I hope the OOT kernel can fulfill your requirements.

colorblank, 2024-03-30:
Really hope you guys can fix the DG1 media-driver support in non-OOT/upstream kernel. It is difficult to build and install the OOT kernel in Niche Linux distributions like Unraid OS. Thus introducing DG1 support  in upstream kernel realy means a lot.

Jexu, 2024-04-01:
Media driver still needs PROD kernel to work on DG1, and loses almost all features in upstream kernel (decode, encode, vp all not work).
@Sherry-Lin do we or kmd have plan to fully support dg1 in upstream kernel?

nyanmisaka, 2024-04-01:
> Media driver still needs PROD kernel to work on DG1, and loses almost all features in upstream kernel (decode, encode, vp all not work).
> @Sherry-Lin do we or kmd have plan to fully support dg1 in upstream kernel?

Unfortunately the maintainer of PROD kernel says DG1 is not *supported* by them.

- https://github.com/intel-gpu/intel-gpu-i915-backports/issues/99#issuecomment-1620558473
- https://github.com/intel-gpu/intel-gpu-i915-backports/issues/161#issuecomment-2024449624

Also, No one seems to want to submit fixes for DG1 media feats from PROD i915 to upstream i915. At this point, DG1 has no where to go.

Sherry-Lin, 2024-04-01:
Yes, that's why PROD i915 is the suggested kernel version for DG1 from KMD team. I'm going to close this issue. Please feel free to re-open if any concerns.

closed 2024-04-01 by Sherry-Lin

nyanmisaka, 2024-04-01:
> Yes, that's why PROD i915 is the suggested kernel version for DG1 from KMD team. I'm going to close this issue. Please feel free to re-open if any concerns.

[image]
[image]
[image]
[image]

@Sherry-Lin Didn't you see that the DG1 is **NOT being supported and tested** with i915 backported **PROD/OOT** driver? The intel devs on the kernel side recommend users to use upstream kernel.

nyanmisaka, 2024-04-01:
Judging from your comments. Intel does not plan to provide any support for DG1 in either PROD/OOT or upstream kernel.

Jexu, 2024-04-02:
FYI, just had a try today with latest drm-tip and media driver that everything seems work fine by applying this draft fix in media driver https://github.com/intel/media-driver/pull/1787. Upstream kernel doesn't support gem buffer object capture, just disable it in media driver.

This is drm-tip version I tried:
21a087f3d1d62518feb57cea8caf101cd4a81b5d (HEAD, cgit/drm-tip) drm-tip: 2024y-04m-01d-19h-35m-47s UTC integration manifest

So, please feel free to give a try to this change.

nyanmisaka, 2024-04-02:
Hi @Jexu, thanks you very much for the patch. With `EXEC_OBJECT_CAPTURE` disabled and `ENABLE_PRODUCTION_KMD=OFF`, the media driver is now *partially working* on DG1 (8086:4095). There are still some noticeable issues that can be fixed or improved.

decode:
H264 & MPEG2 & VC1 & MJPEG & AV1 decoding works
**HEVC & VP9** decoding **hangs** infinitely `(i915 0000:03:00.0: [drm] GPU HANG: ecode 12:4:28fffffd)`

vpp:
Commonly used VPP filters works (scale/csc/crop/deint/overlay/tonemap)

copy:
A little bit **slow** from **my experience**. (IIRC this was faster on older PROD kernel 5.15?)
```
// sw=>vaapi, 1080p nv12, 575fps
ffmpeg -hide_banner -init_hw_device vaapi -f lavfi -i nullsrc=s=1920x1080,format=nv12 -vf hwupload -f null -
frame= 1506 fps=574 q=-0.0 Lsize=N/A time=00:01:00.20 bitrate=N/A speed=  23x

// sw=>qsv, 1080p nv12, 332fps
ffmpeg -hide_banner -init_hw_device qsv -f lavfi -i nullsrc=s=1920x1080,format=nv12 -vf hwupload=extra_hw_frames=16 -f null -
frame=  665 fps=332 q=-0.0 size=N/A time=00:00:26.60 bitrate=N/A speed=13.3x
```

encode:
Both H264 and HEVC encoding works from my initial testing.

kernel:
```
// 6.9.0-rc2-1-drm-tip-git-g5100fcc57dc5
5100fcc57dc5d45b246a0aeb068f4f8062d29b09 drm-tip: 2024y-04m-02d-12h-18m-22s UTC integration manifest
```

dmesg i915:
```
[    7.079481] i915 0000:03:00.0: Force probing unsupported Device ID 4905, tainting kernel
[    7.079493] i915 0000:03:00.0: enabling device (0000 -> 0002)
[    7.079958] i915 0000:03:00.0: [drm] VT-d active for gfx access
[    7.102965] i915 0000:03:00.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
[    7.103016] i915 0000:03:00.0: [drm] Disabling DMC firmware and runtime PM
[    7.752794] i915 0000:03:00.0: [drm] GT0: GuC firmware i915/dg1_guc_70.bin version 70.20.0
[    7.752798] i915 0000:03:00.0: [drm] GT0: HuC firmware i915/dg1_huc.bin version 7.9.3
[    7.758383] i915 0000:03:00.0: [drm] GT0: HuC: authenticated for all workloads
[    7.758767] i915 0000:03:00.0: [drm] GT0: GUC: submission enabled
[    7.758769] i915 0000:03:00.0: [drm] GT0: GUC: SLPC enabled
[    7.758971] i915 0000:03:00.0: [drm] GT0: GUC: RC enabled
[    7.763975] [drm] Initialized i915 1.6.0 20230929 for 0000:03:00.0 on minor 0
[    7.826779] i915 0000:03:00.0: [drm] fb0: i915drmfb frame buffer device
[   31.088904] i915 0000:03:00.0: [drm] GPU HANG: ecode 12:4:28fffffd
[  389.057962] i915 0000:03:00.0: [drm] GPU HANG: ecode 12:4:28fffffd
[  436.203919] i915 0000:03:00.0: [drm] GPU HANG: ecode 12:4:28fffffd
[ 1420.091142] i915 0000:03:00.0: [drm] GPU HANG: ecode 12:4:28fffffd
```

lspci:
```
03:00.0 VGA compatible controller: Intel Corporation DG1 [Iris Xe MAX Graphics] (rev 01) (prog-if 00 [VGA controller])
	Flags: bus master, fast devsel, latency 0, IRQ 121, IOMMU group 0
	Memory at f9000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 7e00000000 (64-bit, prefetchable) [size=4G]
	Expansion ROM at fa000000 [disabled] [size=2M]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express Endpoint, IntMsgNum 0
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit+
	Capabilities: [d0] Power Management version 3
	Capabilities: [100] Latency Tolerance Reporting
	Kernel driver in use: i915
	Kernel modules: i915
```

Jexu, 2024-04-08:
About vp9 and hevc decoding hang, I had a comparasion of the batch buffer dump on PROD and upstrean kernel, unfortunately didn't find any difference.
Same batch buffer given by media driver but leading to different result on both kernel. Something of unknon issue may still exist in upstream kernel. Media driver was never fully verified in upstream kernel, but all features should be verified in prod kernel.

Luckly, draft pr https://github.com/intel/media-driver/pull/1787 could bring back most of media features. If you want, I could prepare a formal fix and get it merged into media driver. By the way, using env 'export INTEL_I915_CTX_CONTROL=1' to disable recoverable ctx could also solve it.

Jexu, 2024-04-08:
More experiment:
Remove decode instructions from batch buffer by submitting an empty batch to kmd, gpu hang still exists. It does have some issue in upstream kernel.

Jexu, 2024-04-09:
I didn't ever try it on DG1, but Xe kmd commits experimental support for DG1 and everything about xe kmd support is already in latest media driver.

nyanmisaka, 2024-07-12:
@Jexu With xe kmd mainlined, I revisited DG1 with it (linux 6.9.5 + xe.force_probe=4905). Now media works fine on DG1, including end-to-end transcoding. No more segfaults or insanely slow on certain codecs. I'd call it very usable, though it still needs time to prove its stability.

[image]

colorblank, 2024-07-23:
I tried replacing the Unraid kernel with Linux 6.9.5 and added xe.force_probe=4908 to the startup script. It can only drive the DG1 but cannot decode properly. Are there any additional steps required?

gesong2077, 2024-10-21:
> Possibilities:
> 
> * Missing firmware
> * Outdated media-driver
> * Outdated vpl-rt

Confirmed that video encode/decode works on Manjaro. HEVC decode/encode does not work on Ubuntu 24.10

[3 more replies]