intel/media-driver #1421: Bitstream is corrupted on occasion.

HiromichiMatsumura, 2022-05-31:
Hi,

We are developing the video recording device using ffmpeg and vaapi for encoding. Currently we have a trouble that the video recording data is corrupted on occasion. When recording for a week, the error is whether it happens once.

We have debugged using CodecDbgSetting.cfg in the media-driver, and we have enabled the following configurations.

  For the debug information to be input to the encoder:
    PicParams:1
    SlcParams:1
    CmdBufferMfx:1
    EncodeRawInputSurface:1

  For the debug information to be output from the encoder:
    Bitstream:1

Recording was performed using the same input image, and we compared the debug information of error case with the debug information of correct case. As the result, only the PAK_Bitstream_I_Frame.dat generated by the Bitstream specification was different. So we are considering there is a problem in the layer below the media-driver.

Is our considertion correct?
We don't know how to debug the lower layer from the media-driver.
Would you please give us some advice to continue to debug?

Here is our environments.

  media-driver: 18.3.0
  OS: Ubuntu 16.04.5 LTS (kernel 4.15.0-51-generic)
  CPU: Intel(R) Xeon(R) CPU E3-1505M v6

If my posting place is wrong, please let me know a suitable posting site. 

Thanks,

XinfengZhang, 2022-05-31:
suppose it is AVC? VME path?

HiromichiMatsumura, 2022-05-31:
Thank you for your early reply.

Yes, AVC.
We use H.264/AVC.
The error corruption occurs at I-Frame only.

Xiaogangli-intel, 2022-06-09:
Hi @HiromichiMatsumura, could you please provide the corrupted picture? and did you record the frame number?

HiromichiMatsumura, 2022-06-09:
Hi Xiaogang,

Thank you for your response!

I'm sorry that I cannot provide the picture because it is confidential.
We are trying to get the shareable picture.
I will share it after I get it.

Yes, I recorded the frame number.
The frame number of the corrupted picture was 211 out of 300 frames.
It was I-frame.

Is that the answer you were hoping for?

Thanks,

HiromichiMatsumura, 2022-06-17:
Hi Xiaogang,

Thank you for your concern.

After that it doesn't occur the error, so we cannot have got the shareble picture.
Please wait a moment.

Thanks,

HiromichiMatsumura, 2022-06-24:
Hi Xiaogang,

Sorry to keep you waiting.

We have been able to get the corrupt picture, so I attach it.
[rev17_M3_xxx_Moment.zip](https://github.com/intel/media-driver/files/8974773/rev17_M3_xxx_Moment.zip)

Block noises are seen at the bottom of the picuture.

Is there something you notice?

Thanks,

Xiaogangli-intel, 2022-09-06:
Hi @HiromichiMatsumura, sorry for leaving for a long time. Did you tried the latest media driver? Because lots of patches were merged.

HiromichiMatsumura, 2022-09-15:
Hi Xiaogangli,

Thank you for your suggestion.
But I think it doesn't make sense to try the latest media driver.

The error are very rare.
So, even if an error does not occur in the latest environment, it is not possible to judge whether there was an effect.

We want to know how to debug the lower layer from the media-driver.
Would you please show us the method?

Xiaogangli-intel, 2022-09-30:
Hi @HiromichiMatsumura, if you just want to debug driver, you can just add break points in driver in gdb/cgdb, run your commands, then you can debug into driver.
For log, you can check https://github.com/intel/media-driver/wiki/Video-Processing-Debug-Tool.

HiromichiMatsumura, 2022-10-03:
Hi Xiaogangli,

Thank you for your useful information!

But investigating the cause of this issue has now been de-prioritized.
I will try using this debugging tool, if I have time.

Thanks,

OttoYang, 2023-10-30:
Hi @HiromichiMatsumura ,  
Since there has been no response for a long time, I will close this issue.  
If you encounter any problems on the encoding, please feel free to report new issues. Thanks.

closed 2023-10-30 by OttoYang