intel/media-driver #1424: [Bug]: vpp psnr mismatch on SKL

wenqingx, 2022-06-02:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

good: intel-medai-driver:b15cab2164ff60456de9ec26c7b1f149d89165fc
bad :intel-media-driver:a5f8afdf821672aebae01649283b0ee9a914649d

### What's the usage scenario when you are seeing the problem?

Others

uartie, 2022-06-02:
@wenqingx please provide the use-case example

wenqingx, 2022-06-07:
gst-launch-1.0 -vf filesrc location=input.yuv num-buffers=10 ! rawvideoparse format=i420 width=1280 height=720 ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=NV12 ! vaapipostproc  brightness=-1.0 ! video/x-raw,format=NV12 ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=I420 ! checksumsink2 file-checksum=false qos=false frame-checksum=false plane-checksum=false dump-output=true dump-location=output.yuv eos-after=10

libva:395e502d162f
gmmlib:0f0d3bfa0c71

After intel-media-driver:a5f8afdf821672aebae01649283b0ee9a914649d2, The psnr values of some vpp cases in gst-vaapi do not match。

wenqingx, 2022-06-07:
After intel-media-driver:a5f8afdf821672aebae01649283b0ee9a914649d2,The actual value of PSNR becomes a fixed value, psnr:actual = [100.0, 100.0, 100.0, 100.0, 100.0, 100.0]，only some vpp cases in the gst-vaapi has this problem on SKL.
gstreamer : e64c6f0b93ce

Sherry-Lin, 2022-06-07:
@wenqingx could you provide a summary what kinds of cases would be impacted?
@FurongZhang https://github.com/intel/media-driver/commit/a5f8afdf821672aebae01649283b0ee9a914649d is HDR enabling patch, is it expected to have PSNR update?

FurongZhang, 2022-06-07:
@wenqingx , @uartie , is this issue caused by this patch https://github.com/intel/media-driver/commit/a5f8afdf821672aebae01649283b0ee9a914649d? 

This patch is not exposing HDR caps on SKL, it should not impact other VPP test cases.

uartie, 2022-06-07:
@FurongZhang it looks like all of our VPP test cases are impacted.  Seems the output is always the same as the input regardless of which operation we apply.  I will setup a local SKL to bisect and see if a5f8afdf821672a is the actual FBC.

FurongZhang, 2022-06-07:
@uartie , if you confirmed it is the actual FBC, is it possible for you to share with me the SKL system setup? I can debug that. Thank you so much in advance!

FurongZhang, 2022-06-07:
Please let me know if ok or not. I am waiting for you online.

uartie, 2022-06-07:
Yes, a5f8afdf821672aebae01649283b0ee9a914649d is definitely the FBC.  We don't see any regression with gst-msdk or gst-va... only seen with gst-vaapi and only SKL.

uartie, 2022-06-07:
Simple reproduce command:
```shell
gst-launch-1.0 -vf videotestsrc num-buffers=1 \
  ! video/x-raw,format=NV12,width=1920,height=1080 \
  ! filesink location=test.nv12

gst-launch-1.0 -vf filesrc location=./test.nv12 \
  ! rawvideoparse format=nv12 width=1920 height=1080 \
  ! vaapipostproc brightness=-1 \
  ! checksumsink2 dump-output=true dump-location=out.nv12

md5sum *.nv12
 367513217a764b3db78f49641412f9f8  out.nv12
 367513217a764b3db78f49641412f9f8  test.nv12
```

FurongZhang, 2022-06-07:
ok

uartie, 2022-06-07:
@FurongZhang According to `vaQueryVideoProcFilters` documentation, it should only return “supported” filters.  And on SKL, the HDR TM is being returned in the supported filters by the driver.  Thus, gst-vaapi tries to call `vaQueryVideoProcFilterCaps` for HDR TM filter because driver said it was supported.  However, since it is not actually supported, the `vaQueryVideoProcFilterCaps` fails and causes the regression.

PR #1433 (2022-06-10): Modify Query VP Filters
Fixed #1424
commit 6c8cdea4 (2022-06-10): Modify Query VP Filters

Fixed #1424
closed 2022-06-10 by intel-mediadev