intel/media-driver #1425: [Bug]: AYUV is supported by vaExportSurfaceHandle only for separate plane

MathieuGirard, 2022-06-02:
### What happened?

Doing a vaExportSurfaceHandle on a AYUV surface with VA_EXPORT_SURFACE_COMPOSED_LAYERS will return VA_STATUS_ERROR_INVALID_SURFACE.

DdiMedia_GetDrmFormatOfCompositeObject is missing AYUV:
https://github.com/intel/media-driver/blob/8097de2de632b34e512e9a2921bde02e637f7945/media_driver/linux/common/ddi/media_libva.cpp#L7043

Proposed fix (around 7056 in DdiMedia_GetDrmFormatOfCompositeObject adds AYUV after YUYV):
{code}
        case VA_FOURCC_UYVY:
            return DRM_FORMAT_UYVY;
        case VA_FOURCC_AYUV:
            return DRM_FORMAT_AYUV;
{code}

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Without support for this DRM surface export wont work with AYUV (ex: Video output, Vulkan support, etc).

### Debug Information

intel-media-driver-22.3.1
libva and libva-utils 2.14.0
intel-gmmlib-22.1.2

/dev/dri is populated with a render and card.

Jexu, 2022-06-06:
@XinfengZhang can you confirm if AYUV and YUY2 are supported by composite object

MathieuGirard, 2022-06-06:
AYUV is the missing format YUY2 is already in the switch case.

AYUV is a packed format (single plane of 32 bits data), there shouldn't be a difference between a group of 1 plane and a composite of 1 plane.

https://docs.microsoft.com/en-us/windows/win32/medfound/recommended-8-bit-yuv-formats-for-video-rendering#ayuv

commit 24c93340 (2022-06-09): [Decode] Add AYUV support when export surface handle

Fixes #1425
closed 2022-06-09 by intel-mediadev