intel/media-driver #1435: [Bug]: DRM_IOCTL_I915_GEM_APERTURE failed: invalid argument

MartinX3, 2022-06-09:
### Which component impacted?

Video Processing

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

$ GST_DEBUG=2 LIBVA_LOG=1 parole 2022-06-09_15-27-00.mk

I can't play the video because of a gstreamer backend error.

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

libva 2.14.0
libva-utils 2.14.0
gmmlib 22.1.3
media-driver 22.4.2

$ ls /dev/dri
```
by-path  card0  card1  renderD128  renderD129
```

$ vainfo
```
vainfo: VA-API version: 1.14 (libva 2.14.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.4.2 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
```

$ GST_DEBUG=2 LIBVA_LOG=1 parole 2022-06-09_15-27-00.mk
```
0:00:00.026902760 213748 0x55dd1189fc90 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:754:gst_element_factory_make_valist: no such element factory "autoimagesink"!
0:00:00.235098653 213748 0x55dd1189fc90 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.235353274 213748 0x55dd1189fc90 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
DRM_IOCTL_I915_GEM_APERTURE failed: Das Argument ist ungültig
[... 37 lines cut ...]
(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.537: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.433984680 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.434172487 213748 0x55dd12149aa0 WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc0> Could not fixate src caps

(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.537: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.434215555 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.434432365 213748 0x55dd12149aa0 WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc0> Could not fixate src caps

(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.537: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.434478017 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.434753062 213748 0x55dd12149aa0 WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc0> Could not fixate src caps

(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.538: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.434797572 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.435021728 213748 0x55dd12149aa0 WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc0> Could not fixate src caps

(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.538: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.435069971 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.435905909 213748 0x55dd12143cc0 WARN           matroskademux matroska-demux.c:6067:gst_matroska_demux_loop:<matroskademux0> error: Internal data stream error.
0:00:00.435925252 213748 0x55dd12143cc0 WARN           matroskademux matroska-demux.c:6067:gst_matroska_demux_loop:<matroskademux0> error: streaming stopped, reason not-negotiated (-4)
```

### Do you want to contribute a patch to fix the issue?

No.

Xiaogangli-intel, 2022-06-10:
Hi @MartinX3, I noticed you have 2 devices, render128 and render129.
could you please provide following info?
```
sudo cat /sys/kernel/debug/dri/128/name
sudo cat /sys/kernel/debug/dri/129/name
```

MartinX3, 2022-06-10:
```bash
$ sudo cat /sys/kernel/debug/dri/128/name
i915 dev=0000:00:02.0 master=pci:0000:00:02.0 unique=0000:00:02.0
```

```bash
sudo cat /sys/kernel/debug/dri/129/name
nvidia-drm dev=0000:02:00.0 unique=0000:02:00.0
```

Jexu, 2022-06-10:
Please check your env setting and void runing iHD driver on nvidia card.

Get more information from previous reporting https://github.com/intel/media-driver/issues/1359 which running iHD driver on AMD card.

MartinX3, 2022-06-10:
I use my intel iGPU for everything except games.
For Games I use my GTX940MX without any encoding hardware.
The GTX940MX is castrated because the hardware vendor wants me to use the iGPU for decoding and encoding.

XinfengZhang, 2022-06-10:
these failure message should happened in initialization stage, vainfo also call these step, but looks it success

MartinX3, 2022-06-10:
Yes, it seems that only h.264/h.265 videos are broken if I see it correctly

Xiaogangli-intel, 2022-06-13:
I wonder if parole chose a wrong device. By the result of vainfo/handbrake/VLC, all works. @MartinX3, do you know how to let parole choose /dev/dri/renderD128?

MartinX3, 2022-06-14:
Sadly I don't see a option to choose the device, but if the application throws `i915` in the error message I assume it already uses the iGPU.
[log.log.zip](https://github.com/intel/media-driver/files/8898521/log.log.zip)

eero-t, 2022-08-04:
> Sadly I don't see a option to choose the device, but if the application throws `i915` in the error message I assume it already uses the iGPU. [log.log.zip](https://github.com/intel/media-driver/files/8898521/log.log.zip)

@MartinX3 There's no mention of "i915" in your attached 60MB log file?

Lines with ERROR on them:
```
$ grep -e ERROR  log.log 
0:01:07.477961298 726649 0x559647906400 INFO        GST_ERROR_SYSTEM gstelement.c:2271:gst_element_message_full_with_details:<matroskademux1> posting message: Internal data stream error.
0:01:07.478092020 726649 0x559647906400 DEBUG                    bin gstbin.c:3695:gst_bin_handle_message_func:<decodebin1> got ERROR message, unlocking state change
0:01:07.478157518 726649 0x559647906400 DEBUG                    bin gstbin.c:3695:gst_bin_handle_message_func:<uridecodebin1> got ERROR message, unlocking state change
0:01:07.478261705 726649 0x559647906400 DEBUG                    bin gstbin.c:3695:gst_bin_handle_message_func:<player> got ERROR message, unlocking state change
0:01:07.478479821 726649 0x559647906400 INFO        GST_ERROR_SYSTEM gstelement.c:2298:gst_element_message_full_with_details:<matroskademux1> posted error message: Internal data stream error.
0:01:07.477961298 726649 0x559647906400 INFO        GST_ERROR_SYSTEM gstelement.c:2271:gst_element_message_full_with_details:<matroskademux1> posting message: Internal data stream error.
0:01:07.478092020 726649 0x559647906400 DEBUG                    bin gstbin.c:3695:gst_bin_handle_message_func:<decodebin1> got ERROR message, unlocking state change
0:01:07.478157518 726649 0x559647906400 DEBUG                    bin gstbin.c:3695:gst_bin_handle_message_func:<uridecodebin1> got ERROR message, unlocking state change
0:01:07.478261705 726649 0x559647906400 DEBUG                    bin gstbin.c:3695:gst_bin_handle_message_func:<player> got ERROR message, unlocking state change
0:01:07.478479821 726649 0x559647906400 INFO        GST_ERROR_SYSTEM gstelement.c:2298:gst_element_message_full_with_details:<matroskademux1> posted error message: Internal data stream error.
```

Warnings, *in order*, with timestamps and identical successive messages removed:
```
$ grep WARN log.log | cut -d' ' -f4- | uniq
WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
WARN               structure gststructure.c:2091:priv_gst_structure_append_to_gstring: No value transform to serialize field 'context' of type 'GstContext'
WARN      matroskareadcommon matroska-read-common.c:713:gst_matroska_read_common_parse_skip:<matroskademux1:sink> Unknown CueTrackPositions subelement 0xf0 - ignoring
WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc1> Could not fixate src caps
WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc1> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, views=(int)2, framerate=(fraction)30/1 in anything we support
WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc1> Could not fixate src caps
WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc1> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, views=(int)2, framerate=(fraction)30/1 in anything we support
WARN                GST_PADS gstpad.c:4351:gst_pad_peer_query:<capsfilter3:src> could not send sticky events
WARN               structure gststructure.c:2091:priv_gst_structure_append_to_gstring: No value transform to serialize field 'params' of type 'GstAllocationParams'
WARN               structure gststructure.c:2091:priv_gst_structure_append_to_gstring: No value transform to serialize field 'meta' of type 'GArray'
WARN               structure gststructure.c:2091:priv_gst_structure_append_to_gstring: No value transform to serialize field 'allocator' of type 'GArray'
WARN               structure gststructure.c:2091:priv_gst_structure_append_to_gstring: No value transform to serialize field 'pool' of type 'GArray'
WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc1> Could not fixate src caps
WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc1> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, views=(int)2, framerate=(fraction)30/1 in anything we support
[... cut ...] f:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, views=(int)2, framerate=(fraction)30/1 in anything we support
WARN           matroskademux matroska-demux.c:6067:gst_matroska_demux_loop:<matroskademux1> error: Internal data stream error.
WARN           matroskademux matroska-demux.c:6067:gst_matroska_demux_loop:<matroskademux1> error: streaming stopped, reason not-negotiated (-4)
...
```

MartinX3, 2023-05-04:
@LhGu I get a short spike in `intel_gpu_top` until I see the error message.

Setting up a compile environment and compiling C code is too hard for me.

LhGu, 2023-05-05:
Hi, @MartinX3 , Just use intel_gpu_top to observe the workload.  You must need to give our more debug detail info, because we could not reproduce it.  Just download media driver code and Use this command **cmake -DCMAKE_INSTALL_PREFIX=/usr  -DBUILD_TYPE=release-internal ../media-driver** to compile it.   Then, you need to replace /etc/igfx_user_feature_next.txt. https://github.com/intel/media-driver/wiki/Video-Processing-Debug-Tool

MartinX3, 2023-05-07:
Compiling in the media-driver folder results in
```
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:73 (execute_process):
  execute_process attempted to output into a file: into a source directory.
Call Stack (most recent call first):
  media_driver/media_top_cmake.cmake:23 (find_package)
  media_driver/CMakeLists.txt:95 (include)
```

LhGu, 2023-05-08:
Hi， pls read [README.md](https://github.com/intel/media-driver#readme)   just need to use cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_TYPE=release-internal ../media-driver cmd to replace original cmake cmd
[image]

LhGu, 2023-05-08:
@MartinX3  you can also find all things in https://github.com/intel/media-driver/wiki   our wiki. You need to compile our release-internal driver to get debug info.