intel/media-driver #1443: What kind of buffers are resRefMbCodeBuffer and resRefMvDataBuffer in intel-media-driver?

YukiOtaMeta, 2022-06-17:
Hello.

I would like to know the role of the following buffer in the intel-media-driver code.

MOS_RESOURCE resRefMbCodeBuffer;
MOS_RESOURCE resRefMvDataBuffer;
git：https://github.com/intel/media-driver/tree/intel-media-18.3.0
header：codec_def_common.h

I assume these are used for the hardware encoding process.
I also assume that the memory area is in the GPU (QSV).

■Question.
・What kind of results are contained in "resRefMbCodeBuffer","resRefMvDataBuffer"?
・Is the process of writing to these buffers done by the GPU (QSV)?
・What are the effects of memory rewrite on these buffers?
   (Will the macroblock be affected?)

Thank you in advance for your cooperation.

YukiOtaMeta, 2022-06-23:
What is the situation?

YukiOtaMeta, 2022-07-06:
Thank you for your response.

If this buffer returns with a different value than normal, should I assume that there was an encoding error? (Without doing anything)

OttoYang, 2023-07-24:
@YukiOtaMeta 
Do you have some other questions of help-needed topics related to this?
Or we can close current ticket?

OttoYang, 2023-07-26:
I'll close this ticket and if you have any questions about the HW encoding, feel free to reopen or submit a new one. Thanks.

closed 2023-07-26 by OttoYang