intel/media-driver #1447: [Bug]: Missing VAProfileVP9Profile0 : VAEntrypointEncSlice for Kaby Lake when using iHD driver

tulipkun, 2022-06-27:
### Which component impacted?

Encode

### What happened?

1. I ran `vainfo` using **iHD** driver
2. I ran `vainfo` using **i965** driver
3. I compared the both of them and **iHD** doesn't have `VAProfileVP9Profile0 : VAEntrypointEncSlice` but **i965** has it.
4. Video conference and Video calls doesn't use video acceleration when my camera is turned on

### What's the usage scenario when you are seeing the problem?

Video Conference

### What impacted?

I noticed that using `intel_gpu_top` yields no changes in the Video when using Google Meet or Facebook. There should be a rise in percentage in Video when my webcam is turned on but there is none. I suspect that there is no encoder for VP9 in the driver for Intel HD 620 (Kaby Lake). I've read the documentation for my GPU and it should have a VP9 8-bit encoding capability.

### Debug Information

1.
intel-media-driver version : 22.3.1
libva version : 2.14.0
libva-utils version : 2.14.0

2.
`ls /dev/dri
by-path  card0  card1  renderD128  renderD129
`

3.
[vainfo.log](https://github.com/intel/media-driver/files/8992749/vainfo.log)

4.
[libva_trace.log](https://github.com/intel/media-driver/files/8992766/libva_trace.log)

5
[dmesg.log](https://github.com/intel/media-driver/files/8992770/dmesg.log)

Xiaogangli-intel, 2022-06-28:
Hi @tulipkun, sorry, iHD driver didn't support VP9 encoding on Kaby Lake, please refer [Feature tabel](https://github.com/intel/media-driver/blob/master/README.md).
[image]

uartie, 2022-06-28:
see #630 and #771