intel/media-driver #1449: [Bug]: 22.4.3 introduces a segfault in CodechalDecode::Allocate

dwlsalmeida, 2022-06-28:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Opening /dev/dri/renderD128 in read-only mode introduces a segfault in CodechalDecode::Allocate as of 22.4.3. This used to work fine in 22.4.2. 

I have managed to pinpoint the issue to this call in `mos_gem_bo_map`:

GDB backtrace:

```
#0  CodechalDecode::Allocate (this=0x5555555daa10, codecHalSettings=0x5555555b1420) at /home/dwls/dev/media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:469
#1  0x00007ffff5374249 in DdiMediaDecode::CreateCodecHal (this=0x5555555b11b0, mediaCtx=0x555555579e00, ptr=0x7fffffff67b0, standardInfo=0x7fffffff66f4) at /home/dwls/dev/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:1201
#2  0x00007ffff538bc65 in DdiDecodeAVC::CodecHalInit (this=0x5555555b11b0, mediaCtx=0x555555579e00, ptr=0x7fffffff67b0) at /home/dwls/dev/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_avc.cpp:710
#3  0x00007ffff53808bc in DdiDecode_CreateContext (ctx=0x555555559590, configId=0, pictureWidth=320, pictureHeight=240, flag=1, renderTargets=0x7fffffffd880, numRenderTargets=6, context=0x7fffffffd83c)
    at /home/dwls/dev/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:1312
#4  0x00007ffff52eee13 in DdiMedia_CreateContext (ctx=0x555555559590, config_id=0, picture_width=320, picture_height=240, flag=1, render_targets=0x7fffffffd880, num_render_targets=6, context=0x7fffffffd83c)
    at /home/dwls/dev/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3168
#5  0x00007ffff7f6ce58 in vaCreateContext () from /usr/lib/libva.so.2
```

Code:
```
           /* and mmap it */
            bo_gem->mem_virtual = drm_mmap(0, bo->size, PROT_READ | PROT_WRITE,
                MAP_SHARED, bufmgr_gem->fd,
                mmap_arg.offset);
            if (bo_gem->mem_virtual == MAP_FAILED) {
```

Which fails with MAP_FAILED and errno == EACESS.

The null pointer eventually bubbles up here:

codechal_decoder.cpp:442

```
        uint8_t *data = (uint8_t *)m_osInterface->pfnLockResource(
            m_osInterface,
            &m_decodeStatusBuf.m_statusBuffer,
            &lockFlagsNoOverWrite);
```

As there are no checks against NULL, it is eventually dereferenced here

```
        // Set IMEM Loaded bit (in DW1) to 1 by default in the first status buffer
        // Set None Critical Error bit to 1 by default in the first status buffer
        // These bits will be changed later after storing register
        if (m_hucInterface)
        {
            m_decodeStatusBuf.m_decodeStatus->m_hucErrorStatus  = (uint64_t)m_hucInterface->GetHucStatusHevcS2lFailureMask() << 32;
```

Causing the issue `segfault at 48 ip 00007f598a337f21 sp 00007f5991aa1720 error 6 in iHD_drv_video.so[7f598a11d000+554000]`

Unfortunately I couldn't look further into this, but my educated guess is that commit 60d841 introduces the problem.

The issue can be reproduced if you patch libva-utils's mpeg2vldemo.cpp to call va_open_display_drm(), and then patch va_open_display_drm to open the drm fd as read only.

Now, I don't know whether this is expected behavior, as I am not well versed in DRM, but I was hoping for a better error message instead of a crash.

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

1. media-driver 22.4.3, libva 2.14.0,

2.  

``` 
ls /dev/dri
                                                                                                                                                                                                    
by-path  card0  renderD128
```

3. vainfo log attached [vainfo.log](https://github.com/intel/media-driver/files/9001491/vainfo.log)

4. trace log attached [libva-trace.log](https://github.com/intel/media-driver/files/9001499/libva-trace.log)

5. GPU does not hang.

### Do you want to contribute a patch to fix the issue?

No.

BobBeckett, 2022-06-28:
due to using `PROT_READ | PROT_WRITE` in the mmap call, `do_mmap()` in the kernel requires `O_RDWR`, returning `-EACCES` otherwise.

The fd used needs to be validated for write permissions

Xiaogangli-intel, 2022-06-29:
Thanks @dwlsalmeida  and @BobBeckett.
Hi @dwlsalmeida, which platform are you working on?

Jexu, 2022-06-29:
Perhaps it can reproduce on any platforms since umd switches to mmap_offset after 22.4.3 which will occur do_mmap.

The prot argument in do_mmap describes the desired memory protection of the mapping and must not conflict with the open mode of the file. In this case, it does confict while the memory could read or write but fd is read only.

Jexu, 2022-06-29:
@dwlsalmeida Please avoid opening the gpu device in read-only mode which will cause memory mapping issue.

I just add some null check to avoid crash as well. @Xiaogangli-intel  @MicroYY  You can also add same check in Encode and VPP side.

Some function of locking resource for reference:
'pfnLockResource'
'Lock'
'LockMosResource'

dwlsalmeida, 2022-06-29:
@Xiaogangli-intel Hey there! It's Manjaro :)

I feel there should be a warning against this somehow. How come you can get a VADisplay backed by a read-only DRM fd if you can't use it without crashing?

I confirm that opening as read-write solves the issue.

BobBeckett, 2022-06-29:
it seems to me that wherever the fd originally enters the driver (if provided by caller), a simple `fcntl(fd, F_GETFL)` should be enough to check for `O_RDWR`, with an appropriate error path if not present.

Jexu, 2022-06-30:
It maybe good to check the fd mode in libva before open media driver.

PR #1451 (2022-07-01): [Decode] Add the null check to avoid crash
Fixes: #1449
commit 904c612f (2022-07-01): [Decode] Add the null check to avoid crash

Fixes: #1449
closed 2022-07-01 by intel-mediadev