intel/media-driver #1453: [Bug]: MJPEG vaapi decoding broken

m911t, 2022-07-06:
### Which component impacted?

Decode

### What happened?

1. Open music file with embedded mjpeg yuv420p on mpv ( or with mpv --hwdec=auto --hwdec-codecs=all <file> "
2. notice how the image is absolutely broken https://i.imgur.com/1aW96cF.png
3. reopen the music file with " mpv --config=no <file> "
4. notice how it's completely fine.

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

Well, mjpeg images (and i assume video) would display just fine.

### Debug Information

libva 2.14.0
libva-utils 2.14.0
intel-gmmlib 22.1.3
intel-media-driver 22.3.1
> "ls /dev/dri"
> by-path  card1  renderD128
vainfo: https://pastebin.com/G8isPxi4
libva trace log: https://pastebin.com/a1QRfDM2

lenovo ideapad 5 15itl05 
i7-1165g7
intel xe graphics

Jexu, 2022-07-06:
>>>notice how the image is absolutely broken https://i.imgur.com/1aW96cF.png
1. I cannot access to the link from my network, please paste the snapshot directly.
2. Can you extract the video from the file and try with ffmpeg decode?
3. The trace log shows invalid surface handle sent to umd by mpv to export the surface, i suppose the surface is for display which causes broken.

m911t, 2022-07-06:
[image]

that glitched image is what should be the album art (which is a mjpeg video according to mpv) 

I used the first file from this page now, " sample_1280x720_surfing_with_audio.mjpeg "
https://filesamples.com/formats/mjpeg

with " ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -i /home/mt/Downloads/sample_1280x720_surfing_with_audio.mjpeg -vcodec rawvideo -acodec copy -f matroska - | ffplay -i - " it plays fine. with mpv it only plays fine without vaapi decoding. I think it might be a mpv issue?

m911t, 2022-07-06:
aha! the bug is with mpv + wayland, not mpv + vaapi 

" mpv --hwdec=auto --hwdec-codecs=all --gpu-context=x11 --config=no <file> " plays it fine

closed 2022-07-06 by m911t