intel/media-driver #1455: [Bug]: HEVC/AVC encoder appends padding bits after "end of sequence" NAL.

HeJunyan, 2022-07-09:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

The command line such as:

gst-launch-1.0 -vf videotestsrc num-buffers=100 ! video/x-raw,format=NV12,width=200,height=100 ! vah264enc rate-control=cbr key-int-max=256 target-usage=1 num-slices=1 b-frames=1 bitrate=1000 ref-frames=2 b-pyramid=0 mbbrc=2 ! filesink location=va2k.h264

can trigger this problem. The dumped coded data "va2k.h264" contains padding bits after the last EOS(end of sequence) NAL.
The EOS NAL should be the last NAL in the whole stream. There should be no padding bits after that nal.

### What's the usage scenario when you are seeing the problem?

Playback, Video Analytics, Video Conference

### Debug Information

You can use gstreamer command:

gst-launch-1.0 -vf videotestsrc num-buffers=100 ! video/x-raw,format=NV12,width=200,height=100 ! vah264enc rate-control=cbr key-int-max=256 target-usage=1 num-slices=1 b-frames=1 bitrate=1000 ref-frames=2 b-pyramid=0 mbbrc=2 ! filesink 
location=va2k.h264

OttoYang, 2023-03-28:
The CBR algorithm in driver will add padding bits to meet the target bitrate.  
User can set a lower bitrate or use VBR if this is undisired.

OttoYang, 2023-03-28:
could we close this issue?

HeJunyan, 2023-03-28:
OK

closed 2023-03-28 by HeJunyan