intel/media-driver #1457: [Bug]:  decode/mpeg2 issue on DG2/ATSM, vaEndPicture return error

fanli-hub, 2022-07-13:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

Ubuntu 20.04.4 LTS
The latest version components

Reproduce cmd:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -i sony-ct1.bits.mpeg2 -vframes 1 -f null -

Error message:
[mpeg2video @ 0x55cc3987cb40] Failed to end picture decode issue: 23 (internal decoding error).
[mpeg2video @ 0x55cc3987cb40] hardware accelerator failed to decode picture

### What's the usage scenario when you are seeing the problem?

Video Analytics

### Do you want to contribute a patch to fix the issue?

No.

Jexu, 2022-07-13:
It is normal clip or error clip of mpeg2 file? Does it happen for other mpeg2 decoding?  How about sample_decoder?

stellawuintel, 2022-07-14:
Auto Created VSMGWL-54531 for further analysis.

fanli-hub, 2022-07-14:
> It is normal clip or error clip of mpeg2 file? Does it happen for other mpeg2 decoding? How about sample_decoder?

It is an error clip. Sample_decode works correctly. And this issue only occurred on DG2/ATSM.

Jexu, 2022-07-19:
It is expected result since it is error clip decoding. Media driver get decode error from hw and return the value to app, sample_decode treats the error(mb error) as success, but ffmpeg doesn't. So both show different result.

feiwan1, 2022-07-25:
ffmpeg-vaapi pass each slice data with a vaBuffer, while sample_decode pass all slice data into one vaBuffer. Does this can cause different vaEndPicture result?

Some debug log:
[MOS]:  CRITICAL - GetResourceInfo:1762: Invalid (nullptr) Pointer.
[MHW]:  CRITICAL - AddMfxPipeBufAddrCmd:984: MOS returned error, eStatus = 0x5
[MHW]:  CRITICAL - AddMfxPipeBufAddrCmd:89: MOS returned error, eStatus = 0x5
[CODEC]:CRITICAL - AddMfxPipeBufAddrCmd:92: MOS returned error, eStatus = 0x5
[CODEC]:CRITICAL - Execute:49: MOS returned error, eStatus = 0x5
[CODEC]:CRITICAL - PackPictureLevelCmds:94: MOS returned error, eStatus = 0x5
[CODEC]:CRITICAL - Submit:64: MOS returned error, eStatus = 0x5
[CODEC]:CRITICAL - Submit:127: MOS returned error, eStatus = 0x5
[CODEC]:CRITICAL - ExecuteActivePackets:297: MOS returned error, eStatus = 0x5
[CODEC]:CRITICAL - Execute:217: MOS returned error, eStatus = 0x5
[LIBVA]:CRITICAL - EndPicture:936: DDI:DdiDecode_DecodeInCodecHal return failure.

Btw, on Legacy platform like CFL/KBL, same cmd and clip doesn't have this issue.

Jexu, 2022-07-25:
Your log shows the resource becoming nullptr before hw execution.
You can step into last two function to check if something wrong and what causes that.

Jexu, 2022-07-27:
First carsh in https://github.com/intel/media-driver/blob/master/media_softlet/agnostic/common/codec/hal/dec/mpeg2/features/decode_mpeg2_basic_feature.cpp#L187 and look like incorrect picture_structure sent by ffmpeg causes this assert.
Suggess to compare the pic/ slice params between ffmpeg and sample decode by capturing libva trace.

PR #1479 (2022-09-06): [Decode] Add null point check for reference
Fixes: #1457
commit cbccd1df (2022-09-06): [Decode] Add null point check for reference

Fixes: #1457
closed 2022-09-06 by intel-mediadev