intel/media-driver #1458: [Feature]: Open source all remaining media kernels

DemiMarie, 2022-07-13:
### What Feature?

All of the media kernels should be open source.

### What's the usage scenario would be benifited?

Content Creation

### What impacted?

Fedora currently only ships decoding support.  This would allow it to ship encoding support.

### Do you want to contribute a patch to develop this feature?

No.

eero-t, 2022-08-17:
> All of the media kernels should be open source.
> ...
> Fedora currently only ships decoding support. This would allow it to ship encoding support.

Please explain your reasoning.  I do not see how that could be a reason for Fedora not enabling encoding support with already existing open source kernels, and hardware.  Especially when (according to driver README) free kernel version already covers all codecs and video processing features...

DemiMarie, 2022-08-17:
> Please explain your reasoning. I do not see how that could be a reason for Fedora not enabling encoding support with already existing open source kernels, and hardware. Especially when (according to [...]

There are several codecs for which encoding requires nonfree media kernels.  HEVC 8/10 bit 422 for one.  Also de-noise, 3DLUT, STD/E, TCC, and HDR10 Tonal Mapping require them.

eero-t, 2022-08-18:
I was asking why them not being available would prevent Fedora from providing the available ones?

Sherry-Lin, 2022-12-02:
From readme table, you could see most of encode features could be supported in free kernel build.
Regarding to HEVC 8/10bit 422, it is not supported in Linux yet in fact even in full feature build. I will submit a patch to correct this information. Thanks for your reminder.