intel/media-driver #1459: [Question]: Macroblock Skip Threshold

BlueWaterCrystal, 2022-07-16:
Hi,

How do you increase the macroblock skip threshold to use more skip MB?

I've tried increasing the values in SkipVal_P_Common but nothing seems to change

Xiaogangli-intel, 2022-07-29:
Hi @BlueWaterCrystal, which platform are you working on? And are you using low-power profile?

BlueWaterCrystal, 2022-07-29:
> Hi @BlueWaterCrystal, which platform are you working on? And are you using low-power profile?
Hi

I'm working on:
OS: Linux Debian 11
Gstreamer:  1.18
CPU: i5-6200U Skylake

The profile is set to none.

OttoYang, 2023-07-28:
@DenWolf Hi Denis, could you please take a look at this question?

DenWolf, 2023-07-28:
Hi @BlueWaterCrystal ,
I understand that you asked this question long time ago, but I hope that maybe it's still actual for you and may be helpful for others, so lets try to resolve this topic.

In general, Skip MB thresholds in encoding controls by increasing **Quantization** ratio -> frames (and part of the frames) become more similar with each other -> this similarity (low delta diff) increase the probability of Skipping MBs

I suppose it's better to start from Driver **API usage** (or higher-level applications) to control MB skipping level and using embedded algorithms (of course with related Quality degradation):
- use higher QP values in CQP mode - up to maximum 51 (I suppose it's good to start from QP=37 and increase it based on experiments)
- use a low Bitrate values in BRC modes

If you have some information about the video stream content like which frames do you want to "skip" or about the data on a part of the frame (for example, during video conference or streaming) - you can use our special media features for such scenarios:
- **ROI** (Region Of Interests) - set the lower/higher quality on some rectangle areas - more information about that on MediaSDK manual - https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdk-man.md#mfxextencoderroi
- **MBQP** - the similar feature as ROI but here you need to fulfil QP mask for the full frame - https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdk-man.md#mfxextmbqp
- use https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdk-man.md#mfxskipmode or https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdk-man.md#mfxextmbdisableskipmap

DenWolf, 2023-07-28:
About the primary question related to **SkipVal_P_Common**:
As you can see in Driver source code - it's a 3-dimentions-array with a thresholds based on frame/slice **QP** (as I mentioned in my previous comment) and transform_8x8 - 
https://github.com/intel/media-driver/blob/55c40038bbbaf33477c31de6a0e509df16d04f87/media_driver/agnostic/gen9_skl/codec/hal/codechal_encode_avc_g9_skl.cpp#L2191

Effect of changing this matrix could be different and related to multiple other factors/metrics/thresholds in AVC encoding algorithm - except the mention QP per frame, it could be affected by Costs of prediction or partition for different modes for example, and so on so forth -> so, effect of some kind of unpredictable here.

@BlueWaterCrystal - feel free to ask any questions you have.

Best regards,
Denis

BlueWaterCrystal, 2023-08-01:
Hi, Thanks for getting back to me.

I believe we came to the same conclusion.

Is MBQP part of FEI? 
Additionally, it looks like FEI may be suitable for this solution, though last time I tested the samples I wasn't able to figure out what it needed for it to work.
 
https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdkfei-man.md

The end goal is if the frame is too big, skip more macroblocks instead of the whole frame.

DenWolf, 2023-08-03:
Hi @BlueWaterCrystal ,
Great that it still could be helpful for you!

_>> The end goal is if the frame is too big, skip more macroblocks instead of the whole frame._
OK, as I right understand, your primary purpose is somehow decrease the frame size in case of it's too big.

For this purpose I see the next solutions:
- using feature **MaxFrameSize** - https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdk-man.md#mfxextcodingoption3 - you can set in MaxFrameSizeI  and/or MaxFrameSizeP to limit frame size in bytes specifically for I-frames and P-/B-frames
- if you using VBR rate control, you can also try to use **LowDelayBRC=ON** - https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdk-man.md#mfxextcodingoption3 - this mode encode frames in "more similar" filesize  without significant size-drop or increasing
- if you are using CBR or VBR - I suppose you can try to use a different HRD buffer settings
- for experiment purpose you can also try to use **ExtBRC=ON**

Some questions from my side:
- which rate control are you using - CBR / VBR / ICQ / CQP / ... ?
- are you using encoder via MediaSDK or via VAAPI directly?
- if possible, maybe you could share the command-line which are you using and input stream as example?

Best regards,
Denis

DenWolf, 2023-08-03:
About MBQP:

It's not related to FEI, MBQP is a separate independent feature.

To use it via MediaSDK, need to set **EnableMBQP=ON** (in ExtCodingOption3 - https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdk-man.md#mfxextcodingoption3) and fulfill **mfxExtMBQP** buffer (https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdk-man.md#mfxextmbqp) with setting QP mask for the whole frame and chosing the mode.

As I mentioned in my previous comment - I suppose that we can resolve your issue using another methods.

DenWolf, 2023-08-23:
Hi @BlueWaterCrystal ,
Did you have a chance to check my latest comments? Is it helpful for you?
In case of some experiments on your side - please help to share your feedback.

Thank you in advance!

Best regards,
Denis

BlueWaterCrystal, 2023-08-24:
Hi @DenWolf,

Sorry for the late reply, thank you for getting back to me. 
I haven't tested it yet, but I'll let you know the results.

CBR, I believe we started with VAAPI but then changed to MediaSDK.
I don't currently have the commands anymore though we did test's with Gstreamer and the examples files. 

Additionally, are there commands that may be beneficial to testing the new parameters? 

Regarding MaxFrameSize, It looks like it can only be used in VBR, can still overshoot and can only be set once?

DenWolf, 2023-08-25:
Hi @BlueWaterCrystal ,
thank you very much for your feedback and provided information about your usage scenarios, it's very helpful!

It's correct - MaxFrameSize (and similar specific features) is applied only for VBR. Please also note, that some settings could "touch" HRD conformance and result Bitrate value (+ quality of the stream of course).
CBR is more strict for HRD settings and Bitrate accurancy, so such features like MaxFrameSize or LowDelayBRC are not applied here.

MaxFrameSizeI/MaxFrameSizeP set specifically for I- and P-/B-frames; it's should be set only on Init(...). If you want to changes MaxFrameSize values during encoding - need to make MediaSDK Reset(..) - in this case it will be "soft" as only BRC encoding algorithm will be reseted, no needs for reallocation resources or similar.

If you have a strong requirement for CBR - I suppose you could try to set a different HRD buffer settings (as I mentioned previously) + you can also disable **NalHrdConformance** (set it to OFF - https://github.com/Intel-Media-SDK/MediaSDK/blob/master/doc/mediasdk-man.md#mfxextcodingoption )

OttoYang, 2023-10-30:
Hi @BlueWaterCrystal , I'd like to know if you have any other questions about this?