intel/media-driver #1460: [License] No explicit grant of license found

rahulmohang, 2022-07-18:
The license header in the following files do not have explicit grant of rights to use or redistribute:

https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal_render_16alignment.h
https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal_render_16alignment.cpp

The license text is just a disclaimer:
"
* INTEL MAKES NO WARRANTY OF ANY KIND REGARDING THE CODE.  THIS CODE IS
* LICENSED ON AN "AS IS" BASIS AND INTEL WILL NOT PROVIDE ANY SUPPORT,
* ASSISTANCE, INSTALLATION, TRAINING OR OTHER SERVICES.  INTEL DOES NOT
* PROVIDE ANY UPDATES, ENHANCEMENTS OR EXTENSIONS.  INTEL SPECIFICALLY
* DISCLAIMS ANY WARRANTY OF MERCHANTABILITY, NONINFRINGEMENT, FITNESS FOR ANY
* PARTICULAR PURPOSE, OR ANY OTHER WARRANTY.  Intel disclaims all liability,
* including liability for infringement of any proprietary rights, relating to
* use of the code. No license, express or implied, by estoppel or otherwise,
* to any intellectual property rights is granted herein.
"

It would be great to have an explicit license grant.

FurongZhang, 2022-08-01:
Received the issues. Will handle that.

closed 2022-08-01 by intel-mediadev

rahulmohang, 2022-09-08:
Thank you @FurongZhang @Sherry-Lin @LhGu @intel-mediadev

PR #1465 (2022-08-01): [License]Add explicit grant of license
Fixed #1460
commit 54b02018 (2022-08-01): [License]Add explicit grant of license

Fixed #1460
Add Licenses for some files