intel/media-driver #1461: [Bug]: The driver should report VA_ENC_SLICE_STRUCTURE_ARBITRARY_ROWS for slice alignment on TGL for LP encoder of HEVC

HeJunyan, 2022-07-20:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

The current driver just reports VA_ENC_SLICE_STRUCTURE_ARBITRARY_MACROBLOCKS for VDEnc HEVC encoder on TGL and older GPUs. But in fact, it require that the start address aligned tothe start address . We already define a VA_ENC_SLICE_STRUCTURE_ARBITRARY_ROWS in va which can match this requirement quite well. So please correct the according query result .

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

The HEVC encoder will hang if the start address does not align to the begin of each row.

### Debug Information

Pipeline will hang.

gst-launch-1.0 -vf videotestsrc num-buffers=100 ! video/x-raw,format=I420,width=1920,height=1080 ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=NV12 ! vah265lpenc rate-control=cqp num-slices=3 bitrate=8000 ! video/x-h265,profile=main ! h265parse ! fakesink

Please note: We already add some hacker in our code to cover this. It's temporary, please delete the hanker and try this command.

stellawuintel, 2022-07-29:
Auto Created VSMGWL-55154 for further analysis.

Xiaogangli-intel, 2022-09-05:
The PR is ready, validation in progress

Tianhaol, 2022-11-07:
@HeJunyan  Could help to check the latest driver?

OttoYang, 2023-03-28:
@HeJunyan Since Gen11 no longer releases new drivers, so could we close this issue?

HeJunyan, 2023-03-28:
OK, because our middleware does not have code related to GEN version check, and if user find hang issue later, I will directly forward the issues to you.

closed 2023-03-28 by HeJunyan