intel/media-driver #1462: [Bug]: [encode/av1] Encode frame failed on DG2

fanli-hub, 2022-07-21:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

Ubuntu 20.04.3 LTS
The latest version components
This issue occurred on ffmpeg-vaapi/gst-msdk.

**Reproduce cmd:**
gst-launch-1.0 -vf filesrc location=720x480p.yuv num-buffers=150 ! rawvideoparse format=i420 width=720 height=480 ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=NV12 ! msdkav1enc rate-control=cqp hardware=true gop-size=30 qpi=140 qpp=140 qpb=140 target-usage=4 num-tile-cols=16 num-tile-rows=4 b-frames=0 ! video/x-av1,profile=main ! av1parse ! matroskamux ! filesink location=output.webm

Error message:
MSDK encode error (the HW acceleration device is busy)
ERROR: pipeline doesn't want to preroll.
ERROR: from element /GstPipeline:pipeline0/GstMsdkAV1Enc:msdkav1enc0: Encode frame failed.
0:00:24.198158246 78628 0x555b98979e90 ERROR                msdkenc gstmsdkenc.c:1126:gst_msdkenc_finish_frame:<msdkav1enc0> failed to get list of frame

**Sample_encode cmd:**
sample_encode av1 -i 720x480p.yuv -o msdk.obu -cqp -trows 4 -tcols 16 -w 720 -h 480 -n 1

### What's the usage scenario when you are seeing the problem?

Video Analytics

### Do you want to contribute a patch to fix the issue?

No.

Xiaogangli-intel, 2022-07-21:
Thanks @fanli-hub, which Linux Kernel version are you using? could you provide vainfo and dmesg?

fanli-hub, 2022-07-22:
Hi @Xiaogangli-intel, I attach vainfo and dmesg to this comment.
kernel: 5.10.54+prerelease292
[vainfo.txt](https://github.com/intel/media-driver/files/9163893/vainfo.txt)
[dmesg.txt](https://github.com/intel/media-driver/files/9163895/dmesg.txt)
7

Xiaogangli-intel, 2022-07-29:
Hi @fanli-hub, could you try removing i915.enable_guc=2 in kernel command line?

stellawuintel, 2022-07-29:
Auto Created VSMGWL-55155 for further analysis.

Xiaogangli-intel, 2022-08-04:
Hi @fanli-hub, did you try modify i915.enable_guc? Any update?

fanli-hub, 2022-08-05:
This issue can still be reproduced on DG2 after modifying i915.enable_guc. The value of "-trows/tcols" will cause this issue.

fanli-hub, 2022-08-12:
Any update on this bug? Thanks.

Xiaogangli-intel, 2022-08-22:
@fanli-hub, we are debugging this issue, will let you know once we have any findings. Thanks.

Xiaogangli-intel, 2022-09-05:
An draft PR is ready, validation in progress.

feiwan1, 2022-10-25:
Hi @Xiaogangli-intel, Is your PR upstreamed?

wenqingx, 2022-11-15:
@Xiaogangli-intel I manually update the driver to the latest, including https://github.com/intel/media-driver/commit/bdab5d1eb  , but the issue does not seem to have been fixed. And I double check CI test report is also failed.thanks

Xiaogangli-intel, 2022-11-15:
it still reports hang?

wenqingx, 2022-11-16:
> it still reports hang?
yes

wenqingx, 2023-03-13:
Wang Fei and Yao Leyu have discussed before. We will remove some cases with unreasonable design. This issue can be closed. Thanks

bai-isaac, 2023-03-16:
Could you please help close this issue if it works for you? Thanks! @fanli-hub

wenqingx, 2023-03-16:
> Could you please help close this issue if it works for you? Thanks! @fanli-hub
[fanli-hub](https://github.com/fanli-hub) has left, please help close it, thanks

bai-isaac, 2023-03-16:
> > Could you please help close this issue if it works for you? Thanks! @fanli-hub
> > [fanli-hub](https://github.com/fanli-hub) has left, please help close it, thanks

Got it, thanks wening! I'll close it.

closed 2023-03-16 by bai-isaac