intel/media-driver #1464: [Bug][regression]: ffmpeg-libvpl block

Bin-CI, 2022-07-29:
### Which component impacted?

Decode, Encode, Video Processing

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

almost test on ffmpeg-libvpl failed
ffmpeg-vaapi, ffmpeg-libmfx, gstreamer-vaapi, gstreamer-libvpl, gstreamer-libmfx not effect

### What's the usage scenario when you are seeing the problem?

Video Conference, Immersive Media

### What impacted?

almost test on ffmpeg-libvpl failed

### Debug Information

passed: https://github.com/intel/media-driver/commit/bffd62a2e6f83403a0fcbacad7a73a47e7000b09
Catch failed: https://github.com/intel/media-driver/commit/d40241cfcef16a86ca3f5f91efcfea94eea91513

Bin-CI, 2022-07-29:
bisect in process it failed on GEN 11+ platform

Bin-CI, 2022-07-29:
it conflict with vpl runtime, not effect msdk runtime

Bin-CI, 2022-07-29:
Cause by https://github.com/intel/media-driver/commit/b8d1bd1601d1c89b4743d1fa474f4be7ececa3fb

uartie, 2022-07-29:
Oddly, it did not appear to affect `gstreamer + vpl`.  So, curious what is different about `ffmpeg + vpl` that exposes this bug?  @xhaihao any insight?

@Bin-CI maybe you can provide a gdb backtrace using debug builds?

uartie, 2022-07-29:
> @Bin-CI maybe you can provide a gdb backtrace using debug builds?
 That is, from our test logs it appears most decode is crashing... so a backtrace could be helpful in this instance.

Xiaogangli-intel, 2022-08-22:
> Cause by [b8d1bd1](https://github.com/intel/media-driver/commit/b8d1bd1601d1c89b4743d1fa474f4be7ececa3fb)

Hi @Bin-CI this commit was reverted by [fca94a6](https://github.com/intel/media-driver/commit/fca94a6d6a631a349980a08e120cdf34ba3038b4).
Does the issue still exist in the latest driver?

Bin-CI, 2022-08-22:
it is not exist, current no time to dump the backtrace for it

Xiaogangli-intel, 2022-08-22:
Thank you @Bin-CI, the issue should be gone by the revert of [fca94a6](https://github.com/intel/media-driver/commit/fca94a6d6a631a349980a08e120cdf34ba3038b4).

Xiaogangli-intel, 2022-09-05:
Hi @Bin-CI, if no more concern about this issue, I will close it.

closed 2022-09-05 by Bin-CI