intel/media-driver #1466: [Bug]: mpv outputs garbage via vaapi  in JSL

YongZhuIntel, 2022-08-08:
### Which component impacted?

Decode, Video Processing

### What happened?

When using mpv --vo=vaapi  and resizing mpv window ,there is a garbage screen appears.  And  it has a recurrence probability of  100% with intel-media driver  22.4.0，However, after being replaced by intel-media driver  20.1.1 the recurrence probability  can be reduced to 20%. 

mpv version: mpv 0.32.0
Linux Distribution and Version: Ubuntu 20.04
Window Manager and version: X11/ Kwin
CPU: Intel®  Pentium® Sliver [N6000@1.10GHz](mailto:N6000@1.10GHz)
GPU driver and version: VA-API version: 1.15.0 (libva 2.7.0)/ iHD driver: 22.4.0

**Sample screenshot**
[image]

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

libva info: VA-API version 1.15.0
libva info: Trying to open /opt/media/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_15
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.15 (libva 2.7.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.4.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP

YongZhuIntel, 2022-08-25:
the Xorg and mpv drm_ioctl are intertwined when garbage appear：
 [71673.633684] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, I915_GEM_CREATE
[71673.633723] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, I915_GEM_SET_DOMAIN
[71673.636638] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, I915_GEM_SET_TILING
[71673.636944] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, DRM_IOCTL_GEM_FLINK
[71673.636968] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, DRM_IOCTL_GEM_FLINK
[71673.637116] [drm:drm_ioctl] comm="mpv/vo" pid=35650, dev=0xe200, auth=1, DRM_IOCTL_GEM_OPEN
[71673.637135] [drm:drm_ioctl] comm="mpv/vo" pid=35650, dev=0xe200, auth=1, I915_GEM_GET_TILING
[71673.637412] [drm:drm_ioctl] comm="mpv/vo" pid=35650, dev=0xe200, auth=1, I915_GEM_SET_DOMAIN
[71673.637441] [drm:drm_ioctl] comm="mpv/vo" pid=35650, dev=0xe200, auth=1, I915_GEM_WAIT
[71673.637543] [drm:drm_ioctl] comm="mpv/vo" pid=35650, dev=0xe200, auth=1, I915_GEM_EXECBUFFER2_WR
[71673.637821] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, I915_GEM_EXECBUFFER2_WR
[71673.637833] [drm:drm_ioctl] comm="mpv/vo" pid=35650, dev=0xe200, auth=1, DRM_IOCTL_GEM_CLOSE
[71673.638046] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, DRM_IOCTL_SYNCOBJ_DESTROY
[71673.638070] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, I915_GEM_MADVISE
[71673.638087] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, I915_GEM_BUSY
[71673.638096] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, I915_GEM_MADVISE
[71673.638107] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, DRM_IOCTL_SYNCOBJ_CREATE
[71673.638321] [drm:drm_ioctl] comm="Xorg" pid=1869, dev=0xe200, auth=1, DRM_IOCTL_CRTC_GET_SEQUENCE

the root cause maybe in DdiCodec_PutSurfaceLinuxHW,if add usleep after   dri_vtable->get_rendering_buffer, the garbage screen disappears

    dri_drawable = dri_vtable->get_drawable(ctx, (Drawable)draw);
    DDI_CHK_NULL(dri_drawable, "Null dri_drawable", VA_STATUS_ERROR_INVALID_PARAMETER);
    buffer = dri_vtable->get_rendering_buffer(ctx, dri_drawable);
    DDI_CHK_NULL(buffer, "Null buffer", VA_STATUS_ERROR_INVALID_PARAMETER);

   usleep(10000);

    bufferObject = DdiMedia_GetSurfaceFromVASurfaceID(mediaCtx, surface);
    DDI_CHK_NULL(bufferObject, "Null bufferObject", VA_STATUS_ERROR_INVALID_SURFACE);
    DdiMediaUtil_MediaPrintFps();
    pitch = bufferObject->iPitch;

Xiaogangli-intel, 2022-09-05:
Thank you @YongZhuIntel, could you help try to add `mos_bo_wait_rendering(drawable_bo)` before `DdiMediaUtil_LockMutex(&mediaCtx->PutSurfaceRenderMutex);
    eStatus = vpHal->Render(&renderParams);`?

YongZhuIntel, 2022-09-06:
@Xiaogangli-intel I have added    mos_bo_wait_rendering(drawable_bo) before DdiMediaUtil_LockMutex(&mediaCtx->PutSurfaceRenderMutex);  but the garbage still appears

    renderParams.uSrcCount                  = 1;
    renderParams.uDstCount                  = 1;
    renderParams.pSrc[0]                    = &Surf;
    renderParams.pTarget[0]                 = &target;
    renderParams.pColorFillParams           = &colorFill;
    renderParams.pColorFillParams->Color    = 0xFF000000;
    renderParams.pColorFillParams->bYCbCr   = false;
    renderParams.pColorFillParams->CSpace   = CSpace_sRGB;

    mos_bo_wait_rendering(drawable_bo);//<--------added 

    DdiMediaUtil_LockMutex(&mediaCtx->PutSurfaceRenderMutex);
    eStatus = vpHal->Render(&renderParams);
    if (MOS_FAILED(eStatus))
    {
        DdiMediaUtil_UnLockMutex(&mediaCtx->PutSurfaceRenderMutex);
        mos_bo_unreference(drawable_bo);
        return VA_STATUS_ERROR_OPERATION_FAILED;
    }

Xiaogangli-intel, 2022-09-06:
Thanks @YongZhuIntel, did you also tried to add `mos_bo_wait_rendering(src_bo)`? Because this issue seems caused by synchronization.

YongZhuIntel, 2022-09-09:
@Xiaogangli-intel I have tried to add mos_bo_wait_rendering(bufferObject->bo); but it doesn't work. 
I also modified MPV so that it won't redraw when the window is resized, but it can't solve the problem of garbage screen when the window is resized during playback.

YongZhuIntel, 2022-10-11:
@LhGu  I have tried the main branch, and the border corruption issue have been fixed.  However,  when resizing the window quickly, there is still a probability (about 1/20) that the garbage screen will appear, which looks like the following:

[image]

XinfengZhang, 2022-10-11:
@YongZhuIntel  ask a question about it, why not use "mpv --hwdec=vaapi --vo=gpu"
  -vo=vaapi will call vaPutSurface , and there are one additional memory copy from vaSurface to X11 DRI drawable

LhGu, 2022-10-11:
Because I could not reproduce this garbage screen, I synced with @YongZhuIntel  offline, he will try again and I will get debug info from his environment if he can reproduce this issue.

LhGu, 2022-10-11:
> @YongZhuIntel ask a question about it, why not use "mpv --hwdec=vaapi --vo=gpu" -vo=vaapi will call vaPutSurface , and there are one additional memory copy from vaSurface to X11 DRI drawable

@XinfengZhang  They found that -vo=vaapi  command has less power cost, so they want to use  -vo=vaapi  command.   --hwdec=vaapi --vo=gpu command has no garbage screen issue by their test.

LhGu, 2022-10-13:
Synced with [YongZhuIntel](https://github.com/YongZhuIntel) offline.  Issue may not be related with media driver. I could not reproduce it in our environment. @YongZhuIntel  reproduced it in their Tablet PC.
Corruption as below:
[image]
[image]

LhGu, 2022-10-13:
It may be related with its display environment from corruption images

closed 2022-10-13 by YongZhuIntel