intel/media-driver #1467: [Bug]: intermittent blank frames in 64CH VPP scale + VPP composite

uartie, 2022-08-08:
### Which component impacted?

Decode, Video Processing

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

On TGL with 64 channel VPP scale + VPP composition using "```default```" scale method, some channels intermittently produce blank frames when using NV12 or YUY2 formats (and maybe others).  It does not appear to be an issue when using YV12 format or when using "```fast```" scale method.

Also note, this is a multi-pass VPP pipeline where scaling is done in separate VPP instance and the composition is done in the final VPP instance.

If we combine all VPP (scale and composition) into a single VPP instance, then this issue cannot be reproduced.

**Multi-pass VPP command-line:**
```shell
export GST_GL_PLATFORM=egl
gst-launch-1.0 -vf \
  vacompositor scale-method=default \
    sink_0::xpos=0 sink_0::ypos=0 \
    sink_1::xpos=480 sink_1::ypos=0 \
    sink_2::xpos=960 sink_2::ypos=0 \
    sink_3::xpos=1440 sink_3::ypos=0 \
    sink_4::xpos=1920 sink_4::ypos=0 \
    sink_5::xpos=2400 sink_5::ypos=0 \
    sink_6::xpos=2880 sink_6::ypos=0 \
    sink_7::xpos=3360 sink_7::ypos=0 \
    sink_8::xpos=0 sink_8::ypos=270 \
    sink_9::xpos=480 sink_9::ypos=270 \
    sink_10::xpos=960 sink_10::ypos=270 \
    sink_11::xpos=1440 sink_11::ypos=270 \
    sink_12::xpos=1920 sink_12::ypos=270 \
    sink_13::xpos=2400 sink_13::ypos=270 \
    sink_14::xpos=2880 sink_14::ypos=270 \
    sink_15::xpos=3360 sink_15::ypos=270 \
    sink_16::xpos=0 sink_16::ypos=540 \
    sink_17::xpos=480 sink_17::ypos=540 \
    sink_18::xpos=960 sink_18::ypos=540 \
    sink_19::xpos=1440 sink_19::ypos=540 \
    sink_20::xpos=1920 sink_20::ypos=540 \
    sink_21::xpos=2400 sink_21::ypos=540 \
    sink_22::xpos=2880 sink_22::ypos=540 \
    sink_23::xpos=3360 sink_23::ypos=540 \
    sink_24::xpos=0 sink_24::ypos=810 \
    sink_25::xpos=480 sink_25::ypos=810 \
    sink_26::xpos=960 sink_26::ypos=810 \
    sink_27::xpos=1440 sink_27::ypos=810 \
    sink_28::xpos=1920 sink_28::ypos=810 \
    sink_29::xpos=2400 sink_29::ypos=810 \
    sink_30::xpos=2880 sink_30::ypos=810 \
    sink_31::xpos=3360 sink_31::ypos=810 \
    sink_32::xpos=0 sink_32::ypos=1080 \
    sink_33::xpos=480 sink_33::ypos=1080 \
    sink_34::xpos=960 sink_34::ypos=1080 \
    sink_35::xpos=1440 sink_35::ypos=1080 \
    sink_36::xpos=1920 sink_36::ypos=1080 \
    sink_37::xpos=2400 sink_37::ypos=1080 \
    sink_38::xpos=2880 sink_38::ypos=1080 \
    sink_39::xpos=3360 sink_39::ypos=1080 \
    sink_40::xpos=0 sink_40::ypos=1350 \
    sink_41::xpos=480 sink_41::ypos=1350 \
    sink_42::xpos=960 sink_42::ypos=1350 \
    sink_43::xpos=1440 sink_43::ypos=1350 \
    sink_44::xpos=1920 sink_44::ypos=1350 \
    sink_45::xpos=2400 sink_45::ypos=1350 \
    sink_46::xpos=2880 sink_46::ypos=1350 \
    sink_47::xpos=3360 sink_47::ypos=1350 \
    sink_48::xpos=0 sink_48::ypos=1620 \
[... 195 lines cut ...]
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0. \
  filesrc location=./testsrc.h264 ! parsebin ! vah264dec  ! queue ! vacompositor0.
```

### What's the usage scenario when you are seeing the problem?

Video Analytics, Video Conference, Others

uartie, 2022-08-08:
[streams.zip](https://github.com/intel/media-driver/files/9284643/streams.zip)

uartie, 2022-08-08:
[image]

stellawuintel, 2022-08-09:
Auto Created VSMGWL-55537 for further analysis.

stellawuintel, 2022-08-09:
Auto Created VSMGWL-55538 for further analysis.

LhGu, 2022-08-24:
Reproduce in latest driver. Some channel failures because of a lot of workloads and timeout  happened on waiting for freeing Vebox State heap.  Maybe we need to change timeout judge condition.

LhGu, 2022-08-25:
> VSMGWL-55538
I think its impact is P2. I will change jirra. Thanks. @uartie

PR #1480 (2022-09-06): Fix sleep issue
Fixed #1467
commit 52505602 (2022-09-06): Fix sleep issue

Fixed #1467
closed 2022-09-06 by intel-mediadev