intel/media-driver #1468: [Bug]: 32 bit build regression: error: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 3 has type ‘uint64_t’ {aka ‘long long unsigned int’} [-Werror=format=]

sebastinas, 2022-08-10:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

The build of 22.5.1 is now failing on Debian's i386 architecture:

```
[ 19%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/common/codec/hal/codechal_encode_avc_base.cpp.o
cd /<<PKGBUILDDIR>>/obj-i686-linux-gnu/media_driver && /usr/bin/c++ -DCLASS_TRACE=0 -DENABLE_KERNELS -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_JSL_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN12_ADLN_SUPPORTED -DIGFX_GEN12_ADLP_SUPPORTED -DIGFX_GEN12_ADLS_SUPPORTED -DIGFX_GEN12_DG1_SUPPORTED -DIGFX_GEN12_RKL_SUPPORTED -DIGFX_GEN12_SUPPORTED -DIGFX_GEN12_TGLLP_CMFCPATCH_SUPPORTED -DIGFX_GEN12_TGLLP_CMFC_SUPPORTED -DIGFX_GEN12_TGLLP_SUPPORTED -DIGFX_GEN12_TGLLP_SWSB_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_CML_SUPPORTED -DIGFX_GEN9_CMPV_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DIGFX_MHW_INTERFACES_NEXT_SUPPORT -DMEDIA_VERSION=\"22.5.1\" -DMEDIA_VERSION_DETAILS=\"\" -DVEBOX_AUTO_DENOISE_SUPPORTED=0 -DX11_FOUND -D_AV1_DECODE_SUPPORTED -D_AV1_ENCODE_VDENC_SUPPORTED -D_AVC_DECODE_SUPPORTED -D_AVC_ENCODE_VDENC_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_FULL_OPEN_SOURCE -D_HEVC_DECODE_SUPPORTED -D_HEVC_ENCODE_VDENC_SUPPORTED -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_RELEASE -D_VP8_DECODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1 -I/<<PKGBUILDDIR>>/media_driver/linux/common/os/i915/include -I/<<PKGBUILDDIR>>/media_driver/linux/common/os/i915/include/uapi -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/os -I/<<PKGBUILDDIR>>/media_driver/linux/common/os -I/<<PKGBUILDDIR>>/media_common/agnostic/common/os -I/<<PKGBUILDDIR>>/media_common/linux/common/os -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/os -I/<<PKGBUILDDIR>>/media_softlet/linux/common/os/osservice -I/<<PKGBUILDDIR>>/media_softlet/linux/common/os -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/cm -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/hw -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/media_interfaces -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/renderhal -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/shared/user_setting -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/shared/mediacopy -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/shared -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen8/cm -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen8/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen8/hw -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen8/renderhal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen8_bdw/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen8_bdw/renderhal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9/cm -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9/hw -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9/renderhal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9_bxt/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9_skl/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9_glk/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9_kbl/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen10/cm -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen10/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen10/hw -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen10/renderhal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen11/cm -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen11/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen11/hw -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen11/renderhal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen11_jsl_ehl/renderhal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen12/cm -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen12/hw/vdbox -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen12/hw -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen12/shared -I/<<PKGBUILDDIR>>/media_driver/agnostic/g12/g12_base/hw/render -I/<<PKGBUILDDIR>>/media_driver/agnostic/Xe_M/Xe_M_base/shared -I/<<PKGBUILDDIR>>/media_driver/agnostic/Xe_R/Xe_HP_base/hw/render -I/<<PKGBUILDDIR>>/media_driver/agnostic/Xe_R/Xe_HP_base/hw/blt -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/common/shared/media_sfc_interface -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/common/shared/scalability -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/m12/m12_0/shared/mediacopy -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/g12/g12_base/renderhal -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/g12/g12_0/renderhal -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/g12/g12_1/renderhal -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/Xe_R/Xe_HP_Base/renderhal -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m8_bdw -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m9_bxt -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m9_skl -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m9_cfl -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m9_glk -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m9_kbl -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m10_cnl -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m11_icllp -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m11_jsl_ehl -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m12_tgllp -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m12_dg1 -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m12_rkl -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m12_adls -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m12_adlp -I/<<PKGBUILDDIR>>/media_driver/media_interface/media_interfaces_m12_adln -I/<<PKGBUILDDIR>>/media_common/agnostic/common/heap_manager -I/<<PKGBUILDDIR>>/media_common/agnostic/common/hw/vdbox -I/<<PKGBUILDDIR>>/media_common/agnostic/common/hw -I/<<PKGBUILDDIR>>/media_common/agnostic/common/shared/user_setting -I/<<PKGBUILDDIR>>/media_common/agnostic/common/shared -I/<<PKGBUILDDIR>>/media_common/agnostic/common/media_interfaces -I/<<PKGBUILDDIR>>/media_common/agnostic/common/renderhal -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/shared/scalability -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/shared/mediacontext -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/shared -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/heap_manager -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/hw/vdbox -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/hw -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/renderhal -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/media_interfaces -I/<<PKGBUILDDIR>>/media_common/agnostic/common/vp/kernel -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/common/vp/kdll -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen8/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9_bxt/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen9_glk/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen10/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen11/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen11_icllp/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen11_icllp/vp/kernel_free -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen11_jsl_ehl/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen12/vp/hal -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/gen12/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen12_tgllp/vp/hal -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen12_tgllp/vp/kernel_free/cmfc -I/<<PKGBUILDDIR>>/media_driver/agnostic/gen12_tgllp/vp/kernel_free/cmfcpatch -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/common/vp/hal/packet -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/common/vp/hal/shared/scalability -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/gen12_tgllp/vp/hal/platform_interface -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/m12/m12/vp/hal/packet -I/<<PKGBUILDDIR>>/media_driver/media_softlet/agnostic/m12/m12_0/vp/hal/feature_manager -I/<<PKGBUILDDIR>>/media_softlet/agnostic/common/vp/cm_fc_ld -I/<<PKGBUILDDIR>>/media_common/agnostic/common/vp/hal -I/<<PKGBUILDDIR>>/media_common/ [... cut ...]
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                         |
      |                         uint64_t {aka long long unsigned int}
```

See https://buildd.debian.org/status/fetch.php?pkg=intel-media-driver&arch=i386&ver=22.5.1%2Bdfsg1-1&stamp=1659479611&raw=0 for the full log and https://bugs.debian.org/1016953 for the Debian bug report.

### What's the usage scenario when you are seeing the problem?

Others

### Do you want to contribute a patch to fix the issue?

No.

sebastinas, 2022-08-10:
There is a partial fix by Simon McVittie in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1016953#10 … but there are more issues.

sebastinas, 2022-09-17:
I have reverted the version in Debian to 22.4.3 until this issue is fixed.

Sherry-Lin, 2022-09-19:
@cqian2  could you take a look with priority?

Xiaogangli-intel, 2022-09-19:
@sebastinas, thanks for reporting this issue, we will check it. I'm very sorry for the inconvenience.

stellawuintel, 2022-09-19:
Auto Created VSMGWL-57029 for further analysis.

cqian2, 2022-09-19:
Would you please help try https://github.com/intel/media-driver/pull/1493 ?
I locally tried and the linux build could pass for both 32bit and 64bit systems.
BTW, we have never tried to build 32bit linux here. I still need to double check w/ folks here about whether or not we had such support requirement here.

sebastinas, 2022-09-21:
This patch only fixes the first issue. It leaves

```
In file included from /<<PKGBUILDDIR>>/media_common/agnostic/common/os/mos_os.h:31,
                 from /<<PKGBUILDDIR>>/media_driver/agnostic/common/os/mos_context.h:30,
                 from /<<PKGBUILDDIR>>/media_driver/linux/common/ddi/media_libva_common.h:37,
                 from /<<PKGBUILDDIR>>/media_driver/linux/common/cm/ddi/media_libva_cm.h:30,
                 from /<<PKGBUILDDIR>>/media_driver/linux/common/cm/hal/cm_def_os.h:41,
                 from /<<PKGBUILDDIR>>/media_driver/agnostic/common/cm/cm_def.h:30,
                 from /<<PKGBUILDDIR>>/media_driver/agnostic/common/cm/cm_kernel.h:30,
                 from /<<PKGBUILDDIR>>/media_driver/agnostic/common/cm/cm_kernel_rt.h:30,
                 from /<<PKGBUILDDIR>>/media_driver/agnostic/common/cm/cm_kernel_ex.h:28,
                 from /<<PKGBUILDDIR>>/media_driver/agnostic/common/cm/cm_kernel_ex.cpp:27:
In static member function ‘static _Ty* MosUtilities::MosNewArrayUtil(size_t) [with _Ty = unsigned char; _Types = {}]’,
    inlined from ‘virtual int32_t CmKernelEx::Initialize(const char*, const char*)’ at /<<PKGBUILDDIR>>/media_driver/agnostic/common/cm/cm_kernel_ex.cpp:192:22,
    inlined from ‘virtual int32_t CmKernelEx::Initialize(const char*, const char*)’ at /<<PKGBUILDDIR>>/media_driver/agnostic/common/cm/cm_kernel_ex.cpp:70:9:
/<<PKGBUILDDIR>>/media_softlet/agnostic/common/os/mos_utilities.h:2790:16: error: argument 1 range [2147483649, 4294967295] exceeds maximum object size 2147483647 [-Werror=alloc-size-larger-than=]
 2790 |     _Ty* ptr = new (std::nothrow) _Ty[numElements]();
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/12/bits/exception_ptr.h:40,
                 from /usr/include/c++/12/exception:168,
                 from /usr/include/c++/12/ios:39,
                 from /usr/include/c++/12/ostream:38,
                 from /usr/include/c++/12/iostream:39,
                 from /<<PKGBUILDDIR>>/media_driver/linux/common/cm/hal/cm_def_os.h:34:
/usr/include/c++/12/new: In member function ‘virtual int32_t CmKernelEx::Initialize(const char*, const char*)’:
/usr/include/c++/12/new:142:26: note: in a call to allocation function ‘void* operator new [](std::size_t, const std::nothrow_t&)’ declared here
  142 | _GLIBCXX_NODISCARD void* operator new[](std::size_t, const std::nothrow_t&) _GLIBCXX_USE_NOEXCEPT
      |                          ^~~~~~~~
```
undfixed.

cqian2, 2022-09-22:
Hi, @sebastinas, 

I used a 64bit Linux machine to force compile 32bit driver (-DARCH=32 -DCMAKE_CXX_FLAGS=-m32 -DCMAKE_C_FLAGS=-m32), tried my patch and it could pass both the issues in the build log. You can find below info (No issue for cm_kernel_ex.cpp):

[ 38%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen10/codec/hal/codechal_vdenc_hevc_g10.cpp.o
[ 39%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen10/codec/hal/codechal_vdenc_vp9_g10.cpp.o
[ 39%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen10/codec/hal/codechal_encode_mpeg2_g10.cpp.o
[ 39%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/agnostic/common/cm/cm_kernel_ex.cpp.o
[ 39%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen10/codec/hal/codechal_encode_vp8_g10.cpp.o
[ 39%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen10/codec/kernel/igcodeckrn_g10.c.o
[ 39%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/agnostic/gen11/codec/hal/codechal_hw_g11_X.cpp.o

Thus the only reason why it still failed on your side as I can think may be your using a real 32bit Linux. 
BTW, since this issue was identified as a build regression, would you please help provide the regression commit thus we can identify which change caused the issues?

sebastinas, 2022-09-22:
Note that the second issue only appeared with GCC 12. Are you building with GCC 12?

Jexu, 2022-09-22:
Did you upgrade the gcc version in this time or just only update media to newer version? Looks like that your compiler treats type of uint32_t and size_t with different value range.

sebastinas, 2022-09-22:
Maybe it's better to split the bug into two: the format flag issue was triggered  by an upgrade of media-driver. That's fixed by the PR.

The other issue is due to GCC 12 emitting new warnings and media-driver building with -Werror by default. Note that this issue was first reported in April (see https://github.com/intel/media-driver/issues/1353#issuecomment-1098300408).

cqian2, 2022-09-22:
Hi, @sebastinas, I just updated the PR and would you please help try whether or not it can solve the second issue now?

sebastinas, 2022-09-22:
I can confirm that this now builds. Shouldn't the check also include `sizeof(_Ty)`?

cqian2, 2022-09-22:
Glad to see it now builds.
We once thought that (numElements > PTRDIFF_MAX) might be just over checking, when we were upgrading to use gcc 12.0.1, then here now, it proves that it is still necessary. But sizeof(_Ty) has no over checking problem.

closed 2022-09-26 by cqian2

SuperSandro2000, 2022-11-21:
Similar issue in 2.6.3

```
/build/source/media_softlet/linux/common/ddi/media_libva_util_next.cpp: In static member function 'static void MediaLibvaUtilNext::MediaPrintFps()':
/build/source/media_softlet/linux/common/ddi/media_libva_util_next.cpp:2199:47: error: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'uint64_t' {aka 'long long unsigned int'} [-Werror=format=]
 2199 |         sprintf(temp,"FPS:%6.4f, Interval:%11lu\n", fps,((uint64_t)tv2.tv_sec)*1000 + (tv2.tv_usec/1000));
      |                                           ~~~~^         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |                                     |
      |                                               long unsigned int                     uint64_t {aka long long unsigned int}
```

cqian2, 2022-11-21:
> Similar issue in 2.6.3
>
> ```
> /build/source/media_softlet/linux/common/ddi/media_libva_util_next.cpp: In static member function 'static void MediaLibvaUtilNext::MediaPrintFps()': [...]

Hi, @SuperSandro2000 , would you please help apply the same change in https://github.com/intel/media-driver/pull/1493 and try again? If fail, please help report a new issue.

SuperSandro2000, 2022-11-21:
I'll open a new issue but I don't know how to represent the ``%11lu`` used here instead of the ``%lu``.

PR #1493 (2022-09-23): Fix the 32 bit build regression
Fixes: #1468