intel/media-driver #1471: Video acceleration stops working with uptime until reboot

david-morris, 2022-08-13:
I have a Jasper Lake chip and video acceleration will sometimes stop working until I reboot. Videos will start stuttering, CPU usage will go up, and intel_gpu_top will show an increase in render/3D load, often to near 100%, while the Video load stays at 0%.

If I force reload i915 (i.e. `rmmod -f` and then an `insmod`), video acceleration will start working again, but it breaks my sound stack so I have to reboot anyway. If I figure out how to unbreak sound, I'll post about it here, but that wouldn't really fix the issue fully.

I've posted about this in a [different issue](https://github.com/intel/media-driver/issues/1298) but I think this is a separate issue.

here's my `vainfo`:
```
$ DISPLAY=:0 vainfo                                                                                                                                                    (08-13 14:20) 
vainfo: VA-API version: 1.15 (libva 2.15.0)                                                                                                                                                                
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.4.4 ()                                                                                                                             
vainfo: Supported profile and entrypoints                                                                                                                                                                  
      VAProfileNone                   : VAEntrypointVideoProc                                                                                                                                              
      VAProfileNone                   : VAEntrypointStats                                                                                                                                                  
      VAProfileMPEG2Simple            : VAEntrypointVLD                                                                                                                                                    
      VAProfileMPEG2Main              : VAEntrypointVLD                                                                                                                                                    
      VAProfileH264Main               : VAEntrypointVLD                                                                                                                                                    
      VAProfileH264Main               : VAEntrypointEncSliceLP                                                                                                                                             
      VAProfileH264High               : VAEntrypointVLD                                                                                                                                                    
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
```
And some info from `lspci` about my graphics chip: 
```
00:02.0 VGA compatible controller: Intel Corporation JasperLake [UHD Graphics] (rev 01) (prog-if 00 [VGA controller])
        DeviceName: Onboard - Video                                                                  
        Subsystem: ASUSTeK Computer Inc. Device 8813                                                 
        Flags: bus master, fast devsel, latency 0, IRQ 135                  
        Memory at 6000000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 4000000000 (64-bit, prefetchable) [size=256M]                                      
        I/O ports at 4000 [size=64]                                                                  
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]                                   
        Capabilities: [40] Vendor Specific Information: Len=0c <?>                                   
        Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00                          
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-                                   
        Capabilities: [d0] Power Management version 2                                                
        Capabilities: [100] Process Address Space ID (PASID)                                         
        Capabilities: [200] Address Translation Service (ATS)                                        
        Capabilities: [300] Page Request Interface (PRI)                                                                                                                                                   
        Kernel driver in use: i915
        Kernel modules: i915
```
And the modules which depend on i915 (recursively):
```
i915                 3432448  25
drm_buddy              20480  1 i915
video                  57344  2 asus_wmi,i915
ttm                    86016  1 i915
drm_dp_helper         167936  1 i915
intel_gtt              28672  1 i915
```
So that's not a lot of info to go off of. I haven't found any errors anywhere when this happens. That might be because there's a lot of background noise in the Kodi log.  Is there a debugging info module parameter I should give to one of these modules so I can post info from the kernel message buffer?

XinfengZhang, 2022-08-15:
is there some suspicious point in dmesg?

david-morris, 2022-08-17:
@XinfengZhang I haven't seen one. Is there a kernel parameter I should set on i915 when I get a chance?

Xiaogangli-intel, 2022-08-22:
@david-morris, just `dmesg | grep i915` is ok.

david-morris, 2022-08-28:
Sorry for the long wait, it took a while before it happened again.
Here's the output:
```
$ sudo dmesg | grep "i915"                                                                                                                                            (08-28 19:46)
[    3.454814] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.457456] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.460929] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/icl_dmc_ver1_09.bin (v1.9)
[    3.489376] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.561643] fbcon: i915drmfb (fb0) is primary device
[    3.621034] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    3.775568] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    4.078285] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
```
And that's after video acceleration had been working for quite a while, so those messages were all from well before the issue came up.

Xiaogangli-intel, 2022-09-05:
There doesn't seem to be any useful information in the dmesg log. @david-morris, could you provide application information or common line you are using?

paulmenzel, 2022-09-07:
s/common/command/

You can configure Linux’ DRM log messages using `/sys/module/drm/parameters/debug`, for example, `echo 0xe` | sude too /sys/module/drm/parameters/debug`.

david-morris, 2022-09-14:
@Xiaogangli-intel I'm using the default Kodi player with the Arch package kodi-standalone. I'll try to get more useful information (e.g. groups of the Kodi user, executable of the player) this weekend.

@paulmenzel I'll check out that sysfs file, thanks for the tip. I'll try running with debug on until the bug pops up.

david-morris, 2022-09-17:
@Xiaogangli-intel here's the information about my kodi player binary:
```
$ sudo pstree kodi                                                                                                                                                    (09-17 10:12)
kodi.bin───23*[{kodi.bin}]

kodi.bin───30*[{kodi.bin}]

systemd─┬─(sd-pam)
        ├─dbus-daemon
        └─pulseaudio─┬─gsettings-helpe───3*[{gsettings-helpe}]
                     └─{pulseaudio}

xinit─┬─(sd-pam)
      ├─Xorg───10*[{Xorg}]
      └─kodi-standalone───kodi───kodi.bin─┬─4*[sh]
                                          └─48*[{kodi.bin}]
$ kodi --version                                                                                                                                                      (09-17 10:13)
19.4 (19.4.0) Git:20220429-nogitfound Media Center Kodi
Copyright (C) 2005-2021 Team Kodi - http://kodi.tv
```

@paulmenzel I set that debug flag _after_ the bug popped up, and the kernel message buffer filled up. I sprunged [the grepped output](http://sprunge.us/pWFvPj) of dmesg.

Jexu, 2022-11-23:
Your description indicates Render/3D workload increases to 100%, but video goes to 0% which means it does's stuck on video decoding. I suspect something about copy, vpp or display causing stuck.
Could you try ffmpeg or sample decoder for decoding only with you content. Or share the libva trace when stucking happens.

david-morris, 2022-11-23:
@Jexu I've recently switched to [lufinkey/kodi-openbox](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwii8ZiHhMT7AhVOQPEDHd9CBrAQFnoECA4QAQ&url=https%3A%2F%2Fgithub.com%2Flufinkey%2Fkodi-openbox&usg=AOvVaw34gR5QDQiJS7TCkHgAz93N) and can't recall the issue recurring since then. If I notice it again, I'll share more diagnostics.  

But if not, it might be an issue with certain types of "kiosk mode" where there's no display manager or window manager.

Jexu, 2022-11-24:
Good to know issue is gone.
If you are interested in, you could ask kodi for more details about the difference between both types of mode to help understand what is the reason.

david-morris, 2022-11-24:
I think it's an issue between a lot of projects. If anyone else is reading this because they're having this problem, I'd suggest trying a kiosk mode for Kodi based on privileges of an autologin user (like kodi-openbox) rather than a dedicated systemd script and a non-login user (like kodi-standalone).

Jexu, 2022-11-25:
It is good guidence for anyone who has same problem.
So far, this issue trends to be solved and i will close it. Just re-open it again if still having any other questions.

closed 2022-11-25 by Jexu