intel/media-driver #1472: [Bug]: build failure for media_libva_putsurface_linux.cpp

renwu58, 2022-08-13:
### Which component impacted?

Build

### What happened?

OS:
Jeffy@Jeffy-PC:~/soft/media-driver/build$ uname -a
Linux Jeffy-PC 5.15.34-amd64-desktop #2 SMP Mon May 16 16:31:30 CST 2022 x86_64 GNU/Linux

I have already build the latest verison of libva with X11 and GMMLib everything is seems fine. but when build the package following the instruction: https://01.org/linuxmedia/quickstart/linux-vaapi-videostack-environment build the media-driver have following error.

```
Jeffy@Jeffy-PC:~/soft/media-driver/build$ make -j8
[  0%] Built target iHD_drv_video_CP
[  0%] Built target iHD_drv_video_SSE2
[  0%] Built target iHD_drv_video_SSE4
[  2%] Built target igfxcmrt
[  8%] Built target iHD_drv_video_mos
Consolidate compiler generated dependencies of target iHD_drv_video_COMMON
[ 23%] Built target iHD_drv_video_VP
[ 67%] Built target iHD_drv_video_CODEC
[ 67%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/ddi/media_libva_putsurface_linux.cpp.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/renderhal/renderhal_dsh_specific.c.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/vp/ddi/media_libva_vp.c.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/vp/ddi/media_libva_vp_tools.c.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/vp/hal/vphal_common_specific.c.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/renderhal/renderhal_linux.cpp.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/hw/mhw_mi_linux.cpp.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/vp/hal/vphal_render_common_specific.c.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/shared/hal_oca_interface.cpp.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/gen8/ddi/media_libva_caps_g8.cpp.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/gen8/ddi/media_sysinfo_g8.cpp.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/gen8/ddi/media_sku_wa_g8.cpp.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/gen9/ddi/media_libva_caps_g9.cpp.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/gen9/ddi/media_sysinfo_g9.cpp.o
[ 69%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/gen9/ddi/media_sku_wa_g9.cpp.o
[ 69%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/gen9_bxt/ddi/media_libva_caps_g9_bxt.cpp.o
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp: In function ‘bool output_dri_init(VADriverContextP)’:
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:166:18: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
     mediaDrvCtx->dri_output = nullptr;
                  ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:178:18: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
     mediaDrvCtx->dri_output = (va_dri_output*) calloc(1, sizeof(struct va_dri_output));
                  ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:179:23: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
     if (!mediaDrvCtx->dri_output){
                       ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:183:18: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
     mediaDrvCtx->dri_output->handle = dso_open(LIBVA_X11_NAME);
                  ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:184:23: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
     if (!mediaDrvCtx->dri_output->handle){
                       ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:185:27: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
         free(mediaDrvCtx->dri_output);
                           ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:186:22: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
         mediaDrvCtx->dri_output = nullptr;
                      ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:190:31: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
     dso_handle = mediaDrvCtx->dri_output->handle;
                               ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:191:32: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
     dri_vtable = &mediaDrvCtx->dri_output->vtable;
                                ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:193:32: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
         dso_close(mediaDrvCtx->dri_output->handle);
                                ^~~~~~~~~~
/home/Jeffy/soft/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:194:27: error: ‘struct DDI_MEDIA_CONTEXT’ has no member named ‘dri_output’
[... 167 lines cut ...]
Others

### What impacted?

The package build failure and cannot install

### Debug Information

## Packages verison

libva: 2.15.0
libva-utils: 2.15.0
gmmlib: 2.15.0
media-driver:  latest master branch 22.5.3 

## ls /dev/dri
```
Jeffy@Jeffy-PC:~/soft/media-driver/build$ ls /dev/dri
ls: cannot access '/dev/dri': No such file or directory
```

```
Jeffy@Jeffy-PC:~/soft/media-driver/build$ vainfo 
error: failed to initialize display
```

XinfengZhang, 2022-08-15:
suppose it is because
#if !defined(ANDROID) && defined(X11_FOUND) 
so, it is X11_FOUND is false

kingskill, 2022-09-07:
> suppose it is because #if !defined(ANDROID) && defined(X11_FOUND) so, it is X11_FOUND is false

So how to fix it?

cqian2, 2022-11-23:
Hi, @renwu58 ,

Would you please help verify it with PR#https://github.com/intel/media-driver/pull/1527 ?

Thanks
Chunqiang

cqian2, 2022-11-30:
I tried PR#1527 but couldn't reproduce the issue. Close it now.
Feel free to reopen it if further reproducibility.

closed 2022-11-30 by cqian2