intel/media-driver #1478: [Bug][TGL][FFmpeg-vaapi] VP9 encode fail with 8K resolution

feiwan1, 2022-08-29:
### Which component impacted?

Encode

### What happened?

1. Linux/ffmepg-vaapi
2. with latest ffmepg&media-driver
3. cmd:
ffmpeg -v debug -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_output_format vaapi -f rawvideo -pix_fmt yuv420p -s:v 7680x4320  -r:v 30 -i 7680x4320_nv12.yuv -vf 'format=nv12,hwupload' -an -c:v vp9_vaapi -rc_mode CQP -g 30 -vframes 1 -y out.ivf
5. error:
[vp9_vaapi @ 0x5596b3970340] Failed to map output buffers: 24 (internal encoding error).
[vp9_vaapi @ 0x5596b3970340] Output failed: -5.
Error submitting video frame to the encoder

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### Debug Information

The issue caused by the incorrect behavior of vaSyncBuffer. Currently in ffmepg-vaapi and if VAAPI version 1.9.0+, then the encode logic will be like this:
    vaSyncBuffer(xxxx, VA_TIMEOUT_INFINITE);
    vaMapBuffer(xxxx);
Then media-driver will return VA_STATUS_ERROR_ENCODING_ERROR for vaMapBuffer.

I checked with the other way:
    vaSyncSurface(xxxx);
    vaMapBuffer(xxxx);
In this way, the pipeline works without any error.

stellawuintel, 2022-08-29:
Auto Created VSMGWL-56237 for further analysis.

feiwan1, 2023-06-16:
The issue still there with latest media-driver/ffmpeg. Hi @cchunbo, any update of this issue?

feiwan1, 2023-06-16:
BTW, DG2 no this issue.

feiwan1, 2023-10-27:
The issue is fixed by using kernel 6.2.0-34-generic. Close this. @cchunbo @XinfengZhang, could you help to check if you need to  close your internal VSMGWL-56237 issue too? Thanks.

closed 2023-10-27 by feiwan1