intel/media-driver #1481: [Bug]: Videos play sluggish with high CPU load on ApolloLake since commit 42ffd7983a32b75cf8403c235d8d0f727191390c

Markus-Pietrek-MSC, 2022-09-02:
### Which component impacted?

Not sure

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

Hello,

with intel-media-driver 22.3.1 used in Yocto Kirkstone videos are played sluggish with high CPU load on ApolloLake CPUs (Intel(R) Atom(TM) Processor E3930).

This was introduced in commit 42ffd7983a32b75cf8403c235d8d0f727191390c "[Media Common] GMM Multi-Adapter Interface Changes".

Can be reproduced with this video https://test-videos.co.uk/vids/bigbuckbunny/mp4/h264/1080/Big_Buck_Bunny_1080_10s_30MB.mp4

gst-launch-1.0 playbin uri=file:Big_Buck_Bunny_1080_10s_30MB.mp4
# which uses the sink xvimagesink by default, so behaves like
gst-launch-1.0 playbin uri=file:Big_Buck_Bunny_1080_10s_30MB.mp4 video-sink=xvimagesink

gst-launch uses 80% CPU load instead of 10% before. The video is also sluggish and it seems a lot of frames are dropped.

The video is played fine with the non-default vaapisink instead using only 10% CPU.

gst-launch-1.0 playbin uri=file:Big_Buck_Bunny_1080_10s_30MB.mp4 video-sink=vaapisink

Best regards,
Markus Pietrek
[libva_log.tar.gz](https://github.com/intel/media-driver/files/9475974/libva_log.tar.gz)

### What's the usage scenario when you are seeing the problem?

Playback

### Debug Information

Since 42ffd7983a32b75cf8403c235d8d0f727191390c

root@intel-corei7-64:~# vainfo
libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.14.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.4.4 (9ca43202d)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD

libva trace has been attached.

[libva_log.tar.gz](https://github.com/intel/media-driver/files/9475978/libva_log.tar.gz)

paulmenzel, 2022-09-07:
@kankanzh, @hanlong1 maybe you have an idea too.

stellawuintel, 2022-09-30:
Auto Created VSMGWL-57485 for further analysis.

kankanzh, 2022-11-09:
Hi @Markus-Pietrek-MSC , i have a draft test change want to have a try to see whether the issue can be reproduced or not, could you share more detailed reproduce steps? only need run "gst-launch-1.0 playbin uri=file:Big_Buck_Bunny_1080_10s_30MB.mp4"? How you check gst-launch uses 80% CPU load? Thanks.

Markus-Pietrek-MSC, 2022-11-09:
Hi @kankanzh,

this command is sufficient. You can check the CPU load with "top" while gst-launch is running. It is but should not be on the top.

Best regards,
Markus

XinfengZhang, 2022-11-09:
have you upgraded the gmmlib?

Markus-Pietrek-MSC, 2022-11-09:
Yocto Kirkstone uses gmmlib v22.1.2

JustPretender, 2023-02-23:
I'm experiencing a similar issue on _"Intel(R) Pentium(R) Silver N5030 CPU @ 1.10GHz"_, while using GStreamer to receive an RTP H.264 stream. The software is the same as the description mentions: Yocto Kirkstone + intel-media-driver 22.3.1.

GStreamer command:
```shell
~# gst-launch-1.0 -v udpsrc port=5000 caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" ! rtph264depay ! vaapih264dec ! videoconvert ! video/x-raw,width=1920,height=1080 ! queue ! waylandsink
```

Just as @Markus-Pietrek-MSC mentions the `top` is showing ~90% CPU usage for the GStreamer pipeline.

I've recorded a couple of seconds with [**perf**](https://www.brendangregg.com/perf.html):
```shell
~# record -p $(pgrep gst) -o /data/gst-intel-media-driver-rtp-h264.perf --call-graph dwarf
```

The interesting part:

```
Samples: 10K of event 'cycles', Event count (approx.): 7099918830
  Children      Self       Samples  Command         Shared Object                       Symbol
-   92.93%     0.00%             0  udpsrc0:src     libc.so.6                           [.] __clone3 (inlined)
     __clone3 (inlined)
     start_thread
     0x7fec967352ec
     0x7fec96735be3
   - 0x7fec968ac22e
      - 92.62% 0x7fec962925bc
         - 92.61% 0x7fec96875c98
            - 92.59% 0x7fec96873a0c
               - 92.51% 0x7fec95d3a2ea
                  - 92.42% 0x7fec95f0468b
                     - 92.39% 0x7fec95f04027
                        - 92.37% 0x7fec95f03df0
                             gst_rtp_base_depayload_push
                             0x7fec96875c98
                             0x7fec96873a0c
                           - 0x7fec95dd2ce1
                              - 92.35% 0x7fec95dcff65
                                 - 0x7fec95dc65d8
                                    - 92.34% 0x7fec95b99251
                                         gst_video_decoder_have_frame
                                       - 0x7fec95dcfb42
                                          - 91.94% 0x7fec95b9ad1a
                                             - gst_video_decoder_finish_frame
                                                - 91.91% 0x7fec95dc6f50
                                                   - 91.90% 0x7fec96875c98
                                                      - 0x7fec96873a0c
                                                         - 91.87% 0x7fec96296bb3
                                                            - 91.85% 0x7fec962974c0
                                                               - 89.75% 0x7fec95ddb912
                                                                  - gst_video_frame_map_id
                                                                     - 0x7fec95bb1d50
                                                                     - 0x7fec95bb1521
                                                                        - 89.73% 0x7fec95be08a1
                                                                             vaGetImage
                                                                           - DdiMedia_GetImage
                                                                              - 89.64% DdiMedia_CopySurfaceToImage (inlined)
                                                                                 - 89.63% DdiMedia_CopyPlane (inlined)
                                                                                      DdiMedia_GetImage
                                                                                      __memmove_sse2_unaligned_erms
```

It leads us to this line https://github.com/intel/media-driver/blob/11127e7697ed9b6d6d1821ce73a08123d26fc606/media_driver/linux/common/ddi/media_libva.cpp#L5219
as expected.
The interesting bit here probably is the `vaGetImage` implementation in the `intel-media-driver` vs `intel-vaapi-driver` which I tested with and couldn't reproduce this problem at all.

I can **still reproduce** the problem even if I revert down to the commit before [42ffd79](https://github.com/intel/media-driver/commit/42ffd7983a32b75cf8403c235d8d0f727191390c) though!

Since it's still open I assume the problem hasn't been solved or does anybody have a solution?

XinfengZhang, 2023-05-06:
+@jay, because look like related with vaGetImage implmentation