intel/media-driver #1482: [Bug]:

krzyk, 2022-09-06:
### Which component impacted?

Build

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

1. With latest builds from master of gmmlib and libva
2. `gcc (Ubuntu 10.3.0-1ubuntu1~20.04) 10.3.0`, `cmake version 3.16.3`
3. Build run using:
```
CMAKE_C_FLAGS_RELEASE=-O3 -march=native -mtune=native -fomit-frame-pointer -DNDEBUG
CMAKE_CXX_FLAGS_RELEASE=-O3 -march=native -mtune=native -fomit-frame-pointer -DNDEBUG

cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE=$CMAKE_C_FLAGS_RELEASE -DCMAKE_CXX_FLAGS_RELEASE=$CMAKE_CXX_FLAGS_RELEASE -DARCH=64 -DBUILD_TOOLS=OFF --build ..
make

```
5. Build fails with:
```
/home/jsmith/dev/inne/media-driver/media_softlet/agnostic/common/codec/hal/enc/av1/pipeline/encode_av1_reference_frames.cpp: In member function ‘void encode::Av1ReferenceFrames::GetFwdBwdRefPicList(CODEC_PICTURE (*)[15])’:
/home/jsmith/dev/inne/media-driver/media_softlet/agnostic/common/codec/hal/enc/av1/pipeline/encode_av1_reference_frames.cpp:571:39: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
  571 |             refsPicList[1][bwdRefNum] = refPic;
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/build.make:5900: media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/__/media_softlet/agnostic/common/codec/hal/enc/av1/pipeline/encode_av1_reference_frames.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:230: media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Just the build

stellawuintel, 2022-09-08:
Auto Created VSMGWL-56642 for further analysis.

krzyk, 2022-09-08:
Just wanted to add that `cmake` warns that there are two macros do nothing:
```
-- /home/jsmith/dev/inne/media-driver/media_driver/agnostic/common/vp/kernel/media_srcs.cmake doesn't exist, macro(media_include_subdirectory) just does nothing
-- /home/jsmith/dev/inne/media-driver/media_softlet/agnostic/common/media_interfaces/mmd/media_srcs.cmake doesn't exist, macro(media_include_subdirectory) just does nothing
```

And my CPU is Tiger Lake (`11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz`), not sure if that is useful info here.

Christa03, 2022-09-14:
@krzyk , we didn't reproduce the error info (Av1ReferenceFrames::GetFwdBwdRefPicList). Can you help share your media_compile_flags_linux.cmake if you change this file?

krzyk, 2022-09-14:
@Christa03 I didn't change any files in the repo, so media_compile_flags_linux.cmake is as in git.

Christa03, 2022-10-11:
@krzyk Can you help update to the latest code?

Xiaogangli-intel, 2022-10-24:
Hi @krzyk, did you check the latest driver? Any update on your side?

closed 2022-10-24 by krzyk