intel/media-driver #1483: [Bug]: [ADL-P][Fmpeg-vaapi][CrOs] VP9 decode fail with 8K resolution file

shinnint, 2022-09-07:
### Which component impacted?

Decode

### Is it regression? Good in old configuration?

Yes, it's good in old version

### What happened?

1. Occur on chrome video player and chrome/ffmpeg-vaapi
2. with ffmpeg and latest media driver (libva.2.1400, iHD) 
3. cmd: 
   ffmepg -v debug -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi -re -i <8K clip file> -f null -
4. error:
[vp9 @ 0x579058f64c40] Failed to end picture decode issue: 23 (internal decoding error)
...
Error while decoding stream #0:0 Input/Output error
...
Error while decoding stream #0:0: Resource temporarily unavailable

### What's the usage scenario when you are seeing the problem?

Playback

### What impacted?

P1 blocking issue

### Debug Information

1. vainfo 
[vainfo_felwinter.txt](https://github.com/intel/media-driver/files/9504944/vainfo_felwinter.txt)
2. 8k video file
given offline already, the file is too large
4. libvatrace 
[libvatrace.zip](https://github.com/intel/media-driver/files/9504955/libvatrace.zip)

Jexu, 2022-09-07:
Can you give the good version and bad version commit if possible?

paulmenzel, 2022-09-07:
Also check if it’s fixed with https://github.com/intel/libva/releases/tag/2.15.0.

stellawuintel, 2022-09-08:
Auto Created VSMGWL-56641 for further analysis.

shinnits, 2022-09-15:
Hi @Jexu,
I can provide bad commit -> Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.1.1 (27bc3fc)
Also I have enabled iHD logs and could see the below logs while ffmpeg the 8K vp9 video file.

[vp9 @ 0x55b0b83ef3b0] Failed to end picture decode issue: 23 (internal decoding error).
[vp9 @ 0x55b0b8403700] Param buffer (type 0, 92 bytes) is 0.=N/A speed=0.0261x    
[vp9 @ 0x55b0b8403700] Slice 0 param buffer (316 bytes) is 0x1.
[vp9 @ 0x55b0b8403700] Slice 0 data buffer (42152 bytes) is 0x2.
[vp9 @ 0x55b0b8403700] Decode to surface 0xa.
[MOS]:  CRITICAL - SubmitCommandBuffer:1363: Command buffer submission failed!
[CODEC]:CRITICAL - DecodePrimitiveLevel:1161: MOS returned error, eStatus = 0x23
[CODEC]:CRITICAL - Execute:1380: Primitive level decoding failed.
[LIBVA]:CRITICAL - EndPicture:916: DDI:DdiDecode_DecodeInCodecHal return failure.

Jexu, 2022-09-16:
Please check your kernel version and if the i915 guc firmware is loaded. From 22.1.1, media driver needs guc scheduling of i915 kernel for ADL-P.

shinnits, 2022-09-16:
Hi @Jexu, 
The kernel version is 5.10.136.
And i915.enable_guc =3, so it seems Enabled guc bit are set?
I will try with the latest media driver but it may take some times since would need to deploy it on ChromeOS.
Besides, may I ask if there are some patches regarding this issue I could try with this issue?

Jexu, 2022-09-16:
Yeah i915.enable_guc =3 will enable guc firmware, can you also share the dmesg log.
I suspect the kernel version maybe incorrect and I will take some time to check which kernel version is the first upstream one that supports guc scheduling on ADL-P. You can also try a newer one, 5.15.xxx e.g.

Jexu, 2022-09-16:
The first upstream kernel for ADL-P with guc schedule should be 5.17-rc5. So you may need to update the kernel to newer one otherwise you need to use old version media driver.

paulmenzel, 2022-09-16:
Can an appropriate error message be shown, when the requirements are not met? Also, where is the new requirement documented?

shinnits, 2022-09-20:
Hi @Jexu,

The enclosed file is the dmesg log when deploying kernel v5.10.
[dmesg_8k_v5.4.txt](https://github.com/intel/media-driver/files/9606240/dmesg_8k_v5.4.txt)

I have tried the more recent version of kernel, v6.00-rc4, ffmpeg with this 8k video file seems fine now.

As paulmenzel mentioned, is there any guideline showing what kind of requirement should be met regarding media driver and kernel ? It would be helpful for us.
Besides, it would be appreciated if there are more detailed information about guc scheduling patches can be shared.

Thanks!

Jexu, 2022-09-21:
For the limitation, you can refer to the page https://github.com/intel/media-driver#known-issues-and-limitations to get the started kernel version.
You could also get what new platforms supported from release note https://github.com/intel/media-driver/releases

Getting more tech details from media driver commit https://github.com/intel/media-driver/commit/697e7102ca41574899d8633417629c2d4ee1ea7a

paulmenzel, 2022-09-21:
@Jexu, probably the information can be found somewhere on the WWW, but it an understandable error message would be much more appropriate, as patches can also be backported to older Linux kernel versions.

paulmenzel, 2022-09-21:
> Getting more tech details from media driver commit https://github.com/intel/media-driver/commit/697e7102ca41574899d8633417629c2d4ee1ea7a

Also, there is no git commit message body detailing the Linux kernel requirements, so people would need to read the diff to know what is going on.

Jexu, 2022-09-21:
> @Jexu, probably the information can be found somewhere on the WWW, but it an understandable error message would be much more appropriate/userfriendly, as patches can also be backported to older Linux kernel versions.

Requirement of media driver version and kernel version for new platforms could be found in release note.
For this case, it is hard for umd to report error message since it depends on i915 reporting something like supported feature list.

jlahtine-intel, 2022-09-21:
> For this case, it is hard for umd to report error message since it depends on i915 reporting something like supported feature list.

@Jexu the HuC loading status already exists in the debugfs as `huc_info` file See [huc_info_show](https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/i915/gt/uc/intel_huc_debugfs.c#L12) from the i915 source.

If media driver adds code to read the status and report supported features to user, we could move it to sysfs.

Jexu, 2022-09-22:
@jlahtine-intel
It seems not formal solution for umd to read something from sys. User even could load/ unload guc and huc manually that will take umd to get unwanted result.
It is much helpful if i915 could add query api to get this info.

shinnits, 2022-09-22:
Hi @Jexu,

I have a further question regarding Guc Scheduling.
I found we have patched "Disable GucSubmission" on ADL-P ChromeOs.

diff --git a/media_driver/linux/gen12/ddi/media_sku_wa_g12.cpp b/media_driver/linux/gen12/ddi/media_sku_wa_g12.cpp
index b0dd89a38b1a..a95068210feb 100644
--- a/media_driver/linux/gen12/ddi/media_sku_wa_g12.cpp
+++ b/media_driver/linux/gen12/ddi/media_sku_wa_g12.cpp
@@ -556,7 +556,6 @@ static bool InitAdlpMediaSku(struct GfxDeviceInfo *devInfo,
 
     MEDIA_WR_SKU(skuTable, FtrE2ECompression, 0);
     MEDIA_WR_SKU(skuTable, FtrAV1VLDLSTDecoding, 1);
-    MEDIA_WR_SKU(skuTable, FtrGucSubmission, 1);
 
     return true;
 }
But still encounter this 8k VP9 decoding issue when using libva.2.1400.0.

Wondering if using < v5.17 kernel, is there any other way to toggle Guc Scheduling?

Jexu, 2022-09-23:
We don't commit supporting ADL-P with media v22.1.1 + kernel < v5.17, that is intermediate state, and it maybe accompanied with potential issue as you saw.
If you don't care about performance, you could try again to disable scalability like osInterface->bHcpDecScalabilityMode=false in line https://github.com/intel/media-driver/blob/master/media_driver/linux/common/os/mos_os_specific.c#L6814

Jexu, 2022-10-13:
If there is no other question, I trend to close this issue now.

Jexu, 2022-10-17:
Feel free to re-open again if having any other questions.

closed 2022-10-17 by Jexu