intel/media-driver #1484: [Bug]: Build bug 在建构到70%的时候出错

kingskill, 2022-09-07:
### Which component impacted?

Build

### What happened?

[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/ddi/media_libva_apo_decision.cpp.o                                                                                                                         
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/ddi/media_libva_putsurface_linux.cpp.o                                                                                                                     
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/media_softlet/agnostic/Xe_M/Xe_HPM/codec/hal/enc/av1/pipeline/encode_av1_vdenc_pipeline_xe_hpm.cpp.o                                                                     
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_caps.cpp: In member function ‘virtual VAStatus MediaLibvaCaps::QueryDisplayAttributes(VADisplayAttribute*, int32_t*)’:
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_caps.cpp:3101:20: error: ‘VADisplayPCIID’ was not declared in this scope
     attrib->type = VADisplayPCIID;
                    ^~~~~~~~~~~~~~
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_caps.cpp:3101:20: note: suggested alternative: ‘VADisplay’
     attrib->type = VADisplayPCIID;
                    ^~~~~~~~~~~~~~
                    VADisplay
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_caps.cpp: In member function ‘virtual VAStatus MediaLibvaCaps::GetDisplayAttributes(VADisplayAttribute*, int32_t)’:
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_caps.cpp:3122:18: error: ‘VADisplayPCIID’ was not declared in this scope
             case VADisplayPCIID:
                  ^~~~~~~~~~~~~~
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_caps.cpp:3122:18: note: suggested alternative: ‘VADisplay’
             case VADisplayPCIID:
                  ^~~~~~~~~~~~~~
                  VADisplay
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/build.make:3800：media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/ddi/media_libva_caps.cpp.o] 错误 1
make[2]: *** 正在等待未完成的任务....
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/media_softlet/agnostic/Xe_M/Xe_HPM/codec/hal/enc/av1/pipeline/encode_av1_vdenc_pipeline_adapter_xe_hpm.cpp.o                                                             
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/media_softlet/agnostic/Xe_M/Xe_HPM/codec/hal/enc/av1/features/encode_av1_basic_feature_xe_hpm.cpp.o                                                                      
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/media_softlet/agnostic/Xe_M/Xe_HPM/codec/hal/enc/av1/features/encode_av1_segmentation_xe_hpm.cpp.o                                                                       
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/media_softlet/agnostic/Xe_M/Xe_HPM/codec/hal/enc/av1/features/encode_av1_vdenc_feature_manager_xe_hpm.cpp.o                                                              
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp: In function ‘VAStatus DdiMediaUtil_AllocateSurface(DDI_MEDIA_FORMAT, int32_t, int32_t, PDDI_MEDIA_SURFACE, PDDI_MEDIA_CONTEXT)’:
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:175:5: error: ‘GMM_RESCREATE_CUSTOM_PARAMS_2’ was not declared in this scope
     GMM_RESCREATE_CUSTOM_PARAMS_2 gmmCustomParams;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:175:5: note: suggested alternative: ‘GMM_RESCREATE_CUSTOM_PARAMS’
     GMM_RESCREATE_CUSTOM_PARAMS_2 gmmCustomParams;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     GMM_RESCREATE_CUSTOM_PARAMS
In file included from /media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_common/agnostic/common/os/mos_os.h:31,
                 from /media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/agnostic/common/os/mos_context.h:30,
                 from /media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_common.h:37,
                 from /media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_util.h:29,
                 from /media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:38:
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:453:25: error: ‘gmmCustomParams’ was not declared in this scope
         MOS_ZeroMemory(&gmmCustomParams, sizeof(gmmCustomParams));
                         ^~~~~~~~~~~~~~~
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_softlet/agnostic/common/os/mos_utilities.h:3025:33: note: in definition of macro ‘MOS_ZeroMemory’
     MosUtilities::MosZeroMemory(pDestination, stLength)
                                 ^~~~~~~~~~~~
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:453:25: note: suggested alternative: ‘gmmParams’
         MOS_ZeroMemory(&gmmCustomParams, sizeof(gmmCustomParams));
                         ^~~~~~~~~~~~~~~
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_softlet/agnostic/common/os/mos_utilities.h:3025:33: note: in definition of macro ‘MOS_ZeroMemory’
     MosUtilities::MosZeroMemory(pDestination, stLength)
                                 ^~~~~~~~~~~~
/media/king/1000G/006_code/work-space/libva/gmmlib/build/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:590:59: error: ‘GMM_CLIENT_CONTEXT’ {aka ‘class GmmLib::GmmClientContext’} has no member named ‘CreateCustomResInfoObject_2’; did you mean ‘CreateCustomResInfoObject’?
         gmmResourceInfo = mediaDrvCtx->pGmmClientContext->CreateCustomResInfoObject_2(&gmmCustomParams);
                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                           CreateCustomResInfoObject
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/build.make:3828：media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/linux/common/ddi/media_libva_util.cpp.o] 错误 1
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/media_softlet/agnostic/Xe_M/Xe_HPM/codec/hal/enc/av1/features/encode_av1_vdenc_const_settings_xe_hpm.cpp.o                                                               
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/media_softlet/agnostic/Xe_M/Xe_HPM/codec/hal/enc/hevc/features/encode_hevc_vdenc_feature_manager_xe_hpm.cpp.o                                                            
make[1]: *** [CMakeFiles/Makefile2:294：media_driver/CMakeFiles/iHD_drv_video_COMMON.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/media_softlet/agnostic/Xe_M/Xe_HPM/codec/hal/enc/hevc/pipeline/encode_hevc_vdenc_pipeline_xe_hpm.cpp.o                                                                   
[ 72%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_CODEC.dir/media_softlet/agnostic/Xe_M/Xe_HPM/codec/hal/enc/hevc/pipeline/encode_hevc_vdenc_pipeline_adapter_xe_hpm.cpp.o     

### What's the usage scenario when you are seeing the problem?

Others

Xiaogangli-intel, 2022-09-07:
Hi @kingskill, did you install the latest libva? If there's old version libva in your machine, please uninstall it and install the latest libva again.

kingskill, 2022-09-08:
> Hi @kingskill, did you install the latest libva? If there's old version libva in your machine, please uninstall it and install the latest libva again.

我是按照下面这个链接里的教程进行的。
https://github.com/intel/media-driver/wiki/How-to-build-and-use-media-driver

另外，
`libva-dev 已经是最新版 (2.14.0-1)。
libva-drm2 已经是最新版 (2.14.0-1)。
libva-drm2-dbgsym 已经是最新版 (2.14.0-1)。
libva-glx2 已经是最新版 (2.14.0-1)。
libva-glx2 已设置为手动安装。
libva-glx2-dbgsym 已经是最新版 (2.14.0-1)。
libva-wayland2 已经是最新版 (2.14.0-1)。
libva-wayland2 已设置为手动安装。
libva-wayland2-dbgsym 已经是最新版 (2.14.0-1)。
libva-x11-2 已经是最新版 (2.14.0-1)。
libva-x11-2-dbgsym 已经是最新版 (2.14.0-1)。
libva2 已经是最新版 (2.14.0-1)。
libva2-dbgsym 已经是最新版 (2.14.0-1)。
有一些软件包无法被安装。如果您用的是 unstable 发行版，这也许是
因为系统无法达到您要求的状态造成的。该版本中可能会有一些您需要的软件
包尚未被创建或是它们已被从新到(Incoming)目录移出。
下列信息可能会对解决问题有所帮助：

下列软件包有未满足的依赖关系：
 libvariable-magic-perl-dbgsym : 依赖: libvariable-magic-perl (= 0.62-1) 但是 0.62-1+b1 正要被安装
E: 无法修正错误，因为您要求某些软件包保持现状，就是它们破坏了软件包间的依赖关系。
`

kingskill, 2022-09-08:
> Hi @kingskill, did you install the latest libva? If there's old version libva in your machine, please uninstall it and install the latest libva again.

My os is: deepin 20.7 (based on debian 10.10)

Xiaogangli-intel, 2022-09-30:
Please check if VADisplayPCIID is defined in /usr/include/va/va.h. if not, please update your libva to the latest version

XinfengZhang, 2022-11-16:
VADisplayPCIID is defined in 2.15.0,  and version check should already been added in #1422 , so I guess there are multiple version libva on your machine , also , other CreateCustomResInfoObject_2 is releated with gmmlib version, latest gmmlib should fix it

closed 2022-11-16 by XinfengZhang