intel/media-driver #1486: [Bug]: no function __vaDriverInit_1_0

moh-eulith, 2022-09-08:
### Which component impacted?

Not sure

### What happened?

I'm seeing random process crashes with VA errors. Installed `vainfo` and it couldn't find a working driver. So I googled around and figured I should install intel/media-driver.

I'm running on a model name      : 12th Gen Intel(R) Core(TM) i7-1260P.
Linux Mint.

There are nasty kernel bugs with this CPU, so I have to run drm-tip kernel with a custom patch (https://gitlab.freedesktop.org/drm/intel/-/issues/6342)

To install media-driver, I did the following:
```
apt install autoconf libtool libdrm-dev xorg xorg-dev openbox libx11-dev libgl1-mesa-glx libgl1-mesa-dev
git clone https://github.com/intel/libva
cd libva
sudo apt-get install -y libdrm-dev libegl1-mesa-dev libgl1-mesa-dev libx11-dev libxext-dev libxfixes-dev libwayland-dev
./autogen.sh --prefix=/usr
make
make install
cd ..
cat /sys/kernel/debug/dri/0/gt/uc/guc_info
cat /sys/kernel/debug/dri/0/gt/uc/huc_info
git clone https://github.com/intel/gmmlib
cd gmmlib/
git checkout intel-gmmlib-22.1.4
mkdir build && cd build
cmake -DMAKE_BUILD_TYPE=Release -DARCH=64 ..
make 
make install
cd ../..
git clone https://github.com/intel/media-driver/
cd media-driver
git checkout intel-media-22.4.4
cd ..
mkdir build_media
cd build_media
cmake ../media-driver
make -j8
make install
export LIBVA_DRIVERS_PATH=/usr/lib/dri/iHD_drv_video.so 
export LIBVA_DRIVER_NAME=iHD
```
running `vainfo` now produces:
```
error: XDG_RUNTIME_DIR not set in the environment.
libva info: VA-API version 1.7.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/dri/iHD_drv_video.so
libva error: /usr/lib/dri/iHD_drv_video.so has no function __vaDriverInit_1_0
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```
(I also tried building HEAD, but got the same results)

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

getting random crashes of a few processes.

### Debug Information

1. see above.
2. 
```
ls /dev/dri
by-path  card0  renderD128
```
3. see above.
4. The only thing I'm running that consistently shows something wrong is `vainfo`. The log from running that is not interesting:
```
[23860.360549][ctx       none]==========va_TraceInitialize
[23860.360561][ctx       none]=========vaInitialize ret = VA_STATUS_ERROR_UNKNOWN, unknown libva error 
[23860.360566][ctx       none]==========va_TraceTerminate
```
5. I don't think the GPU is hanging

### Do you want to contribute a patch to fix the issue?

No.

Jexu, 2022-09-16:
Correct path:
export LIBVA_DRIVERS_PATH=/usr/lib/x86_64-linux-gnu/dri/

moh-eulith, 2022-09-16:
On my machine, `make install` installs into `/usr/lib/dri`
```
# make install
[  2%] Built target igfxcmrt
[  2%] Built target iHD_drv_video_SSE2
[ 35%] Built target iHD_drv_video_COMMON
[ 80%] Built target iHD_drv_video_CODEC
[ 80%] Built target iHD_drv_video_CP
[ 86%] Built target iHD_drv_video_mos
[ 86%] Built target iHD_drv_video_SSE4
[ 99%] Built target iHD_drv_video_VP
[ 99%] Built target iHD_drv_video
[100%] Built target iHD_drv_video_STATIC
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/lib/dri/iHD_drv_video.so
-- Up-to-date: /usr/local/lib/libigfxcmrt.so.7.2.0
-- Up-to-date: /usr/local/lib/libigfxcmrt.so.7
-- Up-to-date: /usr/local/lib/libigfxcmrt.so
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt_g8.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt_g9.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt_g10.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt_g11.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt_g12_tgl.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt_g12_dg1.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_hw_vebox_cmd_g10.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt_def_os.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt_api_os.h
-- Up-to-date: /usr/local/include/igfxcmrt/cm_rt_extension.h
-- Up-to-date: /usr/local/lib/pkgconfig/igfxcmrt.pc
```

moh-eulith, 2022-09-16:
I even tried copying the .so into `/usr/lib/x86_64-linux-gnu/dri/`, but it didn't make any difference.

Jexu, 2022-09-16:
>>>libva info: VA-API version 1.7.0
Your vaapi version is too old that maybe outdate va in system instead of the one you installed.

moh-eulith, 2022-09-16:
There are no build instructions for libva. Using `make && make install` installs into /usr/local/lib, which apparently isn't even used anymore on ubuntu (there is no entry for it in `/etc/ld.so.conf.d/`). If I use `configure --prefix /lib/x86_64-linux-gnu`, then everything ends up in the wrong place (`/lib/x86_64-linux-gnu/lib` -- extra lib at the end). 

Is there a simple configure command to make it put stuff in the right place on ubuntu?

moh-eulith, 2022-09-16:
The following worked: when installing libva, use `./configure --libdir=/lib/x86_64-linux-gnu`.
building `intel-media` still installs `iHD_drv_video.so` into the wrong path (`/usr/lib/dri`), which then has to be copied over manually into `/lib/x86_64-linux-gnu/dri`

Xiaogangli-intel, 2022-09-30:
@moh-eulith, do all work well on your side?
When you build libva on Ubuntu, please always add `--libdir=/usr/lib/x86_64-linux-gnu`, and build media driver with `-DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu`. These will install all libraries into right places.

moh-eulith, 2022-09-30:
It works. Please add those flags to the documentation.

Jexu, 2022-10-25:
I close this issue now and feel free to re-open again if having any other questions.

closed 2022-10-25 by Jexu