intel/media-driver #1489: [Bug]: vapostproc  (scaling with scale-method=default) + vacompositor (composition) observed composition is using VEBOX engine.

SiewHoon, 2022-09-15:
### Which component impacted?

Video Processing

### What happened?

Play around using 4CH decode output pass to vacompositor (scale-method=default/fast)  just doing composition, it was showing is using EU engine.

Play around using 4CH decode output pass to vapostproc (scale-method=default)  just doing scaling, it was showing is using VEBOX engine.

We are doing 64CH decoding, from decode output will be pass to vapostproc to do scaling. Then each 16CH output from vapostproc will be passing to vacompositor doing composition. What we seeing only VEBOX engine is using, no usage in EU engine.

Using metrics_monitor tools from MediaSDK library to look on GPU engine usage.

We are running this 64CH usage using gstreamer-va plugins.

Here is command that we are using to run 64CH inside script.
[script_64CH_vapostproc_fakesink_nv12.sh.txt](https://github.com/intel/media-driver/files/9574314/script_64CH_vapostproc_fakesink_nv12.sh.txt)

### What's the usage scenario when you are seeing the problem?

Others

### Debug Information

libva: 2.15.0
libva-utils: 2.15.0
gmmlib: intel-gmmlib-22.1.4
media-driver: intel-media-22.4.4
gstreamer commit id: e0564b04c6cd08e62904903992b4de4cb50bb4b7
( https://gitlab.freedesktop.org/gstreamer/gstreamer.git)  

Platform:  Intel TGL NUC11PAHi7 
RAM: 32GB
OS: Ubuntu 20.04.4 LTS

stellawuintel, 2022-09-16:
Auto Created VSMGWL-56939 for further analysis.

FurongZhang, 2022-09-20:
I checked this issue. From the script file, it seems VAAPI is called per stream. 

That is to say, Stream0 with default scaling mode. Stream1 with default scaling mode. Then, all workload are in VEBOX+EU. 

Recommend to check with GStreamer.

ceyusa, 2022-09-20:
https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/1451

FurongZhang, 2022-09-22:
@SiewHoon , as we discussed, please discuss it with @uartie and feel free to let me know if any questions. Since we have already had conclusion, I will close this issue as of now. Please feel free to reopen it when you have questions.

closed 2022-09-22 by FurongZhang

uartie, 2022-10-19:
> Please see how to call VAAPI to composite via EU:https://github.com/intel/libva-utils/blob/master/videoprocess/vppblending.cpp#L1155

@FurongZhang Not sure if your link is still correct (since it's not a permalink).  When I clicked it, it took me to this code (which I'm posting as permalink, here): https://github.com/intel/libva-utils/blob/de2bc6821673a0708294f0e5e0452e6094db6762/videoprocess/vppblending.cpp#L1155.

So assuming the code in you link is still the intended target (as my permalink), my response is...

Submitting all of the composite buffers in one call to `vaRenderPicture` is not why the `EU` gets used.  It's because `VA_FILTER_SCALING_FAST` is set in the linked code (https://github.com/intel/libva-utils/blob/de2bc6821673a0708294f0e5e0452e6094db6762/videoprocess/vppblending.cpp#L1125).

Composite will use `VEBOX` if `VA_FILTER_SCALING_DEFAULT` is used, even if we're not scaling at the same time (as long as the format is supported, e.g. NV12).  The other scaling methods push it to the `EU` or when format is not supported on `VEBOX` (e.g. YV12).

Also, this behavior is not just isolated to VPP composite use case.  The same behavior can be seen with various other VPP operations (i.e. it can be replicated with gstreamer's vapostproc).

> https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/1451

@ceyusa It shouldn't matter whether `vaRenderPicture` is called multiple times for each buffer vs. calling it once with a list of buffers.  The call(s) only have to happen between the `vaBeginPicture` and `vaEndPicture`.  The former is actually documented in libva (https://github.com/intel/libva/blob/ed7b12d131fbfaaebb91361e57e44c393897ec39/va/va_vpp.h#L189-L234).  To prove it, I changed vacompositor to submit all buffers in a single call to `vaRenderPicture` (I will post the changes to the gstreamer issue).  I then ran my benchmarks again and did not observe any difference in GPU usage or render speed.

Thus, I don't think there is any issue in current gstreamer implementation or the driver.

@SiewHoon the vacompositor uses default scaling method by default.  Why can't you just set it to `hq` or `fast` to force `EU`?

uartie, 2022-10-19:
One other note, when blend flags are set in the pipeline parameters, it also gets moved onto EU regardless of scale method.  The vacompositor only sets blend flags when user requests surface transparency (i.e. when alpha != 1).

uartie, 2022-10-26:
@FurongZhang @SiewHoon any comment?