intel/media-driver #1490: [Bug]:

stolk, 2022-09-15:
### Which component impacted?

Build

### What happened?

The debug build is broken.

**use after free**

```
In file included from /home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_utilities.h:2778,
                 from /home/stolk/src/media-driver/media_common/agnostic/common/os/mos_os.h:31,
                 from /home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_utilities_next.cpp:34:
/home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_utilities_next.cpp: In static member function ‘static void* MosUtilities::MosReallocMemoryUtils(void*, size_t, const char*, const char*, int32_t)’:
/home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_util_debug.h:606:29: error: pointer ‘oldPtr’ may be used after ‘void* realloc(void*, size_t)’ [-Werror=use-after-free]
  606 |     MosUtilDebug::MosMessage(_level, _compID, _subCompID, MOS_FUNCTION, __LINE__, _message, ##__VA_ARGS__)
      |     ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_util_debug.h:671:5: note: in expansion of macro ‘MOS_DEBUGMESSAGE’
  671 |     MOS_DEBUGMESSAGE(MOS_MESSAGE_LVL_MEMNINJA, _compID, _subCompID, _message, ##__VA_ARGS__)
      |     ^~~~~~~~~~~~~~~~
/home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_util_debug.h:1351:5: note: in expansion of macro ‘MOS_MEMNINJAMESSAGE’
 1351 |     MOS_MEMNINJAMESSAGE(MOS_COMPONENT_OS, MOS_SUBCOMP_SELF, _message, ##__VA_ARGS__)
      |     ^~~~~~~~~~~~~~~~~~~
/home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_utilities.h:2788:5: note: in expansion of macro ‘MOS_OS_MEMNINJAMESSAGE’
 2788 |     MOS_OS_MEMNINJAMESSAGE(                                                                                                                                 \
      |     ^~~~~~~~~~~~~~~~~~~~~~
/home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_utilities_next.cpp:421:13: note: in expansion of macro ‘MOS_MEMNINJA_FREE_MESSAGE’
  421 |             MOS_MEMNINJA_FREE_MESSAGE(oldPtr, functionName, filename, line);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_utilities_next.cpp:412:21: note: call to ‘void* realloc(void*, size_t)’ here
  412 |     newPtr = realloc(ptr, newSize);
      |              ~~~~~~~^~~~~~~~~~~~~~
```

**will never be NULL**

```
/home/stolk/src/media-driver/media_softlet/agnostic/common/codec/hal/codechal_debug.cpp: In member function ‘virtual MOS_STATUS CodechalDebugInterface::DumpRgbDataOnYUVSurface(PMOS_SURFACE, const char*, const char*, MEDIA_DEBUG_STATE_TYPE, uint32_t, uint32_t)’:
/home/stolk/src/media-driver/media_softlet/agnostic/common/codec/hal/codechal_debug.cpp:714:9: error: the address of ‘MOS_SURFACE::OsResource’ will never be NULL [-Werror=address]
  714 |     if (&(tmpRgbSurface->OsResource))
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/stolk/src/media-driver/media_common/agnostic/common/shared/null_hardware.h:30,
                 from /home/stolk/src/media-driver/media_common/agnostic/common/os/mos_os.h:48,
                 from /home/stolk/src/media-driver/media_softlet/agnostic/common/hw/mhw_utilities_next.h:31,
                 from /home/stolk/src/media-driver/media_common/agnostic/common/hw/mhw_utilities.h:29,
                 from /home/stolk/src/media-driver/media_softlet/agnostic/common/shared/media_debug_interface.h:34,
                 from /home/stolk/src/media-driver/media_softlet/agnostic/common/codec/hal/codechal_debug.h:30,
                 from /home/stolk/src/media-driver/media_softlet/agnostic/common/codec/hal/codechal_debug.cpp:27:
/home/stolk/src/media-driver/media_common/linux/common/os/mos_os_specific.h:312:25: note: ‘MOS_SURFACE::OsResource’ declared here
  312 |     MOS_RESOURCE        OsResource;                                             //Surface Resource
      |                         ^~~~~~~~~~
```

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

Building

stolk, 2022-09-15:
NOTE: Debug build also fails when using clang instead. But with a different error:
```
/home/stolk/src/media-driver/media_driver/media_softlet/agnostic/Xe_R/Xe_HP_Base/renderhal/renderhal_xe_hp_base.cpp:108:90: error: result of comparison of constant 8 with expression of type 'MHW_ROTATION' (aka '_MHW_ROTATION') is always true [-Werror,-Wtautological-constant-out-of-range-compare]
    MHW_RENDERHAL_ASSERT(pRenderHalSurface->Rotation >= 0 && pRenderHalSurface->Rotation < 8);
                                                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~
/home/stolk/src/media-driver/media_common/agnostic/common/renderhal/renderhal.h:48:60: note: expanded from macro 'MHW_RENDERHAL_ASSERT'
    MOS_ASSERT(MOS_COMPONENT_CM, MOS_CM_SUBCOMP_RENDERHAL, _expr)
                                                           ^~~~~
/home/stolk/src/media-driver/media_softlet/agnostic/common/os/mos_util_debug.h:570:10: note: expanded from macro 'MOS_ASSERT'
    if(!(_expr))                                                 \
         ^~~~~

```

stolk, 2022-09-15:
Related to #1450 
I'm creating a pull rq that fixes these builds.

Xiaogangli-intel, 2022-09-30:
@stolk, thanks for your report and PR, I have some comments on your PR, please help to check it.

eero-t, 2023-01-18:
@hye5 Any progress with this?

hye5, 2023-01-18:
Added Furong to help review the PR https://github.com/intel/media-driver/pull/1491

FurongZhang, 2023-02-23:
We have reviewed the code changes and working on merging it into the main branch.

VickyZengg, 2023-03-09:
Hi Stock, the PR already merged, can you verify it and close the issue? Thanks!

stolk, 2023-03-09:
Verified with `-D CMAKE_BUILD_TYPE=Debug -D BUILD_TYPE=Debug` and the build now succeeds.

closed 2023-03-09 by stolk