intel/media-driver #1495: [Bug]: build error with libva:8682f9e

Bin-CI, 2022-09-20:
### Which component impacted?

Build

### What happened?

build passed with: https://github.com/intel/libva/commit/a65e00b00120051b2457692f92575db396e938bf
build error with: https://github.com/intel/libva/commit/8682f9e30f2fabf2ccc6f7609db035ed1af44703

### What's the usage scenario when you are seeing the problem?

Others

### What impacted?

media-driver should support the libva commit

Bin-CI, 2022-09-20:
link PR: https://github.com/intel/media-driver/pull/1492

dvrogozh, 2022-09-20:
Yes, that's known that that build fails. We were warning about that in https://github.com/intel/libva/pull/608. Fix is in #1492. And in https://github.com/oneapi-src/oneVPL-intel-gpu/pull/237 for VPL RT. Due to delayed merge process we can't avoid this build break.

uartie, 2022-09-20:
Pity that we can't overrule the merge process.

dvrogozh, 2022-09-20:
> Pity that we can't overrule the merge process.

Yes.