intel/media-driver #1503: [Bug]: Corrupted encode when using HEVC hardware encoder with ffmpeg

eric, 2022-09-30:
### Which component impacted?

Encode

### Is it regression? Good in old configuration?

No, this issue exist a long time

### What happened?

1. In Linux with ffmpeg installed, run:

```
ffmpeg -hide_banner -nostats -hwaccel vaapi -hwaccel_output_format vaapi -noautorotate -fflags discardcorrupt+genpts -copyts -start_at_zero -probesize 8000000 -i testfile.mpg -enc_time_base -1 -max_muxing_queue_size 4096 -muxdelay 0 -map 0:v:0\? -map 0:a\? -ignore_unknown -map 0:s\? -c:v hevc_vaapi -init_hw_device vaapi=intel:/dev/dri/renderD128 -b:v 7232k -minrate 6508k -maxrate 7955k -bufsize 14464k -c:a aac -ac 2 -b:a 384k -filter:a aresample=async=1 -c:s copy -f mpegts testfile-hevc.mpg
```

2. Playback the result in a video player (I used MPV)
3. See the screen has lots of green on it instead of the proper video

[image]

### What's the usage scenario when you are seeing the problem?

Transcode for media delivery

### What impacted?

I am unable to use the hardware HEVC encoder on this system.

### Debug Information

1. libva 2.11.0, gmmlib 21.1.2, media-driver 21.1.3
2. /dev/dri:
 
``` ls -l /dev/dri
total 0
drwxr-xr-x. 2 root root         80 Sep 30 15:40 by-path/
crw-rw----. 1 root video  226,   0 Sep 30 15:40 card0
crw-rw-rw-. 1 root render 226, 128 Sep 30 15:40 renderD128
```
3. [vainfo.log](https://github.com/intel/media-driver/files/9688031/vainfo.log)
4. I tried using `LIBVA_TRACE=/tmp/libva_trace.log` but it did not create any files
5. [dmesg.log](https://github.com/intel/media-driver/files/9688036/dmesg.log)

### Do you want to contribute a patch to fix the issue?

No.

intel-mediadev, 2022-11-16:
Auto Created VSMGWL-59031 for further analysis.

tmm1, 2023-01-09:
@XinfengZhang Do you have any idea about this issue?

tmm1, 2023-06-06:
Same issue also occurs on:

```
OS
Gigabyte Technology Z590I VISION D
Linux Alpine 3.13.2 (kernel: 5.15.79+truenas)

CPU
12 cores / 11th Gen Intel(R) Core(TM) i5-11400 @ 2.60GHz
```

nyanmisaka, 2023-06-07:
@tmm1 Have a try with https://github.com/intel/cartwheel-ffmpeg/blob/5.0/patches/0025-lavc-vaapi_encode_h265-fix-max_transform_hierarchy_d.patch

#1190 is the same issue.

tmm1, 2023-06-09:
@nyanmisaka that fixed it! thanks so much for the pointer

nyanmisaka, 2023-06-10:
@tmm1 Nice to hear that. Which ffmpeg version are you using?

The fix exists in n5.1+ versions. https://github.com/FFmpeg/FFmpeg/commit/9f02e033875185409c861846f209b04a3be339d2

Sherry-Lin, 2023-06-12:
Glad to hear the fix patch merged in latest FFMpeg. I'm going to close this issue. Please feel free to re-open it if any concerns.

closed 2023-06-12 by Sherry-Lin