intel/media-driver #1504: [Bug]: [TGL-U][ffmpeg-vaapi] When HCP scalability is enabled,  8K HEVC decode fps is lower than HCP scalability is disabled

UngTengEn, 2022-10-03:
### Which component impacted?

Decode

### What happened?

1. TGL-U i5-1145G7E; ubuntu 22.05 and kernel 5.15.0-48-generic.
2. Install or build libva 2.15.0 and media-driver 22.4.4 with HCP Scalability support.
3. Install ffmpeg 4.4.2
4. Download this video https://images-assets.nasa.gov/video/First-8K-Video-from-Space/First-8K-Video-from-Space~orig.mp4
5. Decode with ffmpeg-vaapi : "ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi_vld -i <nasa 8k video> -f null -".
6. With HCP scalability enabled, its fps is lower than with HCP scalability disabled.
[image]
[image]

### What's the usage scenario when you are seeing the problem?

Video Analytics

### Debug Information

1. libva 2.15.0 and media driver 22.4.4
2. root@p12dltglu208-Tiger-Lake-Client-Platform:~/Videos# ls /dev/dri
         by-path  card0  renderD128

[libva_trace_hcp_scalability_disabled.zip](https://github.com/intel/media-driver/files/9694940/libva_trace_hcp_scalability_disabled.zip)
[vainfo.log](https://github.com/intel/media-driver/files/9694941/vainfo.log)
[libva_trace_hcp_scalability_enabled.zip](https://github.com/intel/media-driver/files/9694942/libva_trace_hcp_scalability_enabled.zip)

UngTengEn, 2022-10-06:
Can enable or disable HCP Scalability with /etc/igfx_user_feature.txt.

Disable HCP scability with igfx_user_feature.txt :- cat /etc/igfx_user_feature.txt
[KEY]
0x00000001
UFKEY_INTERNAL\LibVa
[VALUE]
Disable MMC
4
1
[VALUE]
Enable HCP Scalability Decode
4
0

Ensable HCP scability with igfx_user_feature.txt :- cat /etc/igfx_user_feature.txt
[KEY]
0x00000001
UFKEY_INTERNAL\LibVa
[VALUE]
Disable MMC
4
1
[VALUE]
Enable HCP Scalability Decode
4
1

Jexu, 2022-10-24:
There is known limitation for scalability when decoding high bitrate content, in this case, we recommend to disable scalability manually.

UngTengEn, 2022-10-25:
Hi,

    Do we know when this will happen, like on what bitrates or video configuration?  Is there any errata for it?  Cause HCP scalability is plan to improve the performance.  So, what kind of recommendations should we suggest to the customer; like how the customer should determine the video configurations that should disable or enable the HCP scalability before playing them?

Thanks

Jexu, 2022-11-03:
Currently, we have the testing data that Virtual tile scalability for 4K bitstream with bitrate of 60Mbps will have perf drop. And 8K testing is under test.

Jexu, 2022-11-18:
I will close this issue since it is expected result under your bistream and please re-open again if having any other questions.